tsduck/tsduck #734: [Q] Pure C (subset) binding

lars18th, 2021-03-05:
Hi @lelegard ,

I don't want to disturb, just make one question if this can have sense at some point:
- The current Python and Java bindings are very powerful. At time we can use a lot of TSDuck functionalities without requiring a complex view of the toolkit. Just with a simple predefined chain (similar to the `tsp` process) and some "event handles" we can generate complex scripts/programs based on TSDuck. However, this is limited to Python, Java and C++ (yes, the last one is included too, but with increased complexity). So any pure C tool, or languages that can include a C binding without troubles (like NodeJS or LUA) are offside.

So even the main bindings will continue to be Python and Java, why not support a plain C binding too? Not full or complex interface, but a restricted API perhaps based on a subset of the current Python version.

What you think?  Completely stupid?

lelegard, 2021-03-05:
Sorry, but there is a simple answer : no way !

C is too dangerous, it lacks all modern tools for resource control. This is a 50-years old language, remember ?

I mean that we can do very stupid things in C++ (as well as in Java and even more easily in Python). But, with strict discipline, the language provides the tools to build robust code. I think I have demonstrated this with TSDuck. With C, you may have the strictest discipline on Earth, the language does not even give you the tools to make your discipline effective.

Any requested COBOL bindings? No, really?

lars18th, 2021-03-05:
OK. So I close it. The idea is for targeting a simple universal subset binding, to include support for other languages using an C API that's universally supported.

Don't worry. :wink: 
Regards.

closed 2021-03-05 by lars18th