tsduck/tsduck #735: [Q] Detailed Python binding documentation

lars18th, 2021-03-05:
Hi @lelegard ,

This issue is not a negative comment on any fault, as the TSDuck is one of the best documented opensource projects. But I need some help regarding how to implement the _side_ processing with TSDuck (discussed in #730):
- In order to set the TSP chain with the parameters `initial_input_packets`, `max_flushed_packets` and `max_input_packets`set to 1, I see in the _source_ of the `tsp.py` class that these members are defined in the class. However, I don't found in the documentation any reference to them (https://tsduck.io/doxy/jpbindings.html).
- A second comment is about how to use the class `TSProcessorArgs` with the Python binding. I can't found any example using it. However, it seems to be necessary to set the `--realtime` option in the `ts::TSProcessor` class.

So, can you help me please? Perhaps one additional example will be sufficient. 
Thank you for a fair and well done job!

lelegard, 2021-03-06:
> However, I don't found in the documentation any reference to them (https://tsduck.io/doxy/jpbindings.html).

The section you mention is just an introduction to Java and Python bindings. All Java and Python classes are documented by Doxygen, just like C++ classes. The class Python TSProcessor is documented here: https://tsduck.io/doxy/classts_1_1tsp_1_1_t_s_processor.html

For other Python classes, you may use the search field in the upper right corner or browse in the left pane: Modules > Other languages > Python bindings.

> A second comment is about how to use the class TSProcessorArgs with the Python binding. 

There is none. The TS processor args are directly set in the TSProcessor instance in Python.

> However, it seems to be necessary to set the --realtime option in the ts::TSProcessor class.

The `--realtime` option is just a command line convenience. It only modifies the default values of various parameters. Each plugin is free to check this option to select different default values. What is useful on the command line is maybe not so important in an application. In programming code I feel it is more important to explicitly set all parameters to understand what is done. This is vital for the maintainability of the application.

lars18th, 2021-03-06:
Hi @lelegard ,
OK. As everything is well explained I close this issue.
Thanks again!

closed 2021-03-06 by lars18th