tsduck/tsduck #736: TBS-6004 modulator card support

perydominet, 2021-03-07:
Hi There,
According to the information from Thierry meanwhile TSDuck does not handle the TBS-6004 DVB-C modulator card. I am looking for a volunteer who would be able to make TSDuck compatible and able to use the TBS-6004 DVB-C modulator card as an output destination. I am able participate in the full testing procedure (multiple input signal, multiple destination TVs), PAT logging, etc. Anybody interested in the TBS-6004 driver creation?
Thank you.
[image]

lelegard, 2021-03-07:
Hi @perydominet,

This is a good initiative. Could you also post directions to the software API and software delivery to interface these modulators? This is the required starting point for a TSDuck output plugin.

perydominet, 2021-03-07:
The only available software support I have found from the manufacturer TurboSight company is located here:
https://www.tbsdtv.com/download/index.html?path=24&id=123
There is a support@ email address, which they are usually responding in 1-3 working days. Email: support@tbsdtv.com

lelegard, 2021-03-07:
I downloaded the docs and software for all their modulator cards (3 models).

I notice that there is no programming documentation, no API, no software library, just a Linux device driver (with source code as most Linux device drivers do).

The application software is just a tarball with a very suspicious content: hundredths of kernel device drivers which can't possibly be all TBS, many PHP and Apache modules which are unrelated to TBS. So, there is no software packaging. Typically some developer has probably manually setup a working system and built an archive from it. I cannot possibly call it a "software product".

Additionally, there is no support for Windows. This is not a showstopper for me since I always insist on the fact that Linux is much better and more stable than Windows to build a technical system. But remember that all other classes of hardware devices which are supported by TSDuck (tuners, Dektec and HiDes devices) are all supported on Linux and Windows.

My conclusion is that the vendor created a proprietary system, without will to open it to external programmers, without software packaging, without programming API or documentation. Maybe someone could reverse their code and develop a hacky plugin for TSDuck but it may not be persistent across updates from TBS or new TBS products.

Before starting a plugin, we must check it it feasible. If TBS does not change its policy towards open source and third-party programming support, I am afraid there won't be any output plugin for TBS modulators.