tsduck/tsduck #738: [Q] [SRT] Operation not supported: Bad parameters

Plombirius, 2021-03-18:
Hi @lelegard ,

We have discovered some issue when we have tried establish SRT connection between Rassberi Pi 4 ("Raspbian GNU/Linux 10 (buster)") and x86 Ubuntu ("Ubuntu 20.04.2 LTS").
When we have started receiver side we got error on sender side "* Error: srt: error during srt_accept(), msg: Operation not supported: Bad parameters" 
What did we wrong?

We started sender with this string: "`tsp -v --debug=1 -I dvb -a 0 --delivery-system DVB-T2 --frequency 498000000 --bandwidth 8-mhz -O srt 5555 --transtype live --messageapi`" and receiver -- "`tsp -v --debug=1 -I srt 192.168.0.242:5555 --transtype live --messageapi -O drop`"

root@raspberrypi:/home/pi# tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.26-2285

root@srt:/home/srt# tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.26-2289

libsrt version on both side - 1.4.3

Thank you in advance!

lelegard, 2021-03-18:
Hi @Plombirius,

Did you try with libsrt 1.4.2?

I see that libsrt 1.4.3 is not yet released (-rc.0 has been out 10 days ago only). I do not know if there is any change that could produce this behavior. Moreover, not being an SRT specialist, I cannot really validate the command.

Maybe @maxsharabayko could answer both questions, 1.4.2 vs. 1.4.3 and validity of the command.

lelegard, 2021-03-18:
Hi,

I did two quick tests, using available configurations I have.

- Linux Ubuntu 20.10 with libsrt 1.4.1 (coming with the distro), both ends on the same machine, same commands as yours. It works.
- Windows 10 with libsrt 1.4.2, both ends on the same machine. Same error as yours: "Error: srt: error during srt_accept(), msg: Operation not supported: Bad parameters".

I do not know if anything changed with SRT version 1.4.2. This looks more like an SRT question rather than a TSDuck question. We definitely need @maxsharabayko's assistance. If he does not answer here, try to report the issue in [Haivision/srt](https://github.com/Haivision/srt), they are aware of TSDuck.

lelegard, 2021-03-18:
Also tested on macOS 11 with libsrt 1.4.2 (installed with Homebrew), same error.

Something must have changed in SRT between version 1.4.1 and 1.4.2.

Plombirius, 2021-03-18:
Thank you! 
I think as temporary solution we are recompiling with 1.4.1 tomorrow

lelegard, 2021-03-18:
> I think as temporary solution we are recompiling with 1.4.1 tomorrow

Please let us know the result. If something has changed in libsrt 1.4.2, we need to know what. In that case, I would recommend to ask the question in [Haivision/srt](https://github.com/Haivision/srt).

KvanSteijn, 2021-03-19:
Since 1.4.2 is the API/ABI changed for SRT: https://github.com/Haivision/srt/releases/tag/v1.4.2

lelegard, 2021-03-19:
Hi @KvanSteijn, thanks for your feedback.

@spnngl, could you have a look at this one please? The `srt` plugin is yours.

sartoricz, 2021-03-20:
Hi, just fyi, I have same issue with libsrt 1.4.3 rc0:

tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.25-2237
Built Mar 18 2021 - 23:52:54
Using GCC 8.3.0, C++ std 2011.03
Web library: libcurl: 7.64.0, ssl: OpenSSL/1.1.1d, libz: 1.2.11
SRT library: libsrt version 1.4.3
Dektec: This version of TSDuck was compiled without Dektec support

tsp -v -I srt 212.96.161.4:12345 --transtype live --messageapi --conn-timeout 100000 -P until --seconds 30 -P analyze --ts -O drop
* Error: srt: error during srt_recv(), msg: Connection was broken
*

lelegard, 2021-03-20:
I am a bit puzzled. I tried to reproduce it on my Mac, where I saw the problem a few days ago and it seems to work again...

On macOS 11.2.3:
~~~
$ tsp --version=all
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.26-2294
Built Mar 19 2021 - 18:22:08
Using Clang 12.0.0 (clang-1200.0.32.29), C++ std 2011.03
Web library: libcurl: 7.64.1, ssl: SecureTransport (LibreSSL/2.8.3), libz: 1.2.11
SRT library: libsrt version 1.4.2
Dektec: This version of TSDuck was compiled without Dektec support
~~~

Sender:
~~~
tsp -d -I file tsduck-test/input/test-001.ts -i -P regulate -O srt 5555 --transtype live --messageapi
~~~

Receiver:
~~~
tsp -d -I srt localhost:5555 --transtype live --messageapi -O drop
~~~

The sender is started first. It waits for a receiver. When the receiver starts, the exchange takes place and continues. When I interrupt the receiver using Ctrl-C, an error message `Connection was broken` is reported one both sides but this looks normal after a user interrupt.

lelegard, 2021-03-20:
More on SRT 1.4.2: 

On Windows, I just did exactly the same thing: rebuild the same version of TSDuck (3.26-2294) with the same libsrt 1.4.2. This time, on Windows, I get the error `Operation not supported: Bad parameters` again. See below.

@KvanSteijn, I understand your point about invalid parameters. However, I do not think that this is as simple as some invalid parameter which would be set in TSDuck. Consider this:

- It works for everybody everywhere with SRT 1.4.1.
- It works sometimes and fails most of the time with SRT 1.4.2 and 1.4.3.
- Please note that the error occurs _in the sender when the receiver connects to the sender_, after waiting for the receiver, not in some direct call with some incorrect parameter.

I really think that SRT folks should investigate this before it happens somewhere else, in a more professional environment.

~~~
$ tsp --version=all
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.26-2294
Built Mar 20 2021 - 18:01:17
Using MSVC 19.28.29913.00, C++ std 1997.11
Web library: WinInet
SRT library: libsrt version 1.4.2
Dektec: DTAPI: 5.43.2.149, Service: 3.8.4.91
$
$ tsp -d -I file test-001.ts -i -P regulate -O srt 5555 --transtype live --messageapi
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d
* Debug: debug level set to 1
* Debug: loading XML file C:\Users\thierry\Documents\tsduck\bin\Release-x64\tsduck.monitor.xml
* Debug: loading XML file C:\Users\thierry\Documents\tsduck\bin\Release-x64\tsduck.monitor.model.xml
* Debug: monitoring configuration loaded, 6 periods
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -I file test-001.ts -i
* Debug: srt: debug level set to 1
* Debug: srt: ====> tsp -O srt 5555 --transtype live --messageapi
* Debug: trying to load "C:\Users\thierry\Documents\tsduck\bin\Release-x64\tsplugin_regulate.dll"
* Debug: registering processor plugin "regulate", status: ok
* Debug: regulate: debug level set to 1
* Debug: regulate: ====> tsp -P regulate
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* regulate: minimum packet burst duration is 2,000,000 nano-seconds
* Debug: file: detected TS file format TS
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: initial input bitrate is 47,802,489 b/s
* Error: srt: error during srt_accept(), msg: Operation not supported: Bad parameters
* Debug: start() error in output plugin srt
* Debug: Report logging thread terminated
~~~

sartoricz, 2021-03-21:
Hi, 

i recompiled tsduck on Debian Stable with libsrt 1.4.1, 1.4.2 and 1.4.3 and here's the result. In short, with libsrt 1.4.3 and 1.4.2 sender crashes with ```Error: srt: error during srt_accept(), msg: Operation not supported: Bad parameters```, with libsrt 1.4.1 sender works, but crashes immediately after listener closes the connection - ```* Error: srt: error during srt_send(), msg: Connection was broken```. SRT listener works fine with all three libsrt versions.

```
uname -a
Linux asgard 4.19.0-13-amd64 #1 SMP Debian 4.19.160-2 (2020-11-28) x86_64 GNU/Linux
```

**libsrt 1.4.3**
```
tsp --version=all
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.25-2237
Built Mar 21 2021 - 01:49:59
Using GCC 8.3.0, C++ std 2011.03
Web library: libcurl: 7.64.0, ssl: OpenSSL/1.1.1d, libz: 1.2.11
SRT library: libsrt version 1.4.3
Dektec: DTAPI: 5.43.2.149

tsp -d -I file --infinite test.ts -P regulate -O srt 12314 --transtype live --messageapi
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d
* Debug: debug level set to 1
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -I file --infinite test.ts
* Debug: srt: debug level set to 1
* Debug: srt: ====> tsp -O srt 12314 --transtype live --messageapi
* Debug: trying to load "/usr/bin/tsplugin_regulate.so"
* Debug: /usr/bin/tsplugin_regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/regulate.so"
* Debug: /usr/bin/regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_regulate.so"
* Debug: /usr/lib64/tsduck/tsplugin_regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/regulate.so"
* Debug: /usr/lib64/tsduck/regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_regulate.so"
* Debug: /usr/lib64/tsplugin_regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/regulate.so"
* Debug: /usr/lib64/regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/local/lib64/tsduck/tsplugin_regulate.so"
* Debug: /usr/local/lib64/tsduck/tsplugin_regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/local/lib64/tsduck/regulate.so"
* Debug: /usr/local/lib64/tsduck/regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_regulate.so"
* Debug: registering processor plugin "regulate", status: ok
* Debug: regulate: debug level set to 1
* Debug: regulate: ====> tsp -P regulate
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* regulate: minimum packet burst duration is 10,000,000 nano-seconds
* Debug: file: detected TS file format TS
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: initial input bitrate is 16,000,011 b/s
* Error: srt: error during srt_accept(), msg: Operation not supported: Bad parameters
* Debug: Report logging thread terminated
```

**libsrt 1.4.2**
```
tsp --version=all
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.25-2237
Built Mar 21 2021 - 09:43:15
Using GCC 8.3.0, C++ std 2011.03
Web library: libcurl: 7.64.0, ssl: OpenSSL/1.1.1d, libz: 1.2.11
SRT library: libsrt version 1.4.2
Dektec: DTAPI: 5.43.2.149

tsp -d -I file --infinite test.ts -P regulate -O srt 12314 --transtype live --messageapi
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d
* Debug: debug level set to 1
[... 66 lines cut ...]
* Debug: /usr/local/lib64/tsduck/regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_regulate.so"
* Debug: registering processor plugin "regulate", status: ok
* Debug: regulate: debug level set to 1
* Debug: regulate: ====> tsp -P regulate
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* regulate: minimum packet burst duration is 10,000,000 nano-seconds
* Debug: file: detected TS file format TS
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: initial input bitrate is 16,000,011 b/s
* Debug: file: input thread started
* Debug: regulate: packet processing thread started
* regulate: regulated at bitrate 16,000,011 b/s
* Debug: regulate: new regulation, burst: 10,000,000 nano-seconds, 106 packets
* Debug: srt: output thread started
* Debug: control commands: found 1 packet processor plugins
* Debug: file: input: got bitrate 16,000,011 b/s
* Debug: file: input: got bitrate 16,000,011 b/s
* Debug: file: input: got bitrate 16,000,011 b/s
* Error: srt: error during srt_send(), msg: Connection was broken
* Debug: srt: output thread aborted after 185,213 packets (185,213 output)
* Debug: regulate: packet processing thread aborted after 185,426 packets, 185,426 passed, 0 dropped, 0 nullified
* Debug: file: input thread aborted after 274,560 packets
* Debug: Report logging thread terminated
```

spnngl, 2021-03-21:
Hi, I just created a PR for this.
It is caused by a missing parameter initialization I had forgotten.

Can you confirm it works for you too ?

lelegard, 2021-03-21:
Thanks @spnngl, this explains why I had the issue one day with libsrt 1.4.2 on macOS and no longer had it the next day _after rebuilding TSDuck_. I do not think that this is due to libsrt 1.4.2 vs previous version. The input value of `addrlen` was just some random value in the stack. So, depending on what previous calls left on the stack, the buffer length value could have been in valid range or outside of it. We were lucky to never get a segmentation fault. If the random uninitialised length value had been inside the valid range (no error from libsrt) but larger than the actual size of the buffer, we could have got a buffer overflow, corrupted memory and random crashes later in some other thread, in other words the worst debug nightmare.

This illustrates the importance of the following rule in the [TSDuck coding guidelines](https://tsduck.io/download/docs/tsduck-coding.pdf):
>  [Rule 119] Always initialize data with predictable values. Use zero for integers and pointers if there is no other meaningful values in the context.

I try to _always_ initialise variables in their declaration in the code, even if the same variable is assigned a couple of lines later. Just to be safe if the code is modified later.

sartoricz, 2021-03-21:
Hi  @spnngl, 
I recompiled tsduck with your patch against libsrt 1.4.3 and it works fine now, at least on Debian Stable.
Sender still crashes after receiver quits: Error: srt: error during srt_send(), msg: Connection was broken

```
tsp --version=all
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.25-2237
Built Mar 21 2021 - 13:49:03
Using GCC 8.3.0, C++ std 2011.03
Web library: libcurl: 7.64.0, ssl: OpenSSL/1.1.1d, libz: 1.2.11
SRT library: libsrt version 1.4.3
Dektec: DTAPI: 5.43.2.149

tsp -d -I file --infinite test.ts -P regulate -O srt 12314 --transtype live --messageapi
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d
* Debug: debug level set to 1
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -I file --infinite test.ts
* Debug: srt: debug level set to 1
* Debug: srt: ====> tsp -O srt 12314 --transtype live --messageapi
* Debug: trying to load "/usr/bin/tsplugin_regulate.so"
* Debug: /usr/bin/tsplugin_regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/regulate.so"
* Debug: /usr/bin/regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_regulate.so"
* Debug: /usr/lib64/tsduck/tsplugin_regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/regulate.so"
* Debug: /usr/lib64/tsduck/regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_regulate.so"
* Debug: /usr/lib64/tsplugin_regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/regulate.so"
* Debug: /usr/lib64/regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/local/lib64/tsduck/tsplugin_regulate.so"
* Debug: /usr/local/lib64/tsduck/tsplugin_regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/local/lib64/tsduck/regulate.so"
* Debug: /usr/local/lib64/tsduck/regulate.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_regulate.so"
* Debug: registering processor plugin "regulate", status: ok
* Debug: regulate: debug level set to 1
* Debug: regulate: ====> tsp -P regulate
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* regulate: minimum packet burst duration is 10,000,000 nano-seconds
* Debug: file: detected TS file format TS
* Debug: file: initial buffer load: 44,620 packets, 8,388,560 bytes
* file: initial input bitrate is 16,000,011 b/s
* Debug: file: input thread started
* Debug: regulate: packet processing thread started
* regulate: regulated at bitrate 16,000,011 b/s
* Debug: regulate: new regulation, burst: 10,000,000 nano-seconds, 106 packets
* Debug: srt: output thread started
* Debug: control commands: found 1 packet processor plugins
* Debug: file: input: got bitrate 16,000,011 b/s
* Debug: file: input: got bitrate 16,000,011 b/s
* Debug: file: input: got bitrate 16,000,011 b/s
* Debug: file: input: got bitrate 16,000,011 b/s
* Debug: file: input: got bitrate 16,000,011 b/s
* Error: srt: error during srt_send(), msg: Connection was broken
* Debug: srt: output thread aborted after 285,745 packets (285,745 output)
* Debug: regulate: packet processing thread aborted after 285,958 packets, 285,958 passed, 0 dropped, 0 nullified
* Debug: file: input thread aborted after 375,092 packets
* Debug: Report logging thread terminated
```

spnngl, 2021-03-21:
@lelegard You're right I should have initialized it. I sait it causes an error on libsrt >= 1.4.2 because before this release srt_accept just checked if `addrlen != NULL` and didn't care of the value, but after it also check if `*addrlen >= sizeof(sockaddr)`.

I will modify my commit with some more details.

@sartoricz It is not a crash in this case, it just let you know why it stopped. When I made the SRT implementation for tsduck I used it for "infinite" streams that's why it is marked as an error.

lelegard, 2021-03-21:
> Sender still crashes after receiver quits: Error: srt: error during srt_send(), msg: Connection was broken

@sartoricz, I agree with @spnngl, this is not a "crash", this is error handling. A crash would be a segmentation fault, an arithmetic trap or other hardware-detected error which forces a program to stop in the middle of execution. Here, a software library detects an error (here, the disconnection of the peer), the application reports the error and properly terminates.

The legitimate question is: how should the `srt` plugin handle the peer disconnection? If possible, the specific status should be checked and no error reported. Maybe a verbose-level message could display "receiver disconnection" or something similar. Other errors, not due to normal disconnection from the peer, shall be reported.

sartoricz, 2021-03-22:
@spnngl , @lelegard, I agree that "crash" is not a best word to describe this behavior and I apologize for choosing it. 
But I still think tsduck (sender) should not end when receiver disconnects - I'd expect some warning and wait for another client.

lelegard, 2021-03-22:
> But I still think tsduck (sender) should not end when receiver disconnects - I'd expect some warning and wait for another client.

This is a matter of specification. You expect this behaviour, someone else may expect another. In front of such dilemma, TSDuck usually provides both alternatives. I would suggest to implement an option (similar to `tsecmg`):
~~~
  -o
  --once
      Accept only one client and exit at the end of the session. By default, wait for another client.
~~~
@spnngl, what do you think? Is there also something similar to implement in the `srt` input plugin?

spnngl, 2021-03-23:
It seems to be a good idea.
Same for input yes, it could wait for another sender if a disconnection happens.

I don't think I will have time to do it in the near future, but if @sartoricz can do it (s)he's more than welcome !

lelegard, 2021-03-25:
Hi,

Since @spnngl does not have time to work in this, I added the option `--multiple` in the `srt` output plugin to wait for another receiver and continue sending when a new receiver connects. The default behaviour remains the same, the sender exits when the receiver disconnects. I have however removed the error message since a peer disconnection is not really an error, just an end of communication.

However, it unfortunately does not work. This is maybe an error on my side, not being an SRT specialist. I have asked the question here: https://github.com/Haivision/srt/issues/1892

lelegard, 2021-03-26:
Hi,

@sartoricz, the issue is fixed now. If you expect the `srt` output plugin to wait for another receiver when the previous one terminates, just add the option `--multiple`.

@spnngl, there were two issues in `SRTSocket` class.
- The first one is that `srt_startup()` and `srt_cleanup()` were unconditionally called in the constructor and destructor of `SRTSocket`. As a consequence, if two `SRTSocket` were open at the same time, the SRT library was shutdown when the first object was destroyed, making the other one unusable, even if it was still open. So, I first fixed this one. And then the second issue emerged.
-  When `srt_accept()` was used while opening an `SRTSocket`, its returned value is a data connection socket. This is a second SRT socket. The first one had become a listener socket after `srt_listen()`. This is the same principle as TCP. But in the code of `SRTSocket` the returned value of `srt_accept()` replaced the initial socket. So, when the `SRTSocket` was closed, the data socket was indeed closed but the listener socket was still there and still listening although its value was lost in `SRTSocket`. When `srt_cleanup()` was called in each `SRTSocket` destructor, everything was cleaned up in SRT and we could not see the issue.

closed 2021-04-16 by lelegard