tsduck/tsduck #739: [Q] Merge two streams

lyncher0, 2021-03-18:
Hello,
I'm trying to merge two VBR streams (with unrelated PCRs) into a CBR stream: -I null -P regulate -P merge stream 1 -P merge stream2 -P pcradjust
What would be the strategy to have tsduck to rewrite PCRs/PTSs of both streams?
Thanks

lelegard, 2021-03-18:
If the two streams are unrelated (ie. distinct services), there is no need to synchronise the PCR between the two. The plugin `pcradjust` should smoothen the PCR values according to the CBR rate. Be sure to test the results, there are not always good, depending on the original jitter.

lyncher0, 2021-03-18:
Thanks for the quick reply.
The goal is to merge an A/V service with the DVB-Teletext service. Even though they are unrelated at the input, they should be merged in the PMT which means that the PMT from DVB-Teletext will be dropped.

This is the resulting stream:

[image]

Which VLC is not able to render the subtitles (even though ffplay renders what is received).

lars18th, 2021-03-20:
Hi @lyncher0 ,

If your goal is to merge a DVB Teletext service from one source with a TV service from another source (both unrelated at source), then you require to synchronize the clocks. Without this synchronization it will be impossible to achieve this goal.

I've requested for a tool to achieve such type of synchronization (see https://github.com/tsduck/tsduck/issues/668#issuecomment-795066446). However, until @lelegard wants to implement it, we have no solution. So I hope at some point he will want to address such use cases.

On the other hand, if all you want to do is a muxing of two different VBR sources, the straightforward solution is: convert each one to CBR and mux them. With TSDuck you can convert any VBR stream to CBR (even in realtime), and after do the muxing of the two in a MPTS with CBR. Using TSDuck this processing is sufficient reliable at time. You can do it in realtime. The only requirement is to work every time with CBR streams. But at start you can convert any VBR to CBR without much effort. If the stream has valid PCR marks, then it will be trivial. If not, then using PTS/DTS timestamps from _known codecs_ you can achieve it too.

I hope this helps.

lyncher0, 2021-08-11:
Hi, after reading the recent TSDuck's new features (svresync) I'm trying to figure out if my scenario is now supported.
What I have are two SPTS CBR streams (A/V and DVB-Teletext) and what I'm looking for as output is a SPTS CBR stream with A/V + DVB-Teletext referenced in the same PMT:
tsp -I av_stream -P merge dvb_teletext_stream -O output

and what I get is an stream all the streams but:
 - PMT only references A/V stream
 - there's another PID with the PMT of the DVB-Teletext stream

I tried to merge the PMT with -P pmt but it seems that injecting a subtitle descriptor is only supported when using a XML.

Even though I could not get the intended output stream (got 2 PMTs instead) the new svresync indeed have done some remapping in PCR/PTS (after a few initial packets which were left untouched).

My questions is:
- what is the most effective approach to merge two PMTs (without needing to use an external XML file to pass all subtitle descriptor params)

Thank you!

lyncher0, 2021-09-29:
Found out that patch-xml did the trick!
Thanks

closed 2021-09-29 by lyncher0