tsduck/tsduck #740: [Q] Elementary stream to TS packets

PeteWilson7, 2021-03-19:
Using TSDuck is it possible to start with an elementary stream, then packatize the stream into a PES and then put the stream into TS packets?

PeteWilson7, 2021-03-19:
I know you can go from TS stream and pull out the ES and PES streams, but am trying to go the opposite direction. ES to PES to TS packets.

lelegard, 2021-03-20:
Hi @PeteWilson7 

There is no such plugin, currently. It should not be too difficult to add if the ES is already structured as PES, packetized elementary stream. It the ES is raw audio or video, without PES packets, then the technical specification of such a plugin shall be studied first.

lars18th, 2021-03-20:
Hi @PeteWilson7 ,

This is a subset of my request #668. At time, without new plugins targeting the PES management, one solution is to mix TSDuck with the very old OpenCaster PES tools.

lars18th, 2021-03-20:
Hi @lelegard ,

> There is no such plugin, currently. It should not be too difficult to add if the ES is already structured as PES, packetized elementary stream. It the ES is raw audio or video, without PES packets, then the technical specification of such a plugin shall be studied first.

From my point of view, it will be easy to target first the encapsulation of a simple PES (valid) or RAW (without any check) streams in TS packets. In the first case the type of the stream should be readed from the PES header, and in the second case (with RAW) the plugin requires to ask to identify it (or leave it unkown, to achieve a burden only encapsulation).

A second step could then be to encapsulate ES (with user identification or some automatic detection) streams inside PES packets.

What do you think? Will you decide to implement these tools sometime in the near future?
Please remember that I personally find this project fascinating. :wink: