tsduck/tsduck #741: [Q] Round recording using TSDuck

Vincent-Dabouineau, 2021-03-20:
I’m considering tsduck for some live multicast UHD events recording. I’m ready to script around the tsduck framework for scheduling some captures using the “time” and the “until” tsp plugins. But how to force tsduck for segmenting for e.g 8 hours of record into 30 minutes files, without MPEG-TS packets lost from this live event? In such use cases, it is difficult to manage one 8 hours of 125Mbps file per day. Thank you

lelegard, 2021-03-21:
> how to force tsduck for segmenting for e.g 8 hours of record into 30 minutes files

Try the `hls` output plugin. HLS usually serves a different purpose but `-O hls --duration 1800` will create 30 minutes segments.

We could consider an enhancement in the `file` output plugin to segment the output in multiple files of a given size. But, in the meantime, `hls` should do the job.

lelegard, 2021-03-21:
>  live multicast UHD events [...] of 125Mbps 

Multicast at 125 Mb/s, really? At that bitrate, I would only imagine an inter-operator feed, not multicast distribution. Out of curiosity, is it possible to get more information on the use case?

Vincent-Dabouineau, 2021-03-21:
For sure, it's all about the France TV UHD streams recording during our french tennis open (next month). Concerning the HLS plugin, may you confirm that there is lo differences on output files that the « classic » tsp -I ip usage (native MPEG-TS)?

lelegard, 2021-03-21:
> For sure, it's all about the France TV UHD streams recording during our french tennis open (next month).

Do you plan to multicast 125 Mb/s to end user's home boxes? I assumed that 40 Mb/s was more that enough for high quality 4K using HEVC encoding.

> Concerning the HLS plugin, may you confirm that there is lo differences on output files that the « classic » tsp -I ip usage (native MPEG-TS)?

Yes, MPEG-TS is the original format for HLS. Fragmented MP4 has been added later in the standard but TSDuck uses TS only.

In practice, the TSDuck `hls` output plugin does this:

- Copy the input TS in output TS files of more or less identical size (fixed file size or fixed duration). This is what you want.
- Add a copy of the PAT and PMT at the beginning of each file. It should not be a problem for you since the result is only two additional TS packets per file.
- Optionally try to locate intra frames as split points. You don't need it.
- Optionally generate an HLS playlist. You don't need it.
- Optionally cleanup obsolete segments with option `--live`. Don't use it.

So, at the end of the day, you get what you want.

Vincent-Dabouineau, 2021-03-21:
> > For sure, it's all about the France TV UHD streams recording during our french tennis open (next month).
> 
> Do you plan to multicast 125 Mb/s to end user's home boxes? I assumed that 40 Mb/s was more that enough for high quality 4K using HEVC encoding.
>
It has nothing to deal with the end users, I’m talking about some UHD HEVC contribution feeds, each with it’s 16 uncompressed audio streams. The broadcasted stream would be around 21Mbps, compliant with the french DTT capabilities.
> Yes, MPEG-TS is the original format for HLS. Fragmented MP4 has been added later in the standard but TSDuck uses TS only.
>
> In practice, the TSDuck `hls` output plugin does this:
>
> * Copy the input TS [...]
Thank you so much for explanations. In such use case, it can be interesting to split files using the duration expected, but handling IDR frames too (for future purpose, playout, editing, joining…). I didn’t know that TSDuck was able to operate at the bitstream level, interesting. Please tell me how to with the HLS plugins.

lelegard, 2021-03-21:
>  Please tell me how to with the HLS plugins.

The usual answer is [RTFM](https://tsduck.io/download/docs/tsduck.pdf) 😉

Vincent-Dabouineau, 2021-03-21:
> >  Please tell me how to with the HLS plugins.
> 
> 
> 
> The usual answer is [RTFM](https://tsduck.io/download/docs/tsduck.pdf) 😉 
> 
> 
> 
> 

Be sure I'm aware of everything around the HLS output plugin. I guess the answer may be to pass the `--label-close label1[-label2]` option? But which is the label for IDR frames inside the HEVC or AVC bitstream?

lelegard, 2021-03-21:
Labels have nothing to do with IDR frames. They are a way for plugins to "label" TS packets so that they are spotted by some later plugin. Here, RFTM means "read chapter 3, reference for the `tsp` command, _packet labelling_ section".

In chapter 4, reference for the `hls` output plugin, RTFM says:
> --intra-close
>
> Start new segments on the start of an intra-coded image (I-frame) of the reference video PID. The reference video PID is the first video PID of the first service in the PAT.
>
> By default, a new segment starts on a PES packet boundary on this video PID.
>
> Note that it is not always possible to guarantee the detection of I-frames if the video coding format is not fully supported, if the start of an intra-image cannot be found in the start of the PES packet which is contained in a TS packet or if the TS packet is encrypted.

To be precise, intra-frames are supposed to be detected in MPEG2, AVC, HEVC and VVC.

lelegard, 2021-03-23:
@Vincent-Dabouineau, 

Since it seems a useful generic feature for long-standing captures of huge files, I added options `--max-size` and `--max-duration` to the `file` output plugin. They automatically close an output file and create a new one (with another name) after either a given size or duration. RTFM as usual for the details.

In that case, this is strictly file splitting. Concatenating the output files gives the exact input. There is not I-frame detection or PAT/PMT reinsertion as `hls` does.

udhayNuk, 2022-11-03:
@lelegard 

tsp -I hls Idiot.mp4 -O hls manifest.m3u8

how to convert for VOD playlist

comment the full tsp mp4 to hls code

Thanking you in advance