tsduck/tsduck #743: [Q] MPTS

csphoniex, 2021-03-22:
Hello

i want make udp input with scrambler then out in mpts udp

lelegard, 2021-03-22:
Hi @csphoniex. "MPTS", "UPD input", "scrambler" and "UDP output" are all supported technical concepts in TSDuck. What is your question?

csphoniex, 2021-03-22:
i want make 16 channel with udp spts input and scrambling then out to mpts

lelegard, 2021-03-22:
If you have 16 distinct SPTS and you want to mux them into one MPTS, then you need a multiplexer. There is currently no accurate MUX in the TSDuck toolkit. This has been a long-standing request but developing an accurate real-time MUX is a lot of work which is not easily compatible with a free open source project.

csphoniex, 2021-03-23:
ok i want scrambe 16 spts and out to 16 spts then i will mux it in other software like mptsd or astra

lelegard, 2021-03-23:
Please be precise. There are many way of doing scrambling. It depends on algorithms, CAS, DRM, ECM or license insertion, key renewal, etc. If you need some help, you need to precisely describe what you want to do, technically speaking. For a start, what is the use case?

csphoniex, 2021-03-23:
can i have any con app like telegram or whatsapp to easy talk with you

csphoniex, 2021-03-23:
i want scrambling channel by key with caid

csphoniex, 2021-03-24:
@lelegard 

??

lelegard, 2021-03-24:
> @lelegard
> ??

Why should I spend time for free to analyze _your_ problem when you can't even take time to clearly explain your technical problem on more than one obscure line with zero technical detail.

csphoniex, 2021-03-24:
i make this config but this only for one spts i want make one file run for 16spts and if available to make remap for 16 spts 

tsp -v --bitrate 3880000 \
    -I ip 225.2.2.1:8002 --local-address 127.0.0.1 \
    -P svrename -p ExpressCAM -n Channel2 \
    -P scrambler "Channel2" --ecmg 192.168.0.201:42101 --super-cas-id 0x0b1c0002 --log-protocol=verbose --channel-id 0x0002 --pid-ecm 0x0203 --cp-duration 10000 --ecm-id 1 --access-criteria 00000001 \
    -O ip 228.2.2.1:8002 --local-address 127.0.0.1 \

lelegard, 2021-03-24:
What's wrong with this command?
Which CAS do you use?
Why do you use nearly 3 hours long crypto-periods?

csphoniex, 2021-03-24:
i want make one file scramble all 16 spts and about cas i want use static cw to can add it in softcam file oscam

csphoniex, 2021-03-25:
i scrambling channel but i cant descrambling in other side 

tsp -v --bitrate 6880000 \
    -I ip 225.2.2.1:8002  --local-address 127.0.0.1 \
    -P svrename -p ExpressCAM -n Channel2 \
    -P scrambler "Channel2" --ecmg 192.168.0.201:42102 --super-cas-id 0x0b1c0002 --log-protocol=verbose --pid-ecm 203 -b 30000 --channel-id 1 --cp-duration 10000 --ecm-id 2 --access-criteria 00000001 \
    -P cat -c -a 0x0B1C/0x01F0/FF0001 \
    -P datainject -r -s localhost:41100  -p 0x01F0 \
    -O ip 228.2.2.1:7002 --local-address 127.0.0.1  \

csphoniex, 2021-03-25:
what wrong in this config

lelegard, 2021-03-25:
TSDuck is a toolbox for digital television engineers.
OSCAM is a toolbox for illegal restreaming of protected contents.
Discussing TV piracy is not allowed here.
This topic is closed and locked.

closed 2021-03-25 by lelegard