tsduck/tsduck #744: [BUG] tspcontrol -t 2000 exit gives "tspcontrol: error connecting socket: Connection refused" error on ubuntu 20.04.

amit2maha1, 2021-03-24:
#### Bug description:
we are using 
`tsp --add-input-stuffing 1/100 --control-port 2000 -I ip -b 16777216 -l 127.0.0.1 225.1.2.3:2000 -P pmt --service 1 --add-pid 1280/0x86 -P spliceinject --service 1 -d --udp 2200 -O file --max-retry 5 -r --retry-interval 1000 C:/wafttech/cuecoder/encoder/assets/recording/8c5b1b5f-90cd-42c0-a974-a0048143472a/1616498079.ts`
to record ts stream

and using `tspcontrol -t 2000 exit` to kill the process

this works fine on windows but gives 'tspcontrol: error connecting socket: Connection refused' error on linux

lelegard, 2021-03-24:
Hi @amit2maha1,

It works for me on Ubuntu 20.10 (which should not make any difference).

Are you sure to run the `tsp` and `tspcontrol` on the same machine?  `tspcontrol` makes a TCP connection to the `tsp` process. The syntax is `tspcontrol -t [address:]port` where the default address is _localhost_. If `tsp` runs on another computer, you need to specify its host name or IP address.

amit2maha1, 2021-03-24:
> Hi @amit2maha1,
>
> It works for me on Ubuntu 20.10 (which should not make any difference).
>
> Are you sure to run the `tsp` and `tspcontrol` on the same machine? `tspcontrol` makes a TCP connection to [...]

yes its running on the same machine:
[image]

amit2maha1, 2021-03-24:
Also, the same thing is working on windows, btw.

lelegard, 2021-03-24:
In that case, it must be a firewall configuration issue on your Ubuntu system.

You can check if `tsp` is actually listening on port 2000 this way;
~~~
$ netstat -ant | grep :2000
tcp        0      0 0.0.0.0:2000            0.0.0.0:*               LISTEN     
~~~
If you see the above line and `tspcontrol` still cannot connect, then you have a firewall issue which does not allow the connection.