tsduck/tsduck #745: [BUG] Running sample plugin's build.ps1 repeatedly leads to errors

APTX, 2021-03-25:
#### Bug description:
Running [`build.ps1`](https://github.com/tsduck/tsduck/blob/master/sample/sample-plugin/build.ps1) many times leads to the following error:
```
Environment variable name or value is too long.
At C:\path\build.ps1:7 char:1
+ $env:Path = $env:Path + `
+ ~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : InvalidOperation: (:) [], ArgumentException
    + FullyQualifiedErrorId : ProviderContentWriteError
```

This also breaks the compiler:
```
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.Cpp.Current.targets(64,5): error MSB4018: The "SetEnv" task failed unexpectedly. [C:\path\project.vcxproj]
```

#### How to reproduce:
Run `build.ps1` enough times in the same terminal.

#### Expected behavior:
Error should not happen no matter how many times `build.ps1` is run.

#### Environment:
 - OS: Windows
 - OS version: 10
 - TSDuck full version: 3.25-2237
 - Installation type: official binary

#### Additional information:
`build.ps1` appends to `Path` every time it is run. Environment variables have a size limit. Eventually the limit is reached.

The main TSDuck `build.ps1` could also be affected (didn't check, but it's enough to see if `Path` gets longer and longer or not).

lelegard, 2021-03-25:
Hi @APTX 

Thanks for reporting this. This sample build script is completely outdated. I just fixed it. The build scripts in the other sample directories and, more importantly, the main TSDuck build scripts were not affected.

APTX, 2021-03-27:
Thanks for the fix!

closed 2021-03-27 by APTX