tsduck/tsduck #746: [Q] CPU

csphoniex, 2021-03-28:
hello

i run ffmpeg and tsp  but tsp use 100% from cpu and ffmpeg is 6% i only use it with audio and video copy option without transcoding 

[image]

lelegard, 2021-03-29:
`tsp` is not just "a command". Depending on the chain of plugins, the CPU may quite normally range from 1% (direct copy of a real-time stream) to 100% x nb CPU (many scrambling or descrambling plugins). So, without the exact command and exact environment, nobody can give a useful advice.

csphoniex, 2021-03-29:
this is my tsp config 

tsp -v \
-I file \
-P continuity --fix \
-P pes \
-P filter --max-payload-size 0 --negate -s \
-P scrambler "c1" --ecmg 192.168.0.201:42102 --ecmg-scs-version 2 --super-cas-id 0x0b1c0003 --pid-ecm 0x1EC1 --ecm-id 1 --channel-id 0x0001 --cp-duration 10000 --access-criteria 00000001 --stream-id 0x1001 \
-P scrambler "c2" --ecmg 192.168.0.201:42102 --ecmg-scs-version 2 --super-cas-id 0x0b1c0003 --pid-ecm 0x1EC2 --ecm-id 1 --channel-id 0x0001 --cp-duration 10000 --access-criteria 00000001 --stream-id 0x1002 \
-P scrambler "c3" --ecmg 192.168.0.201:42102 --ecmg-scs-version 2 --super-cas-id 0x0b1c0003 --pid-ecm 0x1EC3 --ecm-id 1 --channel-id 0x0001 --cp-duration 10000 --access-criteria 00000001 --stream-id 0x1003 \
-P scrambler "c4" --ecmg 192.168.0.201:42102 --ecmg-scs-version 2 --super-cas-id 0x0b1c0003 --pid-ecm 0x1EC4 --ecm-id 1 --channel-id 0x0001 --cp-duration 10000 --access-criteria 00000001 --stream-id 0x1004 \
-P scrambler "c5" --ecmg 192.168.0.201:42102 --ecmg-scs-version 2 --super-cas-id 0x0b1c0003 --pid-ecm 0x1EC5 --ecm-id 1 --channel-id 0x0001 --cp-duration 10000 --access-criteria 00000001 --stream-id 0x1005 \
-P scrambler "c6" --ecmg 192.168.0.201:42102 --ecmg-scs-version 2 --super-cas-id 0x0b1c0003 --pid-ecm 0x1EC6 --ecm-id 1 --channel-id 0x0001 --cp-duration 10000 --access-criteria 00000001 --stream-id 0x1006 \
-P scrambler "c7" --ecmg 192.168.0.201:42102 --ecmg-scs-version 2 --super-cas-id 0x0b1c0003 --pid-ecm 0x1EC7 --ecm-id 1 --channel-id 0x0001 --cp-duration 10000 --access-criteria 00000001 --stream-id 0x1007 \
-P scrambler "c8" --ecmg 192.168.0.201:42102 --ecmg-scs-version 2 --super-cas-id 0x0b1c0003 --pid-ecm 0x1EC8 --ecm-id 1 --channel-id 0x0001 --cp-duration 10000 --access-criteria 00000001 --stream-id 0x1008 \
-P cat -c -a 0x0B1C/0x11F2 \
-P datainject -r -s localhost:41101 -b 50000 -p 0x11F2 \
-P svrename -n "CH 1" -p "EXSAT" c1 --id 1001 \
-P svrename -n "CH 2" -p "EXSAT" c2 --id 1002 \
-P svrename -n "CH 3" -p "EXSAT" c3 --id 1003 \
-P svrename -n "CH 4" -p "EXSAT" c4 --id 1004 \
-P svrename -n "CH 5" -p "EXSAT" c5 --id 1005 \
-P svrename -n "CH 6" -p "EXSAT" c6 --id 1006 \
-P svrename -n "CH 7" -p "EXSAT" c7 --id 1007 \
-P svrename -n "CH 8" -p "EXSAT" c8 --id 1008 \
-O ip 224.1.1.1:1234 --local-address 172.16.1.4 --enforce-burst \

can i use one ecmg for all channels with

lelegard, 2021-03-29:
> can i use one ecmg connection for all channels and reduce cpu usage

This has nothing to do with ECMG. This is just the required CPU for the crypto you request.

The default scrambling is DVB-CSA-2 which is, by design and purposely, fast to implement in hardware and very slow to implement in software. You request 8 simultaneous DVB-CSA-2 scrambling sessions. This would kill any CPU.

The ECM generation and insertion is epsilon. There is nothing you can do, except adding more CPU cores.

Please read the TSDuck user's guide, at the beginning of chapter 4, right after the end of the list of plugins:

> Some plugins are related to the scrambling of TS packets and Conditional Access Systems. Please note the following:
> - The DVB-CSA scrambling algorithm is inherently and purposely very slow with a software implementation. A 3.4 GHz Pentium 4 CPU, for instance, cannot (de)scramble more than 20 Mb/s. Be cautious not to ask for impossible tasks, like real time (de)scrambling of a complete TS on a regular PC.
> - These tsp plugins are implemented for testing Conditional Access Systems, either on the headend or set-top box side. TSDuck does not provide any support to hack or circumvent Conditional Access Systems and will never do so. The CAS-related plugins require and use external CAS provided systems (ECMG, EMMG and smartcards). All secrecy and  proprietary CAS information remain isolated inside these external systems and TSDuck does not attempt to access this type of secret and private information. TSDuck only interacts with these systems using their external communication protocols.

The first bullet mentions a "3.4 GHz Pentium 4 CPU", the Linux machine I used when I wrote this paragraph 15 years ago. Since then, the pure CPU power hasn't much increased. Only the number of cores increased. Since each plugin executes in its own thread, `tsp` takes full advantage of multi-core, up to one core per plugin.

The second bullet is very important too. You already experimented it....

csphoniex, 2021-03-29:
Can i shorten all sessions into a single scrambler session connection

lelegard, 2021-03-30:
> Can i shorten all sessions into a single scrambler session connection

There is no such thing as a "scrambler connection". There may be a confusion with "ECMG connection". But the names are explicit: the ECM Generator generates ECM's. Nothing more. It does not scramble the stream.

The `scrambler` plugin does not connect anywhere to delegate the scrambling to someone else. The `scrambler` plugin, as its name implies, does the scrambling. The plugin directly executes the billions of instructions which encrypt the stream. This is why the `scrambler` plugin consumes so much CPU.

The amount of required CPU depends on the encryption algorithm and the amount of data. Concerning the algorithm, from my previous benchmarks, DVB-CSA2 requires roughly 50 times more CPU than AES and its derivatives. This is by design, to give hackers a hard time (see how effective it is). The amount of data depends on the audio/video bitrate and number of services. Here, you want to scramble 8 services at the same time. If each service uses 4 Mb/s, you have to scramble 32 Mb/s, which is huge for software implementation of DVB-CSA2.

Using a "single scrambler session" is not possible since each instance of the `scrambler` plugin processes one service only. But even it was possible to scramble more than one service in a single instance, that would be worse. 8 instances of the plugin are naturally spread over up to 8 different CPU cores, taking full advantage of parallelism. With only one instance, you would use only one thread and at most one CPU, which is clearly not enough to scramble that amount of data.

So, what you want to do requires a huge amount of CPU. As long as you keep that amount of data and that algorithm, you need that amount of CPU. If you have a multi-core machine (all machines are nowadays), your current configuration makes the best use of parallelism. In short, you are already in the optimal configuration.

csphoniex, 2021-03-30:
ok if i use static control word with ecm pid and caid so can descrambling  by other software like astra cesbo 

can you help me to make this config

closed 2026-04-26 by github-actions[bot]