tsduck/tsduck #748: [BUG] "input plugin pcap not found"

digitaltvguy, 2021-04-02:
#### Bug description:
When attempting to parse a pcap file with tsp an error occurs "input plugin pcap not found"

#### How to reproduce:
Run tsp -I pcap -d <multicast IP:port> <path to output file>

#### Expected behavior:
TS should be extracted

#### Errors and logs:
"input plugin pcap not found"

#### Environment:
 - OS: <macOS>
 - OS version: <11.2.3>
 - TSDuck full version: <3.25-2237>
 - Installation type: <MacOS homebrew>

#### Additional information:
None

lelegard, 2021-04-02:
Hi @digitaltvguy 

The plugin `pcap` was recently added, after the release of version 3.25. It will be released with version 3.26. In the meantime, I suggest to install [a nightly build version](https://tsduck.io/download/prerelease/) or [rebuild from sources](https://tsduck.io/doxy/building.html).

You did not mention the OS type but version 11.2.3 suggests macOS Big Sur. In that case, just do the following:
~~~
git clone https://github.com/tsduck/tsduck.git
cd tsduck
build/install-prerequisites.sh
make -j10 install
~~~

digitaltvguy, 2021-04-02:
Thanks for the quick response.  I’ll give the nightly a try.

Do you think someone could update homebrew with a “HEAD” option for nightlys?

All the best,
Chris

lelegard, 2021-04-03:
> Do you think someone could update homebrew with a “HEAD” option for nightlys?

I do not know has to manage pre-releases in Homebrew. Since installing TSDuck with Homebrew eventually ends up downloading the source archive and compiling it, I assume that an advanced user willing to install a pre-release can simply rebuild the repo as indicated.

digitaltvguy, 2021-04-03:
I added a pull request for a “HEAD” option for latest beta releases in Github homebrew formula.

-Chris

lelegard, 2021-04-04:
> I added a pull request for a “HEAD” option for latest beta releases in Github homebrew formula.

It works, thanks!

In order to avoid other users to do the same mistake as me, the correct syntax to install the latest repo is 
~~~
brew install --HEAD tsduck
~~~

And NOT `brew install tsduck --HEAD`.

digitaltvguy, 2021-04-04:
Thanks for the correction to my syntax

closed 2021-04-16 by lelegard

rodriguesl6, 2024-04-25:
Hi guys,

I tried the pcap to TS command and the result keeps the same: Input pcap not found.

tsp.exe -I Programacao_199_INPUT_%dD%HH%MM%SS -O Programacao_199_INPUT_%dD%HH%MM%SS
* Error: Programacao_199_INPUT_%dD%HH%MM%SS.dll: Impossl localizar o m󤵬o especificado.
* Error: input plugin Programacao_199_INPUT_%dD%HH%MM%SS not found
* Error: Programacao_199_INPUT_%dD%HH%MM%SS.dll: Impossl localizar o m󤵬o especificado.
* Error: output plugin Programacao_199_INPUT_%dD%HH%MM%SS not found

lelegard, 2024-04-25:
Please do not hijack 3-years old obsolete and close issues, open a new issue for new problems.

As far as I can read your errors, there no "Input pcap not found" message. The message is "input plugin Programacao_199_INPUT_%dD%HH%MM%SS not found", which is quite normal since you typed `-I Programacao_199_INPUT_%dD%HH%MM%SS`. You probably wanted `-I pcap Programacao_199_INPUT_%dD%HH%MM%SS`.

rodriguesl6, 2024-04-25:
Got it! You are completly right.

The command:
 tsp.exe -I pcap Programacao_199_INPUT_%dD%HH%MM%SS.pcap

Assumes the pcap plugin. The only point that I could't do is save the ouput in a ts file

lelegard, 2024-04-25:
>  The only point that I could't do is save the ouput in a ts file

Seriously, did you read the user's guide, at least a few examples?

The `file` plugin is used to read/write files: `tsp -I pcap ... -O file out.ts`

This forum is here to help users, not to read the documentation.