tsduck/tsduck #750: [Q] Not able to create a TS from scratch

movingDigital, 2021-04-03:
Hello Thierry,

First of all, amazing tool, thank you so much for your dedication and passion!

I am trying to use TSDuck in order to create a TS from scratch, but unfortunately when I am trying to check the TS output, using TSReader & Stream Xpert, I am not getting anything: no PAT, not PMT.

I have checked with Wireshark that I am getting the network packets correctly though.

Here is my very basic code:

**PAT XML:**
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
	<PAT transport_stream_id="0">
		<service service_id="1" program_map_PID="1"/>
	</PAT>
</tsduck>

**PMT XML:**
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
	<PMT service_id="1">
		<component elementary_PID="7501" stream_type="0x86">
		</component>
	</PMT>
</tsduck>

**Command line:**
>tsp.exe --verbose --bitrate 3000000 --max-flushed-packets 70 -I null -P regulate --packet-burst 14 -P inject test/pat.xml --pid 0 --bitrate 1500 -P inject test/pmt.xml --pid 1 --bitrate 1500 -O ip 172.168.1.221:4500 --packet-burst 7 --enforce-burst --local-address 172.168.1.222

The logs seem to say the command is working:
* inject: loaded 1 sections from test/pmt.xml, repetition rate: unspecified
* inject: loaded 1 sections from test/pat.xml, repetition rate: unspecified
* regulate: minimum packet burst duration is 2,000,000 nano-seconds
* null: initial input bitrate is 3,000,000 b/s
* regulate: regulated at bitrate 3,000,000 b/s
* inject: transport bitrate: 3,000,000 b/s, packet interval: 2,000
* inject: transport bitrate: 3,000,000 b/s, packet interval: 2,000

Do you see any issue in the XML definitions or in the command line?

lelegard, 2021-04-03:
Hi,

When a tool (here, TSReader & Stream Xpert) does not "see" something in your stream, you have to check first if what you are looking for (here, PAT & PMT) is really absent or if the tool just doesn't see it.

In a case like this, where you wonder what's in the TS, start with replacing the network output with a file: `-P until --seconds 5 -O file out.ts` and then see what's in the TS file using commands such as `tsanalyze` or `tstables`.

Doing this on your command shows that the PAT and PMT are there. So why don't TSReader & Stream Xpert detect them? I do not know for sure but I would bet that it is because you placed the PMT on PID 1. This is forbidden. PID 1 is reserved by MPEG for the CAT. Maybe TSReader & Stream Xpert simply consider your stream as invalid because there should be no PMT in PID 1. Just a guess, though.

The usual reserved ranges of PID's are:

- MPEG : 0x0000 - 0x000F
- DVB : 0x0010 - 0x001F
- ISDB : 0x0010 - 0x002F
- ATSC : 0x1FF0 - 0x1FFE

Allocating service data or information in these ranges is looking for troubles.

movingDigital, 2021-04-03:
Hi Thierry,

Thanks for your very quick help! So I changed the PMT PID to 1000, and checked with recording a TS file instead of pushing to the network, and TSReader / StreamXpert are able to find the PAT, but not the PMT :-( 

Here is my code:

PAT:
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
	<PAT transport_stream_id="0">
		<service service_id="1" program_map_PID="1000"/>
	</PAT>
</tsduck>

PMT: 
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
	<PMT service_id="1">
		<component elementary_PID="7501" stream_type="0x86">
		</component>
	</PMT>
</tsduck>

Command: 
>tsp.exe --verbose --bitrate 3000000 --max-flushed-packets 70 -I null -P regulate --packet-burst 14 -P inject test/pat.xml --pid 0 --bitrate 1500 -P inject test/pmt.xml --pid 1000 --bitrate 1500 -P until --seconds 5 -O file C:\out.ts

Do you have any idea of what I can be doing wrong?

lelegard, 2021-04-03:
It does not make sense. If I copy/paste your two XML files and your command, I get this:
~~~
* PAT, TID 0 (0x00), PID 0 (0x0000)
  Version: 0, sections: 1, total size: 16 bytes
  - Section 0:
    TS id:       0 (0x0000)
    Program:     1 (0x0001)  PID: 1000 (0x03E8)

* PMT, TID 2 (0x02), PID 1000 (0x03E8)
  Version: 0, sections: 1, total size: 21 bytes
  - Section 0:
    Program: 1 (0x0001), PCR PID: none
    Elementary stream: type 0x86 (SCTE 35 Splice Info), PID: 7501 (0x1D4D)
~~~
Which is exactly what's in your XML files.

> Do you happen to know why Program shows 700? I haven't defined this anywhere so I am curious.
Also, where does the PID 5001 come from? I also haven't defined it anywhere, so here as well I am curious.

There is also the "PID: 32" instead of 1000 in your PAT. The only explanation I can find is that the XML files you refer in your tsp command are not the same as the XML files you posted (subdirectory `test`?).

movingDigital, 2021-04-04:
Hi Thierry,

Oh my... Indeed, I got two files opened in Notepad++ with the same name, and I was working on the wrong one without even noticing it... Shame on me!

So now that I got this fixed, I still don't get why TSReader is not able to read my PMT.

The PAT XML is:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
	<PAT version="0" transport_stream_id="0">
		<service service_id="1" program_map_PID="32"/>
	</PAT>
</tsduck>
```

The PMT XML is:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
	<PMT version="1" service_id="1" PCR_PID="7501">
		<!-- Description of one SCTE-35 Splice Info PID in the TS -->
		<component stream_type="0x86" elementary_PID="7501">
		</component>
		<registration_descriptor format_identifier="0x43554549">
		</registration_descriptor>
	</PMT>
</tsduck>
```

The command line is: 
`tsp.exe --verbose --bitrate 3000000 --max-flushed-packets 70 -I null -P regulate --packet-burst 14 -P inject test/pat.xml --pid 0 --bitrate 1500 -P inject test/pmt.xml --pid 32 --bitrate 1500 -P until --seconds 5 -O file test/out.ts`

TSReader will only be able to find the PAT, but not the PMT.

However TSTables seems to be getting everything ok:
```
* PAT, TID 0 (0x00), PID 0 (0x0000)
  Version: 0, sections: 1, total size: 16 bytes
  - Section 0:
    TS id:       0 (0x0000)
    Program:     1 (0x0001)  PID:   32 (0x0020)

* PMT, TID 2 (0x02), PID 32 (0x0020)
  Version: 1, sections: 1, total size: 27 bytes
  - Section 0:
    Program: 1 (0x0001), PCR PID: 7501 (0x1D4D)
    Program information:
    - Descriptor 0: Registration (0x05, 5), 4 bytes
      Format identifier: 0x43554549 ("CUEI")
    Elementary stream: type 0x86 (SCTE 35 Splice Info), PID: 7501 (0x1D4D)
```

Do you see any issues/concern/warnings as per what TSTables is showing?

movingDigital, 2021-04-04:
Hi Thierry, I found the problem: the 5 seconds capture was not enough for TSReader to get the PMT table!

After I recorded 15 seconds instead, TS Reader can now display both PAT and PMT without any issue, thank you so much!

lelegard, 2021-04-04:
It is strange that TSReader needs that many occurrences of the PMT to detect it. Maybe it is because you specify an excessively low bitrate for PAT and PMT. 1500 b/s = one TS packet per second = one PAT or PMT per second (or less in case of larger PMT). A standard value would be 15000, 10 times more.

closed 2021-04-04 by movingDigital