tsduck/tsduck #751: [Q] How to get utc_splice_time since 00 hours UTC, January 6th, 1980?

movingDigital, 2021-04-04:
Hello Thierry,

I need to use the SCTE-35 XML table with splice_schedule() command.

According to the ANSI doc, the utc_splice_time is a 32-bit unsigned integer quantity representing the time of the signaled splice event as the number of seconds since 00 hours UTC, January 6th, 1980, with the count of intervening leap seconds
included.

Since it is different than the EPOCH time, have you developed a small piece of code able to get the above, or do you know any useful command able to do the same?

Thanks

lelegard, 2021-04-04:
Hi @movingDigital,

This is a very interesting question which brings other questions. This is why it will take some time to thing about it and make some tests and modifications.

First, checking the code, I realized there was a bug in the display of splice_schedule() commands. The 32-bit value was interpreted as a UNIX time (from Jan 1st 1970). I fixed it (except leap seconds). Only the display was affected.

Concerning your question, how to manipulate the SCTE 35 `utc_splice_time` 32-bit value, I initially thought we should be able to rely on the system time of the computer. Normally, "a time is a time" and an official time is a value which includes the leap seconds. Consequently, an operating system is supposed to manipulate official times. So, if `date1` and `date2` are system dates in seconds, the difference `date2-date1` should include the leap seconds. This is unfortunately not the case.

In the [Wikipedia article on leap seconds](https://en.wikipedia.org/wiki/Leap_second), we have a table of leap seconds. Let's consider two non-leap year (365 days): 2005 has a leap second on Dec 31 while 2006 has none.

How many seconds in a non-leap year, how many seconds in 2005 and 2006, on a macOS and a Ubuntu system (same results):
~~~
$ echo $((365 * 24 * 3600))
31536000
$ echo $(($(date '+%s' -d '2006-01-01 00:00:00 UTC') - $(date '+%s' -d '2005-01-01 00:00:00 UTC')))
31536000
$ echo $(($(date '+%s' -d '2007-01-01 00:00:00 UTC') - $(date '+%s' -d '2006-01-01 00:00:00 UTC')))
31536000
~~~
All the same, no leap second...

And Windows is not better. Using PowerShell on a Windows 10 20H2 system:
~~~
PS C:\> $y2005 = Get-Date -Year 2005 -Month 1 -Day 1 -Hour 0 -Minute 0 -Second 0 -Millisecond 0
PS C:\> $y2006 = Get-Date -Year 2006 -Month 1 -Day 1 -Hour 0 -Minute 0 -Second 0 -Millisecond 0
PS C:\> $y2007 = Get-Date -Year 2007 -Month 1 -Day 1 -Hour 0 -Minute 0 -Second 0 -Millisecond 0
PS C:\> (New-TimeSpan -Start $y2005 -End $y2006).TotalSeconds
31536000
PS C:\> (New-TimeSpan -Start $y2006 -End $y2007).TotalSeconds
31536000
~~~

So, yes, we have a problem I did not anticipated.

Let me think about the best way to solve it.

Stay tuned...

movingDigital, 2021-04-05:
Hi @lelegard 

Thank you so much once again for your swift response! 

I see, noted on the above. Looking at how the `utc_splice_time` must be implemented in the encoder/transcoder when using `spliceSchedule()` command, I really wonder if all the vendors really follow the standard by the book, or not.

While I don't see any difficulty in getting the number of seconds (without the leap ones) since 6th January 1980, I am not so sure implementing the "leap seconds support" is easy, since leap second is unpredictable. I see two scenarios:

1. The system (encoder/transcoder) has access to the information
2. Or not

I am pretty sure that most of the encoder/transcoder do not have access to the information (= number of leap seconds since 6th January 1980). 

Since SCTE-35 allows implementation to ignore the leap second (link [here](https://codesequoia.wordpress.com/2014/02/24/understanding-scte-35/)), I propose to keep it simple in your architecture by not taking care of the number of leap seconds. It would be too difficult to retrieve and maintain, since it means that TSDuck would probably need to remain connected to Internet 24/7 to make sure it is always up-to-date.

However, it may be nice to have a variable that we could use to set the number of leap seconds, in order to decorrelate the calculation of the UTC Splice Time, in having: 
`utcSpliceTime = (UTC Time Since 6th January 1980 without leap seconds) + (number of leap seconds)`

Then, up to us to choose to leap seconds or not, based on the encoder/transcoder implementation.

What do you think?

lelegard, 2021-04-05:
Hi @movingDigital,

> Since SCTE-35 allows implementation to ignore the leap second (link here)

The link points to a blog on "Understanding SCTE-35" and says this:

> SCTE-35 allows implementation to ignore the leap second but it results in about 16 second error.

I don't see why the guy is allowed to say that. Maybe there is some more specific paragraph in SCTE 35 but the standard defines utc_splice_time in section 9.3.2 as follow:

> utc_splice_time – A 32-bit unsigned integer quantity representing the time of the signaled splice event as the number of seconds since 00 hours UTC, January 6th, 1980, with the count of intervening leap seconds included. The utc_splice_time may be converted to UTC without the use of the GPS_UTC_offset value provided by the System Time table. 

The expression _"with the count of intervening leap seconds included"_ is pretty clear. There is no room for ignoring leap seconds.

Maybe the confusion comes from _"may be converted to UTC without the use of the GPS_UTC_offset value provided by the System Time table"_. I do not see this as a permission to ignore leap seconds. I see this as an implementation guideline for systems outside the ATSC world. SCTE 35 comes from the US, an ATSC world, but is also used in DVB and ISDB zones. Time references are managed differently.

- In DVB and ISDB, the time reference is a UTC (or UTC-related) time in MJD/BCD format in the TDT. It is my understanding that leap seconds are applied to UTC. Hence, a UTC time _includes_ leap seconds or, otherwise said, the difference in seconds between two UTC times should include leap seconds.
- In ATCS, the time reference is a GPS time in binary format in the STT. GPS time is a number of seconds since Jan 6 1980 _without_ leap seconds (because the GPS clock is autonomous and uniformely increasing). The STT also contains a _GPS_UTC_offset_ field which indicates the difference in seconds between the GPS and UTC time. This is where leap seconds are.

So, given the format and epoch value, the SCTE 35 time is clearly influenced by the ATSC time. But it emphasizes an important addition: _"with the count of intervening leap seconds included"_. Knowing that, the sentence _"may be converted to UTC without the use of the GPS_UTC_offset value provided by the System Time table"_ should be understood, in my opinion, as _"you may use some other time reference than the ATSC STT"_. This is just my interpretation but it does not seem to allow to ignore leap seconds.

>  I really wonder if all the vendors really follow the standard by the book, or not.

That is the ultimate practical issue with standards: how far are they correctly unambiguously specified, understood and implemented? Even with extremely precise standards, this is an issue. It is even worse when there is some room for interpretation like here. And anyway, the strongest one wins as usual: when a big industrial actor has a wrong implementation, you are kindly but firmly requested to adapt in a "bug-to-bug" fashion since the big guy doesn't give a **** about your demonstration that its implementation is incorrect. Real-life experience, as you may imagine...

For a test toolbox such as TSDuck, to be used in multiple environments, including environments which do not correctly support standards, I think it is wise to support all modes, with and without leap seconds.

My plan is:
- Add an XML "system time" configuration file to TSDuck (other such configuration files already exist for HF bands or LNB). In this file, we specify all known leap seconds (27 so far, not included the initial 10 from the pre-1972 era). Additional leap seconds will be added as IERS will define them.
- In the computation of SCTE 35 utc_splice_time, leap seconds will be included by default since this seems to be the standard. Some option will disable them to cope with non-conformant equipments (a usual strategy in TSDuck).
- In the XML definition of the splice information table, I will allow the dual definition of utc_splice_time: either a uint32 value which is used literally or a UTC date (YYYY-MM-DD hh:mm:ss) to be interpreted as a uint32 with or without leap seconds, depending on the option.
- We may find other interesting usages to this configuration, beyond SCTE 35 utc_splice_time.

Any other suggestion?

moving-digital, 2021-04-05:
Hi @lelegard , this is beautiful, I can't think of anything better.

I have been playing quite a lot with ESAM lately and the standard is relying on UTC Point as there is no PTS information provided (Because it is Out-of-band). The only cue_request_message type allowed is therefore splice_insert (Type 5), and will be inserted according to the desired UTC time. Which in the SCTE35 world is nothing but a splice_schedule (Type 4).

So I really like the idea of allowing the dual definition of utc_splice_time (with or without leap seconds) based on UTC date.

Thank you so much for your dedication once again!

lelegard, 2021-04-06:
The following modifications have been applied:

- The configuration file `tsduck.time.xml` contains the list of officially registered leap seconds. Should there be any need to add other types of system time customization in the future, we will put them here.
- The `utc_splice_time` attribute in `<splice_schedule>` entries in the XML definitions of `<splice_information_table>` now accepts either a 32-bit integer or a "YYY-MM-DD hh:mm:ss" string. In the former case, the binary value is set "as is". In the latter case, an SCTE 35 UTC time is computed. Since the "YYY-MM-DD hh:mm:ss" translation into a UTC time does not include the leap seconds (no operating system does this), the leap seconds are explictly added.
- When a  `<splice_information_table>` is generated (table extraction in XML format), the value of the `utc_splice_time` attribute is a "YYY-MM-DD hh:mm:ss" string. To make sure that the corresponding date is the intended UTC date, the leap seconds are explicitly substracted before conversion from system time to text.
- The option `--ignore-leap-seconds` has been added in all commands and plugins which manipulate tables. With this option, the leap seconds are no longer explicitly added or substracted.
- In the TSDuck user's configuration file (see appendix A in the user's guide), the option `leap.seconds` can be set to `false` to enable this option by default.

moving-digital, 2021-04-06:
Hi @lelegard 

Thanks heaps! Will try this all ASAP. Would it be possible to add the "T" as well as the milliseconds in the UTC Time, in order to make sure to get a SCTE-35 marker frame accurate?
e.g. "2021-04-06**T**17:46:03.**381Z**"

lelegard, 2021-04-06:
The utc_splice_time field is a number of seconds. There is no millisecond and no frame accuracy in a splice_schedule() command. Only splice_insert() commands have a PTS time stamp with 90 kHz accuracy.

There is some inconsistency in the standard about splice_schedule() commands: the utc_splice_time is a number of seconds while the break_duration() in the same command is a number of PTS units.

My understanding is that splice_schedule() are optional (never seen any in practice) and used to warn that an event may occur _around that time_ without guarantee, just to be prepared. But the actual event will be signaled on time with a splice_insert() command, using a PTS accuracy.

Using splice_schedule() as actual and unique mechanism to signal events seems impossible to me since SCTE 35 was originally designed for video splicing for which frame accuracy is crucial (unlike ad replacement in OTT content which is nothing more than program switching). A splice_schedule() command without a corresponding splice_insert() for the same event seems useless to me. Consequently, there is no frame accuracy in splice_schedule() because we just don't care.

moving-digital, 2021-04-06:
Hi @lelegard , 

My bad, I got carried over from the ESAM/POIS OpenCable Specifications allowing a UTC Point precise to the milliseconds as described [here ](https://community.cablelabs.com/wiki/plugins/servlet/cablelabs/alfresco/download?id=150db576-59a1-4cc6-bdd1-abd560997ada) and [here](https://community.cablelabs.com/wiki/plugins/servlet/cablelabs/alfresco/download?id=e7d0f6d3-d1d2-4dfb-ac01-a76b6f7b83f0).

lelegard, 2021-04-06:
>  I got carried over from the ESAM/POIS OpenCable Specifications allowing a UTC Point precise to the milliseconds as described here and here.

I just had a look and it seems that the UTC in _scheduled_ events has a seconds accuracy only in these ESAM documents. Copy/paste from these docs:
~~~
<signal:EventSchedule interval="PT5S">
  <signal:StartUTC utcPoint="2012-09-18T10:00:05Z"/>
  <signal:StopUTC utcPoint="2012-09-18T12:00:00Z"/>
</signal:EventSchedule>
~~~

lelegard, 2021-04-06:
Sorry, the thread was briefly closed by mistake (wrong button).

moving-digital, 2021-04-07:
Hi @lelegard ,

You are right again... :-D I checked all the docs from CableLabs and indeed for both spliceSchedule() and spliceInsert(), there is not notion of milliseconds.

However, I have developed an ESAM server which is inserting an UTC Point such as "2021-04-06T17:46:03**.381Z**" and the encoder/transcoder I am using seems to take it well.

closed 2021-05-09 by moving-digital