tsduck/tsduck #752: [Q] TSduck Dockerfile

themoog, 2021-04-06:
Hi

Has anyone got an example Dockerfile for building tsduck from scratch?

Thanks in advance

Neil

cbernander, 2021-04-06:
I've been using this Dockerfile to build a minimal Ubuntu docker image with tsduck from source. Note that I'm copying the tsduck repo dir into the build container rather than doing a `git clone` inside the container.

Now, it's been a while since I built it so I'm not sure if it will work, but if nothing else maybe you can use it as a starting point.

```
FROM ubuntu:20.04 AS build

ENV DEBIAN_FRONTEND noninteractive

RUN apt-get -y update && \
  apt-get -y upgrade && \
  apt-get --no-install-recommends -y install \
  g++ dpkg-dev doxygen dos2unix graphviz curl locales \
  pcscd libpcsclite-dev libcppunit-dev libcurl4 libcurl4-openssl-dev && \
  apt-get -y clean && \
  sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen && locale-gen

COPY tsduck /usr/local/src/tsduck

ENV LANG en_US.UTF-8  
ENV LANGUAGE en_US:en  
ENV LC_ALL en_US.UTF-8

RUN cd /usr/local/src/tsduck && \
  make -j8 NOPCSC=1 NODTAPI=1 && \
  make install SYSPREFIX=/usr/local

FROM ubuntu:20.04

ENV LD_LIBRARY_PATH /lib:/usr/lib:/usr/local/lib

RUN DEBIAN_FRONTEND=noninteractive \
  apt-get -y update && \
  apt-get -y upgrade && \
  apt-get --no-install-recommends -y install \
  ca-certificates libcurl4 && \
  apt-get -y clean

COPY --from=build /usr/local/bin/ /usr/local/bin/
COPY --from=build /usr/local/lib/libtsduck.so /usr/local/lib/
COPY --from=build /usr/local/lib/tsduck /usr/local/lib/
COPY --from=build /usr/local/share/tsduck /usr/local/share/tsduck
COPY --from=build /usr/local/etc/security/console.perms.d/80-tsduck.perms /usr/local/etc/security/console.perms.d/
COPY --from=build /usr/local/etc/udev/rules.d/80-tsduck.rules /usr/local/etc/udev/rules.d/
```

themoog, 2021-04-07:
Thanks for the very quick response... Just tidying up my lab environments and ill start picking through this, see if I can get SRT to build.... I will post updates :)

themoog, 2021-04-10:
Having some problems building from source with srt support

heres some snippets from the docker file

```
RUN apt install tclsh pkg-config cmake libssl-dev build-essential -y
WORKDIR / 
RUN git clone https://github.com/Haivision/srt.git
WORKDIR srt

RUN ./configure
RUN make
RUN make install

WORKDIR /

RUN git clone https://github.com/tsduck/tsduck.git
WORKDIR tsduck
RUN build/install-prerequisites.sh
RUN make -j10 NOPCSC=1 NODTAPI=1
RUN make install
```

Seems to complete without issues but get the following error when I run tsp -O srt 8001:

**tsp: error while loading shared libraries: libsrt.so.1.4: cannot open shared object file: No such file or directory**

Thanks in advance

Neil

mwolf1989, 2021-08-20:
Hi @themoog 

I added 

```
RUN git clone https://github.com/Haivision/srt.git
WORKDIR srt
RUN ./configure && make && make install
```
to build srt and then you need to copy the libs in the secound container:

```
COPY --from=build /usr/local/lib/libsrt.so.1.4.4 /usr/local/lib/
COPY --from=build /usr/local/lib/libsrt.so.1.4 /usr/local/lib/
COPY --from=build /usr/local/lib/libsrt.so /usr/local/lib/
COPY --from=build /usr/local/lib/libsrt.a /usr/local/lib/
```

I had success with this.

bugre, 2022-05-19:
If it helps someone.. recently built a container, and had to add a few extra copy for libraries..
It's not the cleanest nor probably the most correct to copy libraries  from one to the other layer, but it works for what i needed (multicast and srt transport streams).

```
FROM ubuntu:20.04 AS build

ENV DEBIAN_FRONTEND noninteractive

RUN apt-get -y update && \
  apt-get -y upgrade && \
  apt-get --no-install-recommends -y install \
  g++ dpkg-dev git doxygen tclsh pkg-config cmake libssl-dev build-essential ca-certificates dos2unix graphviz curl locales \
  pcscd libpcsclite-dev libcppunit-dev libcurl4 libcurl4-openssl-dev libedit-dev && \
  apt-get -y clean && \
  sed -i '/en_US.UTF-8/s/^# //g' /etc/locale.gen && locale-gen

ENV LANG en_US.UTF-8
ENV LANGUAGE en_US:en
ENV LC_ALL en_US.UTF-8

RUN git clone https://github.com/Haivision/srt.git
WORKDIR srt
RUN ./configure && make && make install

RUN git clone https://github.com/tsduck/tsduck.git
WORKDIR tsduck
RUN scripts/install-prerequisites.sh
RUN make -j10 NOPCSC=1 NODTAPI=1
RUN make install SYSPREFIX=/usr/local

FROM ubuntu:20.04

ENV LD_LIBRARY_PATH /lib:/usr/lib:/usr/local/lib

RUN DEBIAN_FRONTEND=noninteractive \
  apt-get -y update && \
  apt-get -y upgrade && \
  apt-get --no-install-recommends -y install \
  ca-certificates libcurl4 && \
  apt-get -y clean

COPY --from=build /usr/local/bin/ /usr/local/bin/
COPY --from=build /usr/lib/x86_64-linux-gnu/libpcsclite.so* /usr/lib/x86_64-linux-gnu/
COPY --from=build /usr/lib/x86_64-linux-gnu/libedit.so.2* /usr/lib/x86_64-linux-gnu/
COPY --from=build /usr/lib/x86_64-linux-gnu/libbsd.so.0* /usr/lib/x86_64-linux-gnu/
COPY --from=build /usr/local/lib/libsrt.so.1.4.4 /usr/local/lib/
COPY --from=build /usr/local/lib/libsrt.so.1.4 /usr/local/lib/
COPY --from=build /usr/local/lib/libsrt.so /usr/local/lib/
COPY --from=build /usr/local/lib/libsrt.a /usr/local/lib/
COPY --from=build /usr/local/lib/libtsduck.so /usr/local/lib/
COPY --from=build /usr/local/lib/tsduck /usr/local/lib/
COPY --from=build /usr/local/share/tsduck /usr/local/share/tsduck
COPY --from=build /usr/local/etc/security/console.perms.d/80-tsduck.perms /usr/local/etc/security/console.perms.d/
COPY --from=build /usr/local/etc/udev/rules.d/80-tsduck.rules /usr/local/etc/udev/rules.d/
```

joe820912boy, 2022-05-20:
Hi @bugre 

I have use the dockerfile you edited, but it occurs some problem for me.

[image]

Seems that the lib is not installed? Or did u have this problem before?

joe820912boy, 2022-05-20:
I know the problem, 

[image]

the lib in my docker lib is 1.5, so the copy command not found 1.4.4

and I change the version, now it's work for me!

bugre, 2022-05-20:
Sorry, that has to do with what version of libsrt you've building. I had specific need to build the 1.4.x version. Forgot to mention that.

> [image]
> 
> Seems that the lib is not installed? Or did u have this problem before?