tsduck/tsduck #753: [Q]  Impairments

themoog, 2021-04-07:
Hi Chaps,

Im looking for a way to 'impair' a TS stream for testing. (i.e. drop TS packets for checking CC counters / errors of a downstream device etc...

So for example if I set a pipe line of ffmpeg | tsp input - <impair> -  output

Any-thoughts on if tsduck can help me achieve this? I was looking through the filter functions, but couldn't quite work out if it would help.

Thanks in advacne

Neil

lelegard, 2021-04-08:
Hi @themoog 

The plugin `filter` selects packets, let them pass and drop all others. With option `--negate` (or `-n`), it does the opposite, delete the selected packets and let all others pass. There are a lot of packet selection options.

For instance, the following command drops one packet every 1000 packets:
~~~
tsp ... -P filter -n --every 1000 ...
~~~

With option `--stuffing` (or `-s`), the packets are not dropped but replaced by null packets. You can corrupt the stream without altering the TS bitrate.

The option `--set-label` sets a label on selected packets without dropping any. The labeled packets (and only them) can later be processed by another plugin, whichever it is, using option `--only-label`. 

For instance, the following command sets the value 5 in the continuity counter of one every 200 packets in PID 500:
~~~
tsp ... \
    -P filter --pid 500 --set-label 0 \
    -P filter --only-label 0 --every 200 --negate --reset-label 0 \
    -P craft --only-label 0 --continuity-counter 5 \
    ....
~~~

For a full explanation, RTFM as usual....

lars18th, 2021-04-10:
Hi @lelegard ,

I feel the plugin `craft` could be enhanced with some type of randomness. For example, in the FFmpeg tool exists a filter to generate "random corruptions". I feel that @themoog wants to do some tests simulating a noising channel. Then from my point of view it's sufficient with multiple instances of the `craft` plugin simulating different _random_ modifications. Perhaps you want to consider this suggestion.

Regards.