tsduck/tsduck #754: [BUG]

moving-digital, 2021-04-10:
#### Bug description:
When using the 2 plugins "ip input" and "splicemonitor" in order to get the SCTE-35 commands info displayed in the console, I am not getting any info but instead, lots of binary thrown at the console:
```
.\tsp.exe -I ip 239.5.5.5:5000 -P splicemonitor --all-commands
.\tsp.exe -I ip 239.5.5.5:5000 -P splicemonitor --display-commands
```

#### How to reproduce:
I am using StreamXpress and WinSend as streamer. The streams are MPEG-2 TS using UDP.
Use the above 2 commands to check the behavior and outcome.

#### Expected behavior:
The TSP command shall return the proper SCTE-35 info contained in the TS input, and display them in the console.

#### Errors and logs:
```
����#X�a>��v��\�$�߼ ��� �F�     �.2��� JZ� �j{p����y��  x��J $T�����m?���N�[1Rl��b�ɟ1Inx��@��o]Z��|g��F��   �0��� ��:���
sS+Lp�}C`&Nv�I ^�Jis��"����0����������Z�|g��F��   �0��� ��:���sS+Lp�}C`&Nv�I ^�Jis��"����0����������ZZG � �r�We� �oh�# ;
�؄P�8$�h�� j�Su"X��  - G�G_��5pt���!�/*�z$   S���3�� p>I�oH�;6�Kl� ݬ��lF�S�`�G � �>��Y�� &F�M�\��ο7d!��~x�   �ٙ᪁�i~ �
����(ʞ � k����=�'��� ��=!^ ����`�M J���I�'_7N G?qu�����@� �     ��� }�ۋv�1�ܺ����  3<���&ֶ[HO�;E�� H�1� eP �$,Gj� U �  >x
�$�/v �����88����G � 1L� �9��_��Bɢ�7�3�r[8�D��D�]9 ��f~� �튫Ȉ��Ҳ�uꁣ��� 3���ۙ�mR��}       2� � v�R'�ǽ`�\Oƅj� ���XC߻�e �i h
}�l-U\��� �m
��7x��_�m�6��8�>֞       X� �畉ʪ1fְ|�5\ �`��� ��PC@�L��vuC�WT�j&G � ��יn;��o�[� �`�O��o
�� =�}�� zˁ����pJ�� ���#�D������D��k͗�
� E�O� A�K � f�h1v  � �1�U�Q l2 N�� ��9`be�J� ���V� � � m S ���� ��Ɓ�Fk�
H�yB� n ~�Q�م�%����z �X� ����>�Ƶ�� +G � N� �w�w�I�� `)�d�p ( �;��� J(�xGm�=�)� O�$  0��t@���jHMf8�~o��d� `$ �ʡ!�� ş�  ��
 �-^R�� ������;���� m
```

#### Environment:
 - OS: Windows 64 bits
 - OS version: Server 2012 R2
 - TSDuck full version: 3.25-2237
 - Installation type: Official binary TSDuck-Win64-3.25-2237.exe

#### Additional information:
Am I using the commands correctly?

lelegard, 2021-04-10:
You do not specify any output plugin. The default output (resp. input) plugin is `file`. With this plugin, the default file is standard output (resp. input). The idea is that, by default, `tsp` acts as an element of pipe.

Consequently, the "garbage" you see is simply the output TS binary packets. Specify `-O drop` as output plugin to explicitly drop output packets.

RTFM, `-O` option in `tsp` reference section and also chapter 6 "troubleshooting":

> __6.2 Typical issues with tsp__
> This section lists a few typical issues when using tsp and provides some troubleshooting recipes.
>
> __The screen is garbled with strange characters__
>
> This is typically observed when no output plugin is specified. The default output plugin is file and the default output file for this plugin is the standard output. This is quite useful to pipe transport streams between commands. But if the binary transport stream is sent to the console, the screen is garbled.
>
> __Solution:__ If you don’t need the output transport stream, use -O drop as output plugin.

moving-digital, 2021-04-10:
Hi @lelegard 

Got it working like a charm, thank you so much and sorry for the inconvenience ! Got it, will always specify an Output from now onwards.

closed 2021-04-10 by moving-digital