tsduck/tsduck #756: [Q] How TSDuck handles bitrates with decimals?

lars18th, 2021-04-10:
Hi @lelegard ,

I ask for this because I don't know the answer: How TSDuck handles bitrates with decimals?

The bitrate is expressed as "bits per seconds", but using some modulations (for example with some DVB-T configurations) the bitrate is not an integer value. So when the bitrate is not a value that can be expressed with an integer... how we can handle it with TSDuck?

Please could you add some light on this point?
Thank you.

lelegard, 2021-04-10:
> So when the bitrate is not a value that can be expressed with an integer... how we can handle it with TSDuck?

You cannot, unfortunately.

The initial choice to represent bitrates was integer values in bits/second. At the time, it was more than enough for the target use cases. It could have been different, for instance in db/s or cb/s (decibits/s or centibits/s). But this is not the case.

Technically it is possible to change this. But this would require a careful inspection of all individual lines of code across the whole project (376,830 lines as of today) to apply all modifications. I have no time for this and this won't probably happen any time soon. Not by me at least.

I can design a new `BitRate` class which would carefully encapsulate bitrate representation and computations (I should have probably done this in the first place). But someone else has to perform the overall modifications in the code base.

lelegard, 2021-04-11:
There is maybe a possibility. I have experimented a `FixedPoint` class which is carefully defined to mix with integer arithmetics. If the type `ts::BitRate` is redefined as a fixed-precision, this may work. But there are several steps:
- Update `BitRate` type and make sure everything recompiles.
- Make it does not break anything.
- Explore code in case some integer types were used somewhere instead of `BitRate`

Using such a fixed-point type, the precision can be arbitrary decided for each type. How many decimal digits should be used for bitrates? 2 or 3 maybe?

lars18th, 2021-04-12:
Hi @lelegard ,

Great news!

> Using such a fixed-point type, the precision can be arbitrary decided for each type. How many decimal digits should be used for bitrates? 2 or 3 maybe?

I don't know the correct precission required. However, I feel the max level should be around 10ns. As the PCR clock is running at 27Mhz, so we need more than 37ns of precission to circunvent any drift.

Anyway, regarding the new `ts:BitRate` type with fixed-precision we need to define _where_ to use it. I feel it only requires to be at:
- Plugins that use the PCR and/or DTS/PTS timestamps.
- Input and Output plugins that use tuners or modulators.
- The `tsterinfo` and similar tools.

I suggest to first target the `tsterinfo` tool to almost calculate the bitrates with correct subbits/s units. What you think?

And thank you for working to improve this.
Best regards.

lelegard, 2021-04-12:
>  we need more than 37ns of precission to circunvent any drift.

That is the _maximum_  value in b/s, here 27,027,027 b/s for 1 bit every 37 ns. I was asking for the _minimum_ representable value in bits per second. Currently, we cannot be more precise than, for instance, 12,345 b/s. Do you need to represent 12,345.1 b/s, 12,345.01 b/s, 12,345.001 b/s, etc?

lars18th, 2021-04-12:
Yes, and for this reason a write "_more_ than"... However, if we need 0.001 b/s of precission or only 0.01 b/s, I don't know it. My concern is only to support all DVB-T/T2 modulations.

Perhaps you can start updating the `tsterinfo` tool and check the maximum precission required. You agree?

lelegard, 2021-04-12:
> Perhaps you can start updating the tsterinfo tool and check the maximum precission required. You agree?

We cannot start with one tool or plugin. They are just thin wrappers on top of the TSDuck library. The whole code is impacted. Everything is impacted whenever you change the type `ts::BitRate` from a plain integer to a fixed-point instantiation.

I will start with:
~~~
    typedef FixedPoint<int64_t, 2> BitRate;
~~~
That is to say a 64-bit integer with hundredth precision. I will check if it does not break too many things. The compilation phase may be tedious but rational. The debug / tuning phase, however, may become some black magic...

Then, adding one more decimal is just changing the 2 into a 3. The result of the test suite can be affected however.

lelegard, 2021-04-15:
> Then, adding one more decimal is just changing the 2 into a 3. The result of the test suite can be affected however.

Testing with 2 decimal digits for bitrates is not good. Even a 64-bit integer type is not sufficient to absorb intermediate overflows in some calculations. Specifically, the plugin `reduce` is affected. To smoothen the removal of packets, it works on potentially large windows of packets. This created intermediate overflows and produced incorrect results. To cope with that, I implemented overflow detection in the plugin and automatically adapted the window size down to a value which was free of overflow. Fonctionally, it works. But the results are not good. Running the automated test suite demonstrates that the standard deviation between packets of the same PID is much worse than without fixed-point representation of bitrates (precisely because of the reduced window size).

Changing the number of decimals to 1 only in bitrates, the efficiency of the plugin `reduce` returns to the same value as before.

So, adding decimals to bitrate representations may improve the precision on some points but it also reduces it on others. Using emulated arithmetic on 128-bit integers is not an option for obvious performance reasons. Using floating point types for bitrates does not seem a good option to me for accuracy reasons.

Consequently, we cannot offer more than 1 decimal digit in bitrates, to be confirmed by further testing.

lars18th, 2021-04-15:
Hi @lelegard ,

And it's not possible to do this?
- By default the bitrate uses 64bit integers with 1 decimal digit.
- If the user enables it, it uses 128bit integers with 3 (or 4) decimal digits.

Then, by default the functionality will be the same as now, and only when it's required more precision it will be enabled with some performance penalty. In fact, if the bitrate representation has two parts (64bit with 1 decimal digit; and another 64bit with the second decimal and beyond) every filter can decide if it uses the rounded version (64bit only) or the full precision version (128bit).

What you think?

lelegard, 2021-04-15:
This is not possible. 

First, the source code is written in some way using bitrate as 1) plain integer type (current version), 2) specific fixed-point class (being tested), 3) floating point type, 4) some extended 128-bit integer type or class. Moving from one implementation to another is a one-way trip which needs some work.

Second, based on a specific fixed-point class, the precision (1, 2, 3 digits, etc) is a template parameter and consequently a compile-time constant.

In both cases, you can't compile one binary and allow some command-line option to select the implementation you prefer.

The best I can propose is using a fixed-point type with the precision being 1 decimal digit by default (this seems the best balance so far) but being customizable with a compile-time macro, something like `make BITRATE_PRECISION=2` for 2 decimal digits for instance.

But you must have a full consistent build of the entire TSDuck toolkit with the same precision. Mixing binaries with different bitrate precisions is a recipe for disaster.

Additionally, changing the bitrate precision will maybe improve some use cases but degrade others. The behaviour of some features can be seriously affected (as seen on the `reduce` plugin with 1 vs. 2 decimal digits on bitrates). 

Specifically, the TSDuck test suite is calibrated for one configuration and will fail with any other. I have checked that setting 0 digits precision (but still a fixed-point class) exactly passes the test-suite that was established with the integer type. But, with any other precision, all tests must be reviewed. Using 1 or 2 digits, approximately 110 tests fail out of 1282. I have to check all failed tests one by one to determine whether this is the new normal result (for instance displaying bitrate 12,345,678 b/s instead of 12,345,679 b/s) or if the test is actually broken (for instance the `reduce` plugin with 2 digits).

lelegard, 2021-04-15:
As discussed above, I have updated the code base with the bitrate representation set to a fixed-point template class with one decimal digit. All tests were automatically checked. The only differences were a significant proportion of bitrate values which differed by 1 b/s (2 in rare cases). The difference is likely due to the increased precision in intermediate computations.

So, we are in a stable state now.

As mentioned above, an attempt to set 2 decimals did not change most results (and so the precision is not actually improved) but it broke some plugins (or reduced their efficiency) because of intermediate overflows.

For specific purposes, you may want to entirely recompile TSDuck with another bitrate precision. To use 2 decimal digits for instance:
~~~
$ make BITRATE_DECIMALS=2
~~~

But keep in mind that _everything_ must be consistently recompiled with this new value.

To reject accidental mix of binaries (executables and shared libraries) which were built with distinct values, I have used the same trick as version checking (originally inspired by the way the glibc defines its ABI levels). The main TSDuck shared library exports a symbol which contains the bitrate decimals. All executables and plugin shared libraries import the same symbol. 

See this:
~~~
$ nm libtsduck.so | grep TSDUCK_LIBRARY
0000000000faca74 R TSDUCK_LIBRARY_BITRATE_DECIMALS_1
0000000000faca78 R TSDUCK_LIBRARY_VERSION_3_26_2343
$
$ nm tsp | grep TSDUCK_LIBRARY
                 U TSDUCK_LIBRARY_BITRATE_DECIMALS_1
                 U TSDUCK_LIBRARY_VERSION_3_26_2343
~~~
 
So, if you recompile TSDuck with `make BITRATE_DECIMALS=2`, the new library will export `TSDUCK_LIBRARY_BITRATE_DECIMALS_2` instead of `_1`. If you try to use that library from an "old" executable, that won't work because the executable will try to resolve  `TSDUCK_LIBRARY_BITRATE_DECIMALS_1` which won't be there.

lars18th, 2021-04-15:
Hi @lelegard ,

Thank you for the explanation. Then it will possible to use at time without more complexities 1 decimal in the bitrate. And if we need more precision then recompile with two decimals. In this case, we can use two different binaries: stock and precisse-bitrate. Ok. However, regarding the different DVB-T/T2 modulations, what are the precisse bitrates?

lars18th, 2021-04-15:
Hi @lelegard ,

Don't worry about my previous request. The tool `tsterinfo` calculates this information. However, I'm worried about if it calculates the correct _precise_ value, even if it's compiled with 1 decimal. If it not calculates the correct value with just 1 decimal, then can you incorporate the option to print the `TSDUCK_LIBRARY_BITRATE_DECIMALS_X` bitrate value plus the PRECISE value, both at the same time? Please.

lars18th, 2021-04-15:
Hi @lelegard ,

Checking the info in this page: https://dveo.com/broadcast/bit-rate-calculation-for-COFDM-modulation.shtml
the bitrate of the DVB-T modulation never be a non-fractional number. So, how professional equipment handle it?

lars18th, 2021-04-23:
Hi @lelegard ,

Regarding the precission of the bitrate, I've rethinked about it and I want to share with you my results:
- Inside the TSDuck the only problem with the current support of bitrates with one decimal is when you want to broadcast the resulting transport stream with a modulator using a fractional bitrate.
- This is only harmful for the PCR, as it will drift over time. And nothing else will be affected if the bitrate is very close to the modulation bitrate.
- So we only need to compensate the drift of the PCR over time.

Is this assumption correct?

- If yes, then I feel it will be sufficient to enhace the current `pcradjust` plugin with a new functionality.
- This functionality can be called "compensate bitrate". And it will be based on the assumtion that every N packets a NULL packet _will be inserted_ in the TS. This insertion will be done by the modulator hardware, so it's not present inside the TSDuck's output.
- Then to handle this phantom NULL packet, it will be only necessary to adjust the computation of the PCR to adjust the drift every N packets.
- And the computation of the N packets will be based on two new optional parameters: `--compensate-target-numerator` and `--compensate-target-denominator`. These two values represents the target bitrate expressed as a fractional number, that will be the precise value of the bitrate used in the modulation. Therefore, the calculus could be done based on the difference between the real bitrate (the one used internally by the toolkit, with one decimal) and the broadcast bitrate. And the result will be a X drift over the PCR every N packets.
- So with these integer X and N values the plugin can adjust with full precission the PCR timestamps.
- And to use this functionality we only need to use the most closer lower value of the bitrate, and compensate with the target (modulation) bitrate. As the modulator hardware will insert the required NULL packet.
 
What you think about this idea?

lars18th, 2021-04-30:
Hi @lelegard ,

Please, can you commet if the described enhancement of the `pcradjust` plugin has sense or not?

lars18th, 2021-05-31:
Hi @lelegard ,

In order to maintain the synchronization over large running times (similar to the TDT/TOT drift), can you provide a method to handle the drift over time of the _fractional_ bitrates of DVB-T signals?

lars18th, 2021-07-07:
Hi @lelegard ,

Checking the last commits fixing errors with Dektec devices, I feel that _professional_ equipments use fractions to represent bitrates. It's this assumption right?

If it's true, please try to add some support for native fractional bitrates in TSDuck. I know that you're promoting the TSDuck Toolbox as a _testing_ tool. However, for homemode projects using non-professional equipment (for example with a cheap HiDes UT-100C modulator) your tool is a must have. And in such projects for a continous DVB-T transmission it's necessary that the clock base doesn't drift over time. So even in professional modulators the device can patch/adap/restamp PCR timestamps before the transmission, this is not true with cheap modulators. So in this case, as practically all DVB-T modulation modes use fractional bitrates, then some workaround it's necessary for a continous reliable streaming.

So for this particular scenario, you can provide any easy-and-simple solution?
Please, don't try to implement a complex solution. I only want to maintain a constant straming with the Hides modulator with correct PCR/PTS/DTS values after several days/weeks of run.

In any case, I reiterate: Thank you for this amazing tool!

lelegard, 2021-08-04:
> please try to add some support for native fractional bitrates in TSDuck. 

I tried and that failed...

Let me explain. I have implemented a generic template class named `Fraction` which implements the same operations as `FixedPoint` but maintains a pair of numerator and denominator all the way through. To limit the risk of overflows, the fraction is always reduced after any arithmetical operation (probably at the expense of performance). All unitary tests on `Fraction` pass and the class looks good.

The type `BitRate` is still implemented by default using a one-decimal fixed-point type but, using `make BITRATE_FRACTION=1`, you can recompile everything with `BitRate` being represented as a fraction of 64-bit integers.

When running the TSDuck test suite, there are zillions of errors. Basically, all bitrate values which are computed from many successive operations end up with incorrect values. There must be overflows at some point. Almost all the time.

So, using fractions is a good idea as long as they are limited to a few operations. But the experience seems to demonstrate that this is not applicable in the general case for bitrates. You have to choose between small inaccuracies with fixed-point values or potentially completely invalid values with fractions because of intermediate overflows.

In fact, maintaining the validity of fractions on the long run is not easy. The Dektec API (DTAPI) has recently introduced fractional bitrates (class `Dtapi::DtFractionInt`) for some specific operations, not long accumulations of computations. Even in that simple case, I found an overflow bug with a DVB-S2 rate. I reported it to Dektec and the bug was fixed (at least on those values) with the July 2021 release.

Feel free to play with `make BITRATE_FRACTION=1` and try to implement a solution, maybe some expensive intermediate approximations in the reduction of fractions in case of detected overflow in each arithmetic operation.

lelegard, 2021-08-06:
I added overflow checking in all arithmetic operations on fractions and throw an exception in case of overflow (debug mode only for obvious performance reasons). Because of the nature of fraction arithmetics, additions with different denominators rapidly generate overflows, even on 64 bits integers. Not a single `tsanalyze` operation completes without an overflow exception. So, my conclusion is that using fractional integers on bitrates is not usable in practice.

lelegard, 2021-08-11:
I also added the possibility to use double floating-point to represent bitrates. The results seems good, maybe better than fixed-point, but less accurate than integer fractions (if they did not overflow in intermediate computations).

On a coding standpoint, the three possible representations have been grouped using templates under the common superclass `AbstractNumber`. All those types consistently define their arithmetic and a few common services. Thus, switching representation is simply a few `#ifdef` around `typedef ... BitRate;`. All the rest consistently compiles.

If you want to play with various representations, you can use any of these build commands:
~~~
make -j10 BITRATE_FRACTION=1 BINDIR_SUFFIX=-fraction-clang LLVM=1
make -j10 BITRATE_FRACTION=1 BINDIR_SUFFIX=-fraction
make -j10 BITRATE_FLOAT=1 BINDIR_SUFFIX=-float-clang LLVM=1
make -j10 BITRATE_FLOAT=1 BINDIR_SUFFIX=-float
make -j10 LLVM=1
make -j10
make -j10 DEBUG=1 BITRATE_FRACTION=1 BINDIR_SUFFIX=-fraction-clang LLVM=1
make -j10 DEBUG=1 BITRATE_FRACTION=1 BINDIR_SUFFIX=-fraction
make -j10 DEBUG=1 BITRATE_FLOAT=1 BINDIR_SUFFIX=-float-clang LLVM=1
make -j10 DEBUG=1 BITRATE_FLOAT=1 BINDIR_SUFFIX=-float
make -j10 DEBUG=1 LLVM=1
make -j10 DEBUG=1
~~~

I use them on Linux in a script to compile all possible representations of bitrates, release and debug, gcc and clang. Each build ends up in a distinct directory:
~~~
$ ls -1 tsduck/bin
debug-x86_64-vmubuntu
debug-x86_64-vmubuntu-clang
debug-x86_64-vmubuntu-float
debug-x86_64-vmubuntu-float-clang
debug-x86_64-vmubuntu-fraction
debug-x86_64-vmubuntu-fraction-clang
release-x86_64-vmubuntu
release-x86_64-vmubuntu-clang
release-x86_64-vmubuntu-float
release-x86_64-vmubuntu-float-clang
release-x86_64-vmubuntu-fraction
release-x86_64-vmubuntu-fraction-clang
~~~
Feel free to experiment.

lelegard, 2021-08-19:
Hi @lars18th 

Finally, have a look at section 2.2 in the latest version of the user's guide. It documents 4 different implementations of bitrates  and how to rebuild with each of them.