tsduck/tsduck #757: [FEAT] fork plugin with support to start and stop events

lars18th, 2021-04-18:
#### Is your feature request related to a problem?
I want to execute one script (or shell command) when the execution of the TDusck (`tsp` mainly) starts. And another one when the execution stops.

#### Proposed solution:
My suggestion is to enhance the current `fork` plugin with some new paramters:
- `--no-data`: don't pass any data, just execute the 'command'.
- `--start-event`: executed when the processing starts.
- `--stop-event`: executed when the processing stops.

#### Alternatives:
The alternative is to handle these events from outside the TSDuck. However, in several scenarios it's very useful to have the option to execute some 'commands' directly from the binary, just when some internal events appears. And as the `fork` plugin already exists, then it seems to be easy to expand it with this useful functionality.

#### Additional information:
Perhaps other events can be also supported. For example when the input plugin starts to receive packets. The idea is to implement a _simple_ way to call a 'command' when the _processing_ starts/stops.

lelegard, 2021-04-21:
You seem to want to turn TSDuck into a shell, one more time. We already discussed this.

lars18th, 2021-04-21:
> You seem to want to turn TSDuck into a shell, one more time. We already discussed this.

Sorry. We've discussed this, but every time I found a new scenario where it has sense.
Anyway, I hope at some point you, or another developer, will want to implement this: "a simple event launcher plugin".

Regards.