tsduck/tsduck #759: [Q] break_duration for SpliceInsert() and SpliceSchedule(): Seconds, Milliseconds, or Frames?

moving-digital, 2021-04-20:
Hi @lelegard 

I am playing with the spliceinject plugin, and I need to set a splice duration. As per the table example, the definition is as follows:
`<break_duration auto_return="bool, required" duration="uint33, required"/>`

Is the duration defined:
- In seconds?
- In milliseconds?
- Or in number of frame?

lelegard, 2021-04-20:
Hi @moving-digital,

> I have read in the specs that the duration is a 33-bit field that indicates elapsed time in terms of ticks of the program’s 90 kHz clock.

The XML definitions of tables and descriptors stay as close as possible to the definitions of the standards, using same names and interpretations of the various fields. So, yes, the SCTE-35 break duration is in PTS units, or 90,000 units per second.

> But how do we calculate it? Since 90 kHz = 90000 tickets per second, do I just need to multiply the break duration in seconds, by 90000 ?
>
> If my break is 30s long, does it mean the duration = 30 * 90000 ?

Yes, that's the way it works. Since SCTE-35 was designed for _video splicing_, the idea was just to add the break duration raw value to the starting PTS.

moving-digital, 2021-04-21:
Thank you, Thierry!

closed 2021-04-21 by moving-digital