tsduck/tsduck #760: [BUG] SCTE-35 SpliceSchedule XML returns "Immediate" info

moving-digital, 2021-04-21:
Hi @lelegard 

#### Bug description:
When using `spliceinject` plugin in order to insert some SCTE-35 SpliceSchedule(), the output returns the below messages:
- * spliceinject: loaded file /home/watchfolder/scte35/scte35_schedule_duration.xml, 605 bytes
- * spliceinject: enqueuing SpliceSchedule, immediate
- * spliceinject: injecting SpliceSchedule, immediate, current PTS: 0x000C5A960

As you can see, "immediate" flag seems to be taken into account, but it should not, as I'm using SpliceSchedule().

#### How to reproduce:

The command is:
```
bitrate=1000000;
pcr_per_sec=5;
pcr_distance=bitrate/(pcr_per_sec*188*8);
pcr_pid=102;

tsp --verbose --bitrate " + bitrate + " --max-flushed-packets 7 " +
"-I null " +
"-P regulate --packet-burst 7 " +
"-P filter --every " + pcr_distance + " --set-label 1 " + 
"-P craft --only-label 1 --pid " + pcr_pid + " --no-payload --pcr 0 " + 
"-P continuity --pid " + pcr_pid + " --fix " + 
"-P pcradjust --pid " + pcr_pid + " " +
"-P inject /home/watchfolder/scte35/pat.xml --pid 0 --bitrate 15000 " + 
"-P inject /home/watchfolder/scte35/pmt.xml --pid 100 --bitrate 15000 " + 
"-P spliceinject --files /home/watchfolder/scte35/scte35*.xml --service 101 --delete-files " + 
"-O ip 225.5.5.5:12345 --enforce-burst"
```

The PAT is:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
	<PAT version="0" current="true" transport_stream_id="0x0000">
		<metadata PID="0"/>
		<service service_id="101" program_map_PID="100"/>
	</PAT>
</tsduck>
```

The PMT is:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <PMT version="0" current="true" service_id="101" PCR_PID="102">
    <metadata PID="100"/>
    <registration_descriptor format_identifier="0x43554549"/>
    <component elementary_PID="102" stream_type="0x86">
      <registration_descriptor format_identifier="0x43554549"/>
    </component>
  </PMT>
</tsduck>
```

The SCTE35 XML is:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
	<splice_information_table protocol_version="0" pts_adjustment="0" tier="0x0FFF">
	<metadata PID="100"/>
	<splice_schedule>
		<splice_event
            splice_event_id="0x00000457"
            splice_event_cancel="false"
            out_of_network="true"
            utc_splice_time="2021-04-21 06:30:00"
            unique_program_id="0x0000"
            avail_num="0"
            avails_expected="0">
			<break_duration auto_return="true" duration="2700000"/>		
		</splice_event>
	</splice_schedule>
	</splice_information_table>
</tsduck>
```

#### Expected behavior:
"immediate" should not be used.

#### Environment:
 - OS: Linux
 - OS version: Raspbian Buster
 - TSDuck full version: 3.26-2349
 - Installation type: Official binary tsduck_3.26-2349.raspbian10_armhf

#### Additional information:

lelegard, 2021-04-21:
Hi @moving-digital,

This is not a bug, this is the expected behaviour.

From the reference section of the `spliceinject` plugin in the TSDuck user's guide:

> __Injection principles__
> The whole point about splice information is synchronization with video. There are roughly two classes of splice events:
> - Non-immediate splice_insert() and time_signal() commands. These commands contain a specific PTS value for the event. This PTS refers to a time stamp in the video and audio PID’s of the service.
> - Everything else.
>
> Any splice command in the "everything else" category is injected as soon as possible after reception.
>
> A non-immediate splice_insert() or time_signal() command is injected a specific number of times (2 by default) within a short period of time (2 seconds by default) preceding the specified PTS timestamp in the video stream.

In your case, "immediate" does not refer to the `splice_immediate_flag` in a `splice_insert()` command. It just means that the command will be immediately injected.

There is maybe a misunderstanding on the word "schedule". A SCTE 35 splice command is something that is sent to an equipment (a _video splicer_ in the original spec). Consequently, a `splice_schedule()` command is an instruction for the video splicer to schedule an action at some future time. But the command itself shall be sent immediately. The "scheduling" operation shall take place in the receiver (the video splicer), not the sender (TSDuck).

More generally, using `splice_schedule()` commands is questionable. It means that you trust the receiving equipment to safely record a command long before its execution. This may be fine in the original sense of a professional intermediate video splicer (typically a satellite-to-terrestrial retransmitter). But is it really appropriate for a domestic receiver, TV set or set-top box? This kind of equipment is switching channels all the time (well, the user does) and is powered on and off. When you send a `splice_schedule()` command a couple of hours ahead for a given TV channel, the receiving equipment will likely no longer watch the same channel at the event time. Conversely, when watching a TV channel at the expected time of a splice event, any `splice_schedule()` command that was sent for this channel a couple of hours ago was probably never received by the TV set.

In such a situation, a "scheduled event" is a back-end concept only. If your back-end system receives a scheduled ad replacement in the future for a given channel, it should probably record that event in a database and send a `splice_insert()` command a few minutes before the event. This event will be kept in memory by the `spliceinject` plugin and sent the specified number of times at the specified time interval just before the event.

moving-digital, 2021-04-22:
I see, my bad , thank you, Thierry!

closed 2021-04-22 by moving-digital