tsduck/tsduck #761: [Q] No multicast output

Ulfila80, 2021-04-21:
Hello @lelegard!

Can not understand what I'm doing wrong.

I've tried a simple command:

```
tsp -v --bitrate 1300000 -I ip -l 192.168.100.20 239.230.2.4:3001 -O ip -l 192.168.100.20 239.230.2.44:3044      
* ip: joining multicast group 239.230.2.4:3001 from local address 192.168.100.20
* ip: initial input bitrate is 1,300,000 b/s
```
I can play new stream 239.230.2.44:3044 via VLC on PC where tsp works, I can see packet with tcpdump on eth0

```
acs@osss-control:~$ sudo tcpdump -i eth0 dst port 3044
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
16:09:04.103429 IP 192.168.100.20.35379 > 239.230.2.44.3044: UDP, length 1316
16:09:04.109932 IP 192.168.100.20.35379 > 239.230.2.44.3044: UDP, length 1316
16:09:04.123046 IP 192.168.100.20.35379 > 239.230.2.44.3044: UDP, length 1316
```
But there is no multicast on my local network. Dektec analyzer doesn't see any stream on udp://239.230.2.44:3044 in another PC.

I thought there was something wrong with my multicast setup, but I tried ffmpeg

`ffmpeg -re -i udp://239.230.2.4:3001 -c copy -f mpegts 'udp://239.230.2.44:3044??pkt_size=1316&localaddr=192.168.100.20'`

and multicast appeared on my network. The Dektec analyzer recognizes it.

Please, help to find out why multicast stream from tsp dosn't appear in local network.

**Environment:**

- OS: Linux
- OS version: Debian 10.3
- TSDuck full version: 3.26-2349
- Installation type: Official binary tsduck_3.26-2349.debian10_amd64.deb

lelegard, 2021-04-21:
Hi @Ulfila80,

It should work. It does work on my side. I reproduced your command, just changing the local addresses to match the system. And it works on a Debian 10.9.

Configuration:
- Ubuntu 20.10 VM : send multicast to 239.230.2.4:3001 using tsp
- Debian 10.9 VM : tsp receives multicast  from 239.230.2.4:3001 and sends multicast to 239.230.2.44:3044, using the same local interface for receive and send (it seems to be what you are doing).
- Ubuntu system : another tsp command receives multicast  from 239.230.2.44:3044 (and it works).

This validate the round-trip of multicast packets using tsp.

There must be some specificity on your system. Any firewall, selinux or apparmor restriction on tsp sending multicast?

lars18th, 2021-04-21:
Hi @Ulfila80 ,

Please can you try this?
```
ffmpeg -re -i udp://239.230.2.4:3001 -c copy -f mpegts pipe:1 | tsp -v -O ip -l 192.168.100.20 239.230.2.44:3044 
```

Ulfila80, 2021-04-21:
@lelegard 

There are no firewall, selinux or apparmor on my PC. It very strange but multicast from ffmpeg is working.

@lars18th 

I tried your command. Same reaction. I can see multicast packets with tcpdump and play stream with VLC on local PC (ip:192.168.100.20) but can not see them in local network.

Ulfila80, 2021-04-21:
I found what was the problem.
In routes of 192.168.100.20 PС I have route for multicast streams:

`224.0.0.0    192.168.100.254    240.0.0.0     UG    0      0     0  eth0`

I  deleted it and multicast from tsp appeared in local network. 192.168.100.254 is my Cisco router that all of my IRDs are connected to.

lelegard, 2021-04-22:
> Can somebody explain how this route affects tsp?

Just a guess: This route instructs to send all your multicast packets (because you are in that range) to 192.168.100.254. My guess is that the IP packets exit your interface with the multicast IP address 239.230.2.44 as destination but the Ethernet frames have the MAC address of the gateway 192.168.100.254 as destination instead of the multicast MAC address corresponding to 239.230.2.44. Thus, other systems on the same subnet listening for multicast traffic won't see those Ethernet frames.

I assume that you can easily verify this using Wireshark.

Now, it does not explain why the same multicast packets are seen when sent by ffmpeg or VLC. Maybe those applications do something more than what TSDuck does on multicast output.

When the `ip` output plugin creates a socket for multicast output and a local interface is specified, it sets the socket option `IP_MULTICAST_IF` to say "all multicast output traffic from this socket should go through this interface". There is nothing special in addition to this.

If you have time, it should be interesting to restore that route and inspect and compare the traffic which is generated by tsp and ffmpeg using Wireshark. Especially have a look at the destination MAC address and IP address. If there is a difference, we need to investigate why.

lelegard, 2021-04-22:
I have another (and probably better) idea. If you have the opportunity to run Wireshark on ffmpeg and tsp output with your route set, also have a look at the TTL field (time to live).

With the route set, the UDP datagrams are probably delivered to the gateway in a unicast Ethernet frame (the IP packet is sent to the router). Then, if your gateway is configured to deliver multicast, the packet is forwarded in a multicast Ethernet frame. This is the real multicast packet that you see on the network. It is sent _from the gateway_ and not from your origin server. More precisely, the multicast IP packet is sent by the origin server as a layer-3 packet but received in a layer-2 frame coming from the gateway.

This is where the default options of tsp and ffmpeg may differ. The default TTL of the tsp `ip` plugin is the one of the system, usually 1 for multicast. Having a multicast TTL to N means that the packet is dropped by the Nth gateway, in your case the first one since the default multicast TTL is 1. This explains why you never see the multicast Ethernet frame. If ffmpeg uses a different TTL value, then it can cross that number of gateways. With the tsp `ip` plugin, you have to specify the TTL value if you don't want the system default.  As a test, restore the route and use `-O ip --ttl 2` (or some higher value).

Let us know if the TTL value was the cause of the problem.

lars18th, 2021-04-23:
Hi,

See this line: https://github.com/FFmpeg/FFmpeg/blob/f7c3484b2659063043100e8194d5790d2aa1a73c/libavformat/udp.c#L137
FFMpeg default TTL for UDP output is 16.

So @Ulfila80 , please make the test with `-O ip --ttl 16` with TSDuck. And try also to repeat the FFMpeg test with:
```
ffmpeg -re -i udp://239.230.2.4:3001 -c copy -f mpegts 'udp://239.230.2.44:3044??pkt_size=1316&localaddr=192.168.100.20&ttl=1'
```

Ulfila80, 2021-04-23:
Hello everyone!

I tried different settings for the multicast route.

Short report:

`route 224.0.0.0/4 gw 192.168.100.254 dev eth0`

tsp with ttl 1,2,3 and any other doesn't work.
ffmpeg with ttl 1 and any other works.

`route 224.0.0.0/4 gw 0.0.0.0 dev eth0`

tsp with ttl 1 works.
ffmpeg with ttl 1 works.

`no route for multicast`

tsp with ttl 1 works.
ffmpeg with ttl 1 works.

I've made stream captures with WireShark for all cases. I'm not deep involved in packet analyzing but I found one difference that seems to me is the key.

Ethernet field of MPEG_TS packet by tsp ttl=1 (route with gw 192.168.100.254)

> Ethernet II, Src: HonHaiPr_72:cf:da (d0:27:88:72:cf:da), Dst: **Cisco_5a:cc:40 (58:97:bd:5a:cc:40)**
>     Destination: **Cisco_5a:cc:40 (58:97:bd:5a:cc:40)**
>         Address: **Cisco_5a:cc:40 (58:97:bd:5a:cc:40)**
>         .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
>         **.... ...0 .... .... .... .... = IG bit: Individual address (unicast)**
>     Source: HonHaiPr_72:cf:da (d0:27:88:72:cf:da)
>         Address: HonHaiPr_72:cf:da (d0:27:88:72:cf:da)
>         .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
>         .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
>     Type: IPv4 (0x0800)

Ethernet field of MPEG_TS packet by fmpeg ttl=1 (route with gw 192.168.100.254)

> Ethernet II, Src: HonHaiPr_72:cf:da (d0:27:88:72:cf:da), Dst: **IPv4mcast_66:02:2c (01:00:5e:66:02:2c)**
>     Destination: **IPv4mcast_66:02:2c (01:00:5e:66:02:2c)**
>         Address: **IPv4mcast_66:02:2c (01:00:5e:66:02:2c)**
>         .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
>         **.... ...1 .... .... .... .... = IG bit: Group address (multicast/broadcast)**
>     Source: HonHaiPr_72:cf:da (d0:27:88:72:cf:da)
>         Address: HonHaiPr_72:cf:da (d0:27:88:72:cf:da)
>         .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
>         .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
>     Type: IPv4 (0x0800)

When route gw set to 0.0.0.0 Ethernet fields of tsp and ffmpeg are identical:

> Ethernet II, Src: HonHaiPr_72:cf:da (d0:27:88:72:cf:da), Dst: IPv4mcast_66:02:2c (01:00:5e:66:02:2c)
>     Destination: IPv4mcast_66:02:2c (01:00:5e:66:02:2c)
>         Address: IPv4mcast_66:02:2c (01:00:5e:66:02:2c)
>         .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
>         .... ...1 .... .... .... .... = IG bit: Group address (multicast/broadcast)
>     Source: HonHaiPr_72:cf:da (d0:27:88:72:cf:da)
>         Address: HonHaiPr_72:cf:da (d0:27:88:72:cf:da)
>         .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
>         .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
>     Type: IPv4 (0x0800)

I attached captured with Wireshark streams. 4 files: tsp/ffmpeg with gw 192.168.100.254 and tsp/ffmpeg with gw 0.0.0.0.
[WireShark.zip](https://github.com/tsduck/tsduck/files/6367283/WireShark.zip)

lelegard, 2021-04-23:
Your post exactly illustrates what I suspected: In the presence of the route, the packet is sent in an Ethernet frame with the unicast MAC address of the router. Without the route, the packet is immediately sent in an Ethernet frame with the corresponding multicast MAC address.

However, I was wrong with the second part: Even with the route, ffmpeg sends the packet in an Ethernet frame with the multicast MAC address. There is probably something that ffmpeg does to bypass the route. But I do not know what.

Then, it seems that the router is not configured for multicast since the packet is not relayed as a multicast packet after the router. This is consistent with the fact that the TTL does not change the behaviour. The packet from ffmpeg is received not because of TTL being 16 but because the packet is directly sent as multicast, bypassing the route.

lelegard, 2021-04-25:
@Ulfila80,

I have reproduced this difference of behaviour between tsp and ffmpeg using two Linux VM using the private virtual LAN inside VMware. On the first VM, I declared the same type of route, used the same tsp and ffmpeg commands. On the second VM, I run Wireshark. I can see exactly the same behaviour: the multicast IP packet is sent in a unicast Ethernet frame to the router with tsp and in a multicast Ethernet frame with ffmpeg. To characterise the exact differences between the two in terms of system calls, I ran the two commands using `strace`. The only difference is the setting of the socket option `IP_MULTICAST_IF` with tsp.

As a test, I temporarily disabled this option in the code and tsp immediately switched to the same behaviour as ffmpeg. 

At this point, this becomes complicated. This option is used for a reason. It was not used in the beginning. But some users reported routing errors on multi-homed systems with no multicast route and a default route to another interface. In that case, the multicast traffic went the wrong way, using the other interface. I cannot remember, however, if this problem was general or specific to Linux or Windows. So, to cope with the potential lack of non-ambiguous routing for multicast, I added the socket option `IP_MULTICAST_IF`. When `--local-address` is used, it was then guaranteed that the multicast traffic was sent to the intended interface.

So, the socket option `IP_MULTICAST_IF` is necessary in some cases. We cannot simply drop it.

In such situations, the best solution is to add a command line option. I added option `-f` or `--force-local-multicast-outgoing` to force the socket option `IP_MULTICAST_IF` and it is no longer set by default.

The situation is now the following:
- On systems with only one interface, there is no ambiguity, it works by default.
- On systems with proper multicast routes defined at system level, it works by default (same behaviour as ffmpeg, the multicast IP packets go out in multicast Ethernet frames).
- On systems without multicast routes but the default route is the same as the intended route, it works by default.
- On _some_ systems (Linux? Windows? all?, I can't say) with no proper multicast route and a default route on another interface, you may be obliged to specify `-f` or `--force-local-multicast-outgoing`.

I can't really decide if the correct behaviour of the system, in the presence of a multicast route, should be to send a unicast Ethernet frame to the router or a multicast Ethernet frame. However, I am convinced that switching from multicast to unicast Ethernet MAC addresses just because of the socket option `IP_MULTICAST_IF` is a Linux bug.

closed 2021-04-29 by Ulfila80