tsduck/tsduck #763: TSECMG as a production ecmg with a custom cipher implementation

nikpakar, 2021-04-22:
Hello everyone,

I have asked a broader question before and got a very constructive response from developers. Since then i have studied the implementation and the code more and narrowed down my scope as below.

to implement a custom cipher in tsecmg to encrypt the ecm payload which im able to implement the decryption in the STB side to get the CW extracted.

My question is, if the clear ECM's security is the concern, with such cipher implementation, would the tsecmg be able to act as a production ecmg in a commercial mux/symulcrypt setup with around 60 channels.

Thanks a lot for all the responses in advance.

-nik

lelegard, 2021-04-22:
TSDuck has a very liberal BSD license. So, you may reuse all or part of the code in many ways, according to the BSD License. If you want to implement a production ECMG based on the code of `tsecmg`, you are allowed to, as long as you are ready to take responsibility for what happens in production.

That being said, an ECMG alone is useless. It shall be part of a broader Conditional Access System (CAS). All experienced CAS engineers will tell you that the ciphering of ECM's is important but not the main issue. The main issue is access rights distribution, service keys distribution and secure usage in the device (smart card or secure chipset). So, it involves developing back-end systems such as subscribers authorization systems and EMM generators. A CAS is usually a complex system and a matter of professionals.

Also, as a last word, all cryptologists will tell you that the concept of "custom cipher" is the worst thing to do. You may start with https://www.schneier.com.

nikpakar, 2021-04-22:
Many thanks for your detailed explanation. My use case does not require much of those full CAS features and it only requires common encryption for all STBs as its a Free-to-View service. We dont want to control who watch what. all we need is to restrict all watch all but only using the specific STB. So all we want is to encrypt using rolling keys to avoid bruce force and use a secure mechanism to transport the keys over ECM to STBs. All STBs will watch all content.

Once again many thanks for your very helpful response as always. Highly appreciate it.

nikpakar, 2021-04-26:
Hello Thierry, a narrowed down question on the same scope.

Can the tsecmg handle asynchronous work i.e. even though it connects to a single client (Mux), there are asynchronous ECM requests coming from 50+ ecm channels. or does the implementation simply works on one synchronous ECM.

Sorry for repeat questions and highly appreciate your kind response.

-Nik

lelegard, 2021-04-26:
`tsecmg` has been developed to test the integration of a real SCS with a CAS. This is a test tool to exercise many possible configurations of a real SCS. It has not been designed for production.

I have developed a production ECMG many years ago. Its design was completely different. It was fully asynchronous. It managed a potentially unlimited number of clients. It managed smart-cards / HSM for protection of backend crypto. It managed asynchronous reception and handling of status messages. Its multi-thread architecture was based on a single-thread reactor pattern for all client processing and one thread per smardcard / HSM. It had internal monitoring and recovery for resiliency. So, in a few words, it was completely different from `tsecmg`.

To be clear, if you want a cheap ECMG for small unsafe production, you may use a cheap development based on `tsecmg` design. If you want multi-client robust and fully DVB-SimulCrypt compliant ECMG (asynchronous handling of status messages), you need to redesign a new one from scratch. Nothing is free in this world....

closed 2021-04-27 by nikpakar