tsduck/tsduck #764: [Q] Splicemonitor plug-in not detecting SCTE-35 signaling

cc17926, 2021-04-22:
Hi!
I seem to be having a problem where tsduck, while either processing a multicast transport stream, or recorded transport stream (with SCTE-35 signaling) is not detecting splicing signals. I am still new to understanding the full depth of how SCTE-35 works, however it seems the plug-in is not recognizing these commands are coming in. 

I have ran tsduck's tsanalyzer on a recorded stream, and it does identify the presence of these commands.

For troubleshooting, I have ran tsduck to filter the output of the SCTE-35 pid when these signals are present, and hope someone with a better set of eyes can see if I am missing something, or have identified a bug.

Thank you!

[only_scte35.txt](https://github.com/tsduck/tsduck/files/6359554/only_scte35.txt)
[only_scte35.zip](https://github.com/tsduck/tsduck/files/6359558/only_scte35.zip)

lelegard, 2021-04-22:
The provided sample only contains the SCTE-35 PID. To understand your issue, we would need a piece of the complete transport stream and the precise tsp command. The `splicemonitor` plugin analyzes splice commands _in the context of their service_, typically in relation with the PTS of the video and audio PID's. 

With only the splice command PID, we can extract or display the SCTE-35 tables (plugin `tables` or command `tstables`) but we cannot _monitor_ their activity without PMT, video and audio context.

And by the way, unless you work on an ATSC stream, PID 19 seems invalid to carry SCTE-35 commands. It is in the reserved range for DVB and ISDB.

cc17926, 2021-04-22:
> The provided sample only contains the SCTE-35 PID. To understand your issue, we would need a piece of the complete transport stream and the precise tsp command. The `splicemonitor` plugin analyzes [...]

Yes, I can see where you're coming from. Unfortunately, the video content I cannot post in the public domain, however I have (temporarily) enabled my email to be shown in my profile so I can exchange this with you for further analysis.

As far as command, the tsp command is as follows:
`tsp -I file test.ts -P splicemonitor -a -O drop`

Thank you!

lelegard, 2021-04-22:
Could you at least display the PMT of the service?

You can send a link to a full TS at the TSDuck email contact (https://tsduck.io -> Web resources -> Contact).

cc17926, 2021-04-22:
As requested, here is the information regarding the service.
`===============================================================================
|  TRANSPORT STREAM ANALYSIS REPORT                                           |
|=============================================================================|
|  Transport Stream Id: ....... 4520 (0x11A8)  |  Services: .............. 1  |
|  Bytes: ........................ 90,841,600  |  PID's: Total: .......... 9  |
|  TS packets: ...................... 483,200  |         Clear: .......... 9  |
|     With invalid sync: .................. 0  |         Scrambled: ...... 0  |
|     With transport error: ............... 0  |         With PCR's: ..... 1  |
|     Suspect and ignored: ................ 0  |         Unreferenced: ... 0  |
|-----------------------------------------------------------------------------|
|  Transport stream bitrate, based on ....... 188 bytes/pkt    204 bytes/pkt  |
|  User-specified: ................................... None             None  |
|  Estimated based on PCR's: ................ 4,036,265 b/s    4,379,777 b/s  |
|-----------------------------------------------------------------------------|
|  Broadcast time: ................................... 180 sec (3 min 0 sec)  |
|-----------------------------------------------------------------------------|
|  Srv Id  Service Name                              Access          Bitrate  |
|  0x0001  (unknown) .................................... C    3,998,717 b/s  |
|                                                                             |
|  Note 1: C=Clear, S=Scrambled                                               |
|  Note 2: Unless specified otherwise, bitrates are based on 188 bytes/pkt    |
===============================================================================

===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 4,495, PID's: 2 (clear: 2, scrambled: 0)                       |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C       37,547 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C        7,417 b/s  |
|  0x0000  PAT .......................................... C        7,417 b/s  |
|  0x1FFF  Stuffing ..................................... C       30,129 b/s  |
|=============================================================================|
|  Service: 0x0001 (1), TS: 0x11A8 (4520), Original Netw: 0x0000 (0)          |
|  Service name: (unknown), provider: (unknown)                               |
|  Service type: 0x00 (Undefined)                                             |
|  TS packets: 478,705, PID's: 9 (clear: 9, scrambled: 0)                     |
|  PMT PID: 0x0010 (16), PCR PID: 0x0011 (17)                                 |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Undefined .................................... C    3,998,717 b/s  |
|  0x0010  PMT .......................................... C       14,851 b/s  |
|  0x0011  AVC video (1280x720, high profile, level 3.2,  C    3,471,948 b/s  |
|  0x0013  SCTE 35 Splice Info .......................... C           33 b/s  |
|  0x0014  ATSC AC-3 Audio (eng, AC-3, 3/2 (L,C,R,SL,SR)  C      328,798 b/s  |
|  0x0015  ATSC AC-3 Audio (spa, AC-3, mono, @48,000 Hz,  C      134,686 b/s  |
|  0x0018  unknown (0xC0) ............................... C       47,446 b/s  |
|  0x0019  Scientific Atlanta ECM ....................... C            0 b/s  |
|  0x001A  unknown (0xC0) ............................... C          952 b/s  |
|  0x001E  Scientific Atlanta ECM ....................... C            0 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
===============================================================================
`

And I have sent the video example via the contact instructions.

lelegard, 2021-04-22:
OK, I found the reason. This is a bug in TSDuck. There is one thing in your SCTE 35 sections which is different from all other streams which were processed so far by TSDuck. This small silly difference is hidden in the field _splice_command_length_ and is not correctly handled by TSDuck.

In the document ANSI/SCTE 35 2017, section 9.2.1. _Semantic definition of fields in splice_info_section()_:
> __splice_command_length__ – a 12 bit length of the splice command. The length shall represent the number of bytes following the _splice_command_type_ up to but not including the _descriptor_loop_length_. Devices that are compliant with this version of the standard shall populate this field with the actual length. The value of 0xFFF provides backwards compatibility and shall be ignored by downstream equipment.

TSDuck does not correctly handle the last sentence: _"value of 0xFFF provides backwards compatibility and shall be ignored by downstream equipment"_. And all your SCTE 35 sections have 0xFFF in this field while all other streams I have seen so far have the actual length of the command.

Arguably, _"devices that are compliant with this version of the standard shall populate this field with the actual length"_ means that your SCTE 35 generator uses an outdated version of the standard. But this is not a good reason for TSDuck to not handle it correctly. I will fix this asap.

In the plugin `splicemonitor`, all splice info sections are fully analyzed. Incorrect sections (or supposed to be incorrect by TSDuck) are ignored. This is the reason why the plugin does not display anything. No SCTE 35 section is seen because they are all considered incorrect because of the 0xFFF in the command length.

Note that the command `tstables` still displays most of the section. This command (and the plugin `tables`) displays what it can display and stops at end of data. It should probably display that a part of the section is missing.

I will let you when this is fixed. Probably by tomorrow. What kind of system do you use? Windows, Linux, macOS? Do you use pre-built binaries or do you rebuild yourself?

cc17926, 2021-04-23:
> OK, I found the reason. This is a bug in TSDuck. There is one thing in your SCTE 35 sections which is different from all other streams which were processed so far by TSDuck. This small silly [...]

I appreciate your swift and thorough analysis of this, and addressing this bug! I am using this on standard release of the Raspberry Pi build. I look forward to testing this out for you!

lelegard, 2021-04-23:
This is fixed in commit 24a967dd3c8a1befe08e6af49fe41a1f548748a7. I have checked that the analysis of your SCTE sections are correct and that the plugin `splicemonitor` properly reports the sections. I have added the provided binary section in the test suite to avoid further regression.

You may pull the repo and rebuild. I have started a build on my Raspberry Pi but it takes ages to build on this small device. I will publish it in the prerelease area when ready.

cc17926, 2021-04-23:
> This is fixed in commit [24a967d](https://github.com/tsduck/tsduck/commit/24a967dd3c8a1befe08e6af49fe41a1f548748a7). I have checked that the analysis of your SCTE sections are correct and that the plugin `splicemonitor` properly reports the sections. I have added the provided binary section in the test suite to avoid further regression.
> 
> You may pull the repo and rebuild. I have started a build on my Raspberry Pi but it takes ages to build on this small device. I will publish it in the prerelease area when ready.

That's great to hear! I will await your build to be available to run this. Thank you again!

lelegard, 2021-04-23:
The build is completed. Try will find packages for Raspberry here: https://tsduck.io/download/prerelease/

Please let us know if this works for you.

cc17926, 2021-04-24:
Yes! This works great now! Thank you very much for taking the time to address this bug and swiftly putting out a fix for it.

closed 2021-04-24 by lelegard