tsduck/tsduck #766: [Q] Is it possible to get the PTS of a service containing data PID only?

moving-digital, 2021-04-23:
Hi @lelegard 

I have created a service containing a single data PID only (SCTE-35), and I would like to retrieve its PTS, which I don't succeed in doing.

When using `pcrextract`, there is no PTS information available. Only the PCR info are avail.

However, when using the `spliceinject` plugin, the current PTS is returned in the console, meaning the PTS is available somehow:
* spliceinject: enqueuing SpliceInsert out @0xFFFFFFFFFFFFFFFF, immediate
* spliceinject: injecting SpliceInsert out @0xFFFFFFFFFFFFFFFF, immediate, **current PTS: 0x01A629320**

My question would be: where is this PTS, and how can I access it from TSP?

Thanks

lelegard, 2021-04-23:
Hi @moving-digital,

You are looking for something that does not exist. PTS means "presentation time stamp" and is fully linked to a video codec (and also audio codec by extension for the sake of synchronisation). A PTS (or DTS) is present in the PES header. PES is used to transport audio, video and some types of data. The format of an elementary stream is either PES or section. Only PES PID's can carry PTS and DTS. A PID carrying sections (SCTE 35 or any other) do not carry PES and consequently can't have PTS.

A PCR, on the other hand, is located in the TS adaptation field (an extension of the TS header). This is one level below PES vs section. A PCR can potentially be present in any TS packet, although it is usually present in audio and video PID's only, except when you create an "artificial" PCR PID for the sole purpose of having a clock reference on the service.

The message of the `spliceinject` plugin is a bit misleading. SCTE 35 messages use the PTS as clock reference because SCTE 35 is supposed to drive _video splicing_ and the PTS is the video presentation time stamp. So, every timing information is related to the PTS reference. The plugin `spliceinject` tries to synchronise on the actual PTS of the service. But, in the absence of PTS, it uses the PCR as a fallback, knowing that PTS = PCR/300 + epsilon. In the complete absence of PTS, the epsilon can be neglected because there is no video (there can't be any video without PTS).

So, what you see from the `spliceinject` plugin is a synthetic PTS value which was derived from a PCR, not an actual PTS value which was read from the stream. The [code here](https://github.com/tsduck/tsduck/blob/ebd3e5d5401ea486b119f5c60e17f67295a1e3d1/src/tsplugins/tsplugin_spliceinject.cpp#L612) is pretty explicit.

moving-digital, 2021-04-24:
Hi @lelegard 

Got it! In this case, it seems there is still a way for me to use `spliceInsert "Normal"` instead of `spliceInsert "Immediate"`.

The idea behind it that since the accuracy of `spliceInsert "Immediate"` is really bad (Have done around 20 measurements once the encoder inserts the splice. The splice point is between 5 to 30 frames after the reception of the splice request by the encoder... More than 1sec accuracy is a big deal).

What I am thinking about is to use `spliceInsert "Normal"` in doing the following:

1. Right before sending the splice request, I need to get the synthetic PTS value derived from the PCR (PTS = PCR / 300)
2. I add a bit of delay. To play safe, I will start with 4sec which should be ok to get the encoder "ready" to insert the splice on-time, and with the same accuracy of `time of splice point = time of reception of splice request + 4 sec`.

The question would be: what is the best TSP plugin to get the real-time PCR value? I've played with `pcrextract` which is great, but returns a list. Is there any other plugin giving a similar result, but providing only one single PCR value at the time the command is launched?

lelegard, 2021-04-24:
Something isn't clear: you introduced the topic mentioning a service with a data PID only and now you mention an encoder ready for splicing. So, there is some video. And there must be PTS in that video.

closed 2021-05-09 by moving-digital