tsduck/tsduck #767: [Q] Continue reading from PIPE when out-of-sync

lars18th, 2021-04-23:
Hi @lelegard ,

I want to use `tsp` with PIPE input for STSDIN. But the process that sends the TS data, sometimes will insert some corrupted data. So, what I want is to overcome this discarding garbage until the next SYNC data (0x47). Then my questions are:
- Wich parameter I need to continue reading if the input sync is out?
- And wich parameter I need to use if the _first_ data from the pipe is not a TS packet?

Now, the processing stops if the first data is not a valid TS packet. So please help me.
Thank you.

lars18th, 2021-04-26:
Hi,

I list here different reasons to support the "safe running in case of spurious data" optional mode in `tsp`:
- When an external producer process generates some data that it isn't part of the TS stream, but it writes in the buffer/pipe. In this case, we can discard the data and continue.
- When the input data is not in TS format data. In this case in the debug print we can see the data to identify the problem.
- When multiple sources write to the same pipe, and we want to process the mix discarding non TS data.
- When the input source exits and restarts connecting to the same pipe. In this case, we want to continue running discarding the closing of the pipe, or and small chunk of data not related with the TS stream.

I see almost some scenarios where this have sense. So please, consider to implement it. I can't see any side effect. You agree @lelegard ?
Regards.

lars18th, 2021-04-26:
Hi @lelegard ,

You think it's sufficient with `any-process | tsresync | tsp ...` support?
From my point of view, the `teresync` functionality could be good to be introduced inside the `tsp` INPUTS as a generic support.
But, perhaps I'm wrong. So I appreciate if you can comment what's the best way to overcome non-TS-packets in the flow.

Thank you for your patience.

lars18th, 2021-04-26:
Sorry,

My suggestion is more or less to add one option to convert this block:
https://github.com/tsduck/tsduck/blob/1f5c70798940033c666b4feeed4b9dcc6f104c17/src/libtsduck/plugin/private/tstspInputExecutor.cpp#L282
Into an optional warning. And instead of `// Ignore subsequent packets` continue reading forever.
Then using the `tsresync` tool or something similar in the tsp input we can handle the case when non-TS-data is present in the flow and we want to continue reading.

I feel this is KISS compatible and not very complex.
Regards.

lelegard, 2021-04-26:
It is not as simple as it seems.

"Continuing reading" is not only reading, it is analyzing the structure of input. Normally, an input plugin is responsible for providing TS packets. The code you point out is at tsp-level and simply checks that the input looks like valid TS packets. Some input plugins extracts packets from a "binary magma". This is the case for `file`  (multiple formats, determined in initial phase only), `pcap`, `ip` and `srt` (extracting TS packets from UDP datagrams using various encapsulations formats such as RTP but not only), `dvb` on Windows (because of experiences corruption or desynchronization in DirectShow), etc.

So, simply adding some resynchronization layer inside tsp is not correct because it does not take into account the work that was done upstream in the input plugin. For instance, in the `file` plugin, if a part of an M2TS file is corrupted, trying to resynchronize in tsp would be hopeless since the `file` plugin skips 4 bytes every 192 bytes. If the input is desynchronized and the input plugin continues to skip 4 bytes every 192 bytes, you will never get a correct stream in tsp, even with a resynchronization layer inside tsp.

So, all resynchronization work shall be done inside each input plugin, one by one, depending in the global binary format from which TS packets are extracted. Some plugins already do that such as datagram-based plugins (`ip`, `srt`). For all others, this must be addressed one by one, with various levels of difficulties. I have no plan to enter this for now.

lars18th, 2021-04-26:
Hi @lelegard ,

Thank you for the explanation. Now I understand the complexity of implementing this in a general way. However, I'm happy if some of the input plugins have almost some protection over potential spureous data.

However, for the _particular case of PIPE inputs_, why not provide a _simple_ enhancement (for sure, pure optional)? At time, if the first byte is not 0x47 then it's impossible to start the processing. And I want to overcome this. So, I prefer to disable this check, and in case of any other non-TS-data, advance until the next valid TS packet.

Please, can you almost target this particular case? Without it, it's very complex to integrate TSDuck with external tools that use non-reliable sources.
Thank you!

lelegard, 2021-04-26:
> However, for the particular case of PIPE inputs, 

There is no pipe input, just the `file` plugin which also transparently handles pipes when the standard input is a pipe.

> why not provide a simple enhancement (for sure, pure optional)? At time, if the first byte is not 0x47 then it's impossible to start the processing. And I want to overcome this. So, I prefer to disable this check, and in case of any other non-TS-data, advance until the next valid TS packet.

If you look at the documentation of the `file` input plugin and section 2.1.2 of the TSDuck user's guide, you will see that the plugin already handles non pure TS input files (see option `--format`). So, the plugin already analyzes the input data, trying to guess the actual file format. If you add the constraint that the input can be desynchronized or corrupted, that makes a double guess. 

> Please, can you almost target this particular case? Without it, it's very complex to integrate TSDuck with external tools that use non-reliable sources.

Use or write an adapter to make it reliable (if `tsresync` is not enough).

lars18th, 2021-04-26:
Hi @lelegard ,

Thank you another time. After much reading I continue with some questions. So please help me to understand this:

- With the current implementation of the `file` input, why if the _first_ byte is not a 0x47 sync byte the `tsp` process exits? Any idea to overcome this?

And regarding the use of the `tsresync` tool previous to the `tsp`:

- Why we need to instantiate _two_ buffers, and fill the first to just output the data to the second one? This increases a lot the delay. So I prefer if the code of the `tsresync` will be included (optionally, of course) inside the `file` input. I feel the code will be the same, or not?

Regards.

lelegard, 2021-04-26:
Why don't you just look at the code to understand the complexity?

lars18th, 2021-04-26:
> Why don't you just look at the code to understand the complexity?

Please don't think that I don't try to understand as much as I can.

My current problem is that I'm using this to overcome the exit of the TSDuck if some non-TS-data outputs from the ingest process: `external_process | tsresync -c -m 1318 -p 188 -s 13180 | tsp --realtime ...`

However, the problem here is how to determine the best `-m` and `-s` values.
- For the `-m` parameter, perhaps it's best to use 564 bytes. This assumption is because in order to guarantee a consistent TS synchronization, a _three_ consecutive marks at 188 bytes in between, is the objective. So, what's the best value for a _realtime_ flow?
- More complex it the `-s`parameter. As I want to reduce the latency to the minimun, then I use here 10 * 7 * 188 bytes. But this generates troubles with VBR streams. If the souce process lost some packets at one point, then the flush of the packets is not done based on time (the buffer of `-s` is size-based controlled). Then the problem is here that the `tsp` process only receives a chunk of packets.

I can understand that maintain the simplicity it's a must in this project (and in general too). However, compared with the UDP network input, the file input is less reliable. So, for this reason I prefer to use the network streaming input, even using the loopback for interconnect two local processes. But the lack of event-control of subprocess from TSDuck (i.e. #757) makes impossible to execute a _reliable_multiprocess_execution_. You pointed a lot of times that you don't think it's a good idea to implement a subprocess plugin event manager. And this is acceptable. But, now we don't have a reliable _filtered_ communication using PIPES. So, what's the solution?

Please, try to see the specific scenario to resolve: run more than one process, including `tsp` in the chain, as one unique flow, and exit _only_ if a severe error occurs. So in the event of spurious data, the processing continues. And if only one process dies, then all the others exit. You have any suggestion to solve this particular use case?

Thank you for taking the time to answer so patiently all these questions.
Regards.