tsduck/tsduck #768: [Q] tsp -I hls gives a synchronization lost error

vandanachadha, 2021-04-23:
tsp  -I hls http://50.239.254.203/media/knorzydr/live.m3u8
* Error: hls: synchronization lost after 0 packets, got 0x00 instead of 0x47

Pls suggest how to fix this. Tried with -P filter -v  but still the same error

lelegard, 2021-04-23:
Hi @vandanachadha,

The message means that what you are receiving is not a transport stream. Use `tsp --debug ...` to view the returned HTTP headers, which may give clues about what is going on.

Apart from misconfigured servers or other network error, the only "normal" situation I can think about is HLS serving fMP4 files instead of TS files. The HLS protocol can serve transport stream segments (TS) or fragmented MPEG 4 segments (fMP4). TSDuck is a TS toolbox and supports only TS format. The HTTP headers, as seen using `--debug` should help you.

lars18th, 2021-04-25:
Hi @lelegard ,

This issue is identical (or similar) to #767. I suggest to add some "continue with incoming illegal data" option to the input. The idea is quite simple: if the _input_ (and only at the input) has a data that is not a valid TS 188 bytes packet starting with 0x47 sync byte, then try to resync and continue... instead of exit.

This will make the processing of the TSDuck toolkit more robust when connecting it with other external tools.
I hope you want to implement it.
Thank you!

lelegard, 2021-04-25:
This specific issue has probably nothing to do with desynchronized TS input. Here, what is supposed to be an HLS segment over HTTP does not start with 0x47. This is must probably completely different, either fMP4 HLS or an error page from a misconfigured HLS server.

lars18th, 2021-04-26:
Hi @lelegard ,

Yes, the root cause can be different. But if we have the option to continue the processing and remove non TS data, and if we can print the content of this spurious data, then it will be more easy to identify the cause. And futhermore, the execution of any TSDuck processing will be more robust to glitches caused by other processes. So I feel the advantage of this "safe running in case of spurious data" it's worth putting some effort into it.

vandanachadha, 2021-04-26:
@lelegard I should have added that the same hls with tsswitch command works fine, but not with tsp. Why is this?

Also yes, the segment files are .m4s, are you saying that only .ts can used for segment files, when using tsp? Is there any way to convert .m4s to .ts and pass them to tasp, or does that have to happen outside of tsduck, at the hls source only?

Command which works
```
tsswitch -v  --remote 4444 --delayed-switch --cycle 1 --receive-timeout 2000  -I  hls http://50.239.254.203/media/knorzydr/live.m3u8     -O   file /var/www/vandana/public/reports/recording_210423.ts
```

I also ran the hsp --debug, output is
```
tsp --debug -I hls http://50.239.254.203/media/knorzydr/live.m3u8
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp --debug
* Debug: debug level set to 1
* Debug: hls: debug level set to 1
* Debug: hls: ====> tsp -I hls http://50.239.254.203/media/knorzydr/live.m3u8
* Debug: file: debug level set to 1
* Debug: file: ====> tsp -O file 
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: hls: downloading http://50.239.254.203/media/knorzydr/live.m3u8
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.14.0 (Ubuntu)
* Debug: hls: HTTP header: Date: Mon, 26 Apr 2021 07:38:52 GMT
* Debug: hls: HTTP header: Content-Type: application/vnd.apple.mpegurl
* Debug: hls: HTTP header: Content-Length: 316
* Debug: hls: announced content size: 316 bytes
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Last-Modified: Sun, 25 Apr 2021 03:15:26 GMT
* Debug: hls: HTTP header: ETag: "6084dece-13c"
* Debug: hls: HTTP header: Accept-Ranges: bytes
* Debug: hls: HTTP header: Expires: Mon, 26 Apr 2021 07:38:51 GMT
* Debug: hls: HTTP header: Cache-Control: no-cache
* Debug: hls: HTTP header: Cache-Control: no-cache
* Debug: hls: MIME type: application/vnd.apple.mpegurl
* hls: downloaded live.m3u8, master playlist, static, 1 media playlists
* hls: selected playlist: video.m3u8, 1280x720, 2,380,983 b/s
* Debug: hls: downloading http://50.239.254.203/media/knorzydr/video.m3u8
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.14.0 (Ubuntu)
* Debug: hls: HTTP header: Date: Mon, 26 Apr 2021 07:38:52 GMT
* Debug: hls: HTTP header: Content-Type: application/vnd.apple.mpegurl
* Debug: hls: HTTP header: Content-Length: 363
* Debug: hls: announced content size: 363 bytes
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Last-Modified: Mon, 26 Apr 2021 07:38:50 GMT
* Debug: hls: HTTP header: ETag: "60866e0a-16b"
* Debug: hls: HTTP header: Accept-Ranges: bytes
* Debug: hls: HTTP header: Expires: Mon, 26 Apr 2021 07:38:51 GMT
* Debug: hls: HTTP header: Cache-Control: no-cache
* Debug: hls: HTTP header: Cache-Control: no-cache
* Debug: hls: MIME type: application/vnd.apple.mpegurl
* Debug: hls: unexpected segment file extension in reference URI: video_105366.m4s
* Debug: hls: unexpected segment file extension in reference URI: video_105367.m4s
* Debug: hls: unexpected segment file extension in reference URI: video_105368.m4s
* Debug: hls: unexpected segment file extension in reference URI: video_105369.m4s
* Debug: hls: unexpected segment file extension in reference URI: video_105370.m4s
* hls: downloaded video.m3u8, media playlist, updatable (live), 5 segments, 4 seconds/segment
* Debug: hls: internal push-input thread started
* Debug: hls: downloading segment http://50.239.254.203/media/knorzydr/video_105366.m4s
* hls: downloading from http://50.239.254.203/media/knorzydr/video_105366.m4s
* hls: MIME type: unknown, expected size: unknown
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: nginx/1.14.0 (Ubuntu)
* Debug: hls: HTTP header: Date: Mon, 26 Apr 2021 07:38:52 GMT
* Debug: hls: HTTP header: Content-Type: application/octet-stream
* Debug: hls: HTTP header: Content-Length: 749650
* Debug: hls: announced content size: 749650 bytes
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Last-Modified: Mon, 26 Apr 2021 07:38:38 GMT
[... 23 lines cut ...]
    0120:  00 00 05 DE 00 00 05 DD 00 00 19 15 00 01 00 00  ................
    0130:  00 00 05 DD 00 00 05 DE 00 00 02 2F 00 01 00 00  .........../....
    0140:  00 00 05 DE 00 00 05 DD 00 00 14 25 00 01 00 00  ...........%....
    0150:  00 00 05 DD 00 00 05 DE 00 00 02 DE 00 01 00 00  ................
    0160:  00 00 05 DE 00 00 05 DD 00 00 01 A2 00 01 00 00  ................
    0170:  00 00 05 DD 00 00 05 DE 00 00 11 24 00 01 00 00  ...........$....
    0180:  00 00 05 DE 00 00 05 DD 00 00 02 AE 00 01 00 00  ................
    0190:  00 00 05 DD 00 00 05 DE 00 00 0E 48 00 01 00 00  ...........H....
    01A0:  00 00 05 DE 00 00 05 DD 00 00 05 B7 00 01 00 00  ................
    01B0:  00 00 05 DD 00 00 05 DE 00 00 10 44 00 01 00 00  ...........D....
    01C0:  00 00 05 DE 00 00 05 DD 00 00 10 C1 00 01 00 00  ................
    01D0:  00 00 05 DD 00 00 05 DE 00 00 0F C9 00 01 00 00  ................
    01E0:  00 00 05 DE 00 00 05 DD 00 00 10 45 00 01 00 00  ...........E....
    01F0:  00 00 05 DD 00 00 05 DE 00 00 10 44 00 01 00 00  ...........D....
    0200:  00 00 05 DE 00 00 05 DD 00 00 10 45 00 01 00 00  ...........E....
    0210:  00 00 05 DD 00 00 05 DE 00 00 17 3A 00 01 00 00  ...........:....
    0220:  00 00 05 DE 00 00 05 DD 00 00 09 50 00 01 00 00  ...........P....
    0230:  00 00 05 DD                                      ....

* Debug: hls: error pushing packets
* Debug: hls: Web transfer is interrupted by application
* hls: HLS playlist completed
* Debug: hls: internal push-input thread completed
* Debug: Report logging thread terminated
```

lelegard, 2021-04-26:
This is very clear:

~~~
* Debug: hls: unexpected segment file extension in reference URI: video_105366.m4s
* Debug: hls: unexpected segment file extension in reference URI: video_105367.m4s
....
* Debug: hls: data at lost of synchronization:
    0000:  00 00 00 24 73 74 79 70 69 73 6F 6D 00 00 00 00  ...$stypisom....
    0010:  69 73 6F 38 6D 70 34 31 64 61 73 68 61 76 63 31  iso8mp41dashavc1
    0020:  63 6D 66 73 00 00 00 34 73 69 64 78 01 00 00 00  cmfs...4sidx....
~~~

As I speculated in a previous reply, your HLS server produces segments in fMP4 format, one of the allowed formats for HLS with TS. TSDuck is a "Transport Stream toolbox" and manages TS format only. The purpose of TSDuck is not to manage HLS in all its forms. Its purpose is to manage TS in all (or most of) its possible input mechanisms, HLS being one of them.

TSDuck cannot be used to receive HLS data in fMP4 format.

lars18th, 2021-04-26:
Hi @lelegard ,

I agree with the fMP4 format support.

However, regarding the _optional_ support of "safe running in case of spurious data"... can we continue in #767, please?