tsduck/tsduck #769: Sample input file for tsemmg

nikpakar, 2021-04-26:
Hello everyone,

Does anyone has a sample input file/format for the tsemmg ?

Thanks in advance for the help.

-Nik

lelegard, 2021-04-26:
The input file contain sections in binary or XML format. See the TSDuck user's guide for more details.

If your question is linked to #763, you make me worry. Your attempt to build a production CAS based on TSDuck is not appropriate. The dynamics of a real EMMG is completely different from what `tsemmg` does. Unlike ECM's which are requested by the muxing system at the cadence of an SCS, an EMMG is driven by the CAS crypto-design. EMM scheduling and broadcasting is a CAS-specific "art".

Here `tsemmg`, just like `tsecmg`, is a test and integration tool for a muxing & broadcasting system. It only sends or generates fake data as EMM's with various configurations to exercise the MUX.

nikpakar, 2021-04-26:
Hi Thierry,

Once again many thanks for your response. I understand your concern and totally agree with you. As i mentioned my approach is quite lighter, simpler than a generic CAS. EMM shall only send an EMM once for one time activation purpose and there is no need to continuous EMM broadcast. This EMM (encrypted in RSA) will carry a AES128 key which will be saved in STB flash for the purpose of decrypting the ECM and thats it. No smart cards, STB chipset itself does this simple RSA and AES decryption and that serves the purpose.

Having said that, i totally understand what you said and i will probably give it a start with tsemmg and tsecmg and once i get the concept working, may be will look into redo the both generators from the scratch.

Hence your kind and valuable feedback is highly appreciated.

Btw, i went through the user guide few times, but could not find a specific EMM xml format. Would highly appreciate your kind help to locate the correct sample xml section.

Thanks again
-nik

lelegard, 2021-04-26:
> Btw, i went through the user guide few times, but could not find a specific EMM xml format. Would highly appreciate your kind help to locate the correct sample xml section.

There is no standard EMM format. Each CAS has its own proprietary (and secret) EMM format. In XML, you should use `<generic_short_table>`. The table ids for EMM's are usually in the range 0x82 to 0x8F, using CAS-specific allocation. So your pseudo-EMM's should look like this:
~~~
<generic_short_table table_id="0x86">
  01 02 34 6F DE AD BE EF ....
</generic_short_table>
~~~

nikpakar, 2021-04-26:
Thank you so much. Highly appreciate your help.

closed 2021-04-26 by nikpakar