tsduck/tsduck #770: [Q] Video tearing when switching between videos using tsswitch.

amit2maha1, 2021-04-27:
Hello there.
- We are using tsswitch to play a bunch of videos in sequence
- Using control port we are switching from one video to the next video using nodejs
- We are getting video tearing when switching the videos
- Is the "Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation" related to this issue?

Following is the command being used
` tsswitch --infinite --remote 2100 -I fork "tsp -I file C:/wafttech/cuecoder/encoder/assets/recording/436379cd-11e6-4ef0-a198-58c69b139803/1619441726.ts -P regulate --pcr-synchronous" -I fork "tsp -I file C:/wafttech/cuecoder/encoder/assets/recording/436379cd-11e6-4ef0-a198-58c69b139803/1619441769.ts --infinite -P regulate --pcr-synchronous" -I fork "tsp -I file C:/wafttech/cuecoder/encoder/assets/recording/436379cd-11e6-4ef0-a198-58c69b139803/1619441772.ts -P regulate --pcr-synchronous" -I fork "tsp -I file C:/wafttech/cuecoder/encoder/assets/recording/436379cd-11e6-4ef0-a198-58c69b139803/1619441774.ts --infinite -P regulate --pcr-synchronous" -I fork "tsp -I file C:/wafttech/cuecoder/encoder/assets/recording/436379cd-11e6-4ef0-a198-58c69b139803/1619441777.ts -P regulate --pcr-synchronous" -O ip -l 127.0.0.1 225.1.2.3:2000`

* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation
* Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation

lelegard, 2021-04-27:
There are two unrelated issues: video tearing and the out of sync PCR messages.

You must have two sorts of video tearing. First, when staying on the same source, when an input file loops back from end to start of file. All audio and video streams are broken. The player suddenly receives packets (from the beginning of file) right after unrelated packets (from the end of file). The time stamps are broken, the video frames are broken. Depending on the codec and the player, the effect may be more or less ugly. After the first I-frame in the file, the video should properly restart.

The second sort of video tearing is when you switch inputs. The video player suddenly receives packets which are completely unrelated to previous ones because they come from another TS source. The effect may be worse. For instance, if the two PAT's have, by accident, the same version number and the structure of the PID's and services are different, the switch may simply turn into a black screen. If the PAT and PMT version numbers are different, the transition will range from smooth to ugly. It depends on how your player manages these disruptions. After the first I-frame from the new input, the video should properly restart.

The warning messages "out of sequence PCR" are reported by the regulate plugins in the various source commands. They are due to the loopback of the input files. All along the file, the PCR clock reference increases and then suddenly decreases back to the value at the beginning of the file. The message adds "maybe source was cycling", which is exactly the case. This is just a harmless warning to inform that the regulation restarted from another reference.

amit2maha1, 2021-04-27:
> There are two unrelated issues: video tearing and the out of sync PCR messages.
>
> You must have two sorts of video tearing. First, when staying on the same source, when an input file loops back from [...]

First of all, thank you for replying. You are right, I get both kinds of video tearing. Is there anyway to avoid these tearings? Can we do something in tsswitch command to fix these tearing?

lelegard, 2021-04-27:
>  Is there anyway to avoid these tearings? Can we do something in tsswitch command to fix these tearing?

These are direct consequences of what you are doing; creating a "broken" video stream because of a sudden disruption in the video encoded stream (file loop back or new input). This is a matter of video, not transport stream. The only way to solve it is a matter of video tools, not TS tools. Either the input switch is managed by a video encoder (not tsswitch) which creates a smooth video transition between inputs or use a more resistant media player which nicely handles the transition between incomplete video frames.

amit2maha1, 2021-04-27:
> These are direct consequences of what you are doing; creating a "broken" video stream because of a sudden disruption in the video encoded stream (file loop back or new input). This is a matter of [...]

I see. Btw if we capture a live stream with tsp and stop with tcpcontrol. will it end clean?

lelegard, 2021-04-27:
> Btw if we capture a live stream with tsp and stop with tcpcontrol. will it end clean?

What do you mean by "clean"? The result is clean at transport stream level. The end of the output is at a TS packet boundary, not packet will be missing (unless already missing during original reception). But it won't be clean at video level. It will be exactly the same as what you experiment with tsswitch.

A transport stream is a pipe and TSDuck is the plumber. The plumbing is clean. But what is flowing through the pipes is not controlled by TSDuck when cutting the pipe: audio, video, data, subtitles, carousels, MPE, T2-MI, teletext, etc.

amit2maha1, 2021-04-28:
> What do you mean by "clean"? The result is clean at transport stream level. The end of the output is at a TS packet boundary, not packet will be missing (unless already missing during original [...]

I see, thank you.

closed 2026-04-26 by github-actions[bot]