tsduck/tsduck #771: Error using dektec  DTA-2174B

izharm, 2021-04-27:
Hi,

I am trying to work with Dektec DTA2174B 4 ports card, in ASI mode.  The device is detected by tsdektec (see below),

trying to use the output port I get the following error:

```
tsp -v  -I null  -O dektec -d 1
* null: unknown initial input bitrate
* dektec: using Dektec device 1, output channel 0 (DTA-2174B port 4)
* Error: dektec: output device SetTxMode error: DTAPI_E_INVALID_MODE (DTAPI status 4108)
```

it may be that the default configuraiotn for the output port is SDI and not ASI.

from dektec SDK :"Note: For ASI/SDI input channels, first configure the output port for ASI or SDI with SetIoConfig, group IOSTD. If an ASI-specific transmit mode is applied to a channel that is configured for SDI (or vice versa), an error will be returned"

Thanks,

Izhar

```
tsdektec -av

DTAPI: 5.43.2.149, DtPcie: 1.11.1.263, Dta: 4.28.1.271, DtaNw: -1.-1.-1.-1

* Device 0: DTA-2162 (Dual GigE ports)
  Physical ports: 2
  Channels: input: 2, output: 2
  Input 0: DTA-2162 port 1 (top), TS-over-IP, IP 0.0.0.0, MAC 00:14:F4:04:90:0E
      (Network port supports failover, Transparent mode, MPEG-2 transport
      stream, Transport stream over IP, Internal TS rate clock reference)
  Input 1: DTA-2162 port 2, TS-over-IP, IP 0.0.0.0, MAC 00:14:F4:04:90:0F
      (Network port supports failover, Transparent mode, MPEG-2 transport
      stream, Transport stream over IP, Internal TS rate clock reference)
  Output 0: DTA-2162 port 1 (top), TS-over-IP, IP 0.0.0.0, MAC 00:14:F4:04:90:0E
      (Network port supports failover, Transparent mode, MPEG-2 transport
      stream, Transport stream over IP, Internal TS rate clock reference)
  Output 1: DTA-2162 port 2, TS-over-IP, IP 0.0.0.0, MAC 00:14:F4:04:90:0F
      (Network port supports failover, Transparent mode, MPEG-2 transport
      stream, Transport stream over IP, Internal TS rate clock reference)
  Subsystem id: 0x0000 (DTA-2162)
  Subsystem vendor id: 0x0000
  Device id: 0x0872
  Vendor id: 0x1A0E
  Serial number: 0000000080DD8887
  Firmware version: 3 (0x00000003)
  Firmware variant: 0 (0x00000000)
  PCI bus: 179, slot: 0
  Customer id: 302204
  Engineering change level: Rev 3
  Manufacture id: 04
  Production date: 2020.03
  Part number: DTA-2162
  Serial number: 2162002055

* Device 1: DTA-2174B (Quad 3G-SDI/ASI ports (1x12G) with genlock)
  Physical ports: 6
  Channels: input: 3, output: 1
  Input 0: DTA-2174B port 1 (top), ASI, SDI (DMA-rate test mode, Locked to a
      genlock reference, Uni-directional input, Uni-directional output, Double
      buffered output, ASI output signal can be inverted, High-level Matrix API
      support, Transparent mode, MPEG-2 transport stream, Transmit on timestamp,
      3G-SDI, DVB-ASI transport stream, HD-SDI, SD-SDI, 1080p/50 lvl A, 1080p/50
      lvl B, 1080p/59.94 lvl A, 1080p/59.94 lvl B, 1080p/60 lvl A, 1080p/60 lvl
      B, 1080i/50, 1080i/59.94, 1080i/60, 1080p/23.98, 1080p/24, 1080p/25,
      1080p/29.97, 1080p/30, 1080psf/23.98, 1080psf/24, 1080psf/25, 1080psf/29.
      97, 1080psf/30, 720p/23.98, 720p/24, 720p/25, 720p/29.97, 720p/30,
      720p/50, 720p/59.94, 720p/60, 525i/59.94, 625i/50)
  Input 1: DTA-2174B port 2, ASI, SDI (DMA-rate test mode, Locked to a genlock
      reference, Uni-directional input, Uni-directional output, Double buffered
      output, ASI output signal can be inverted, High-level Matrix API support,
      Transparent mode, MPEG-2 transport stream, Transmit on timestamp, 3G-SDI,
      DVB-ASI transport stream, HD-SDI, SD-SDI, 1080p/50 lvl A, 1080p/50 lvl B,
      1080p/59.94 lvl A, 1080p/59.94 lvl B, 1080p/60 lvl A, 1080p/60 lvl B,
      1080i/50, 1080i/59.94, 1080i/60, 1080p/23.98, 1080p/24, 1080p/25,
      1080p/29.97, 1080p/30, 1080psf/23.98, 1080psf/24, 1080psf/25, 1080psf/29.
      97, 1080psf/30, 720p/23.98, 720p/24, 720p/25, 720p/29.97, 720p/30,
      720p/50, 720p/59.94, 720p/60, 525i/59.94, 625i/50)
  Input 2: DTA-2174B port 3, ASI, SDI (DMA-rate test mode, Locked to a genlock
      reference, Uni-directional input, Uni-directional output, Double buffered
      output, ASI output signal can be inverted, High-level Matrix API support,
      Transparent mode, MPEG-2 transport stream, Transmit on timestamp, 3G-SDI,
      DVB-ASI transport stream, HD-SDI, SD-SDI, 1080p/50 lvl A, 1080p/50 lvl B,
      1080p/59.94 lvl A, 1080p/59.94 lvl B, 1080p/60 lvl A, 1080p/60 lvl B,
      1080i/50, 1080i/59.94, 1080i/60, 1080p/23.98, 1080p/24, 1080p/25,
      1080p/29.97, 1080p/30, 1080psf/23.98, 1080psf/24, 1080psf/25, 1080psf/29.
      97, 1080psf/30, 720p/23.98, 720p/24, 720p/25, 720p/29.97, 720p/30,
      720p/50, 720p/59.94, 720p/60, 525i/59.94, 625i/50)
  Output 0: DTA-2174B port 4, ASI, SDI (DMA-rate test mode, Locked to a genlock
      reference, Uni-directional input, Uni-directional output, Double buffered
      output, ASI output signal can be inverted, High-level Matrix API support,
      Transparent mode, MPEG-2 transport stream, Transmit on timestamp, 3G-SDI,
      DVB-ASI transport stream, HD-SDI, SD-SDI, 1080p/50 lvl A, 1080p/50 lvl B,
      1080p/59.94 lvl A, 1080p/59.94 lvl B, 1080p/60 lvl A, 1080p/60 lvl B,
      1080i/50, 1080i/59.94, 1080i/60, 1080p/23.98, 1080p/24, 1080p/25,
      1080p/29.97, 1080p/30, 1080psf/23.98, 1080psf/24, 1080psf/25, 1080psf/29.
      97, 1080psf/30, 720p/23.98, 720p/24, 720p/25, 720p/29.97, 720p/30,
      720p/50, 720p/59.94, 720p/60, 525i/59.94, 625i/50)
  Subsystem id: 0x0000 (DTA-2174B)
  Subsystem vendor id: 0x0000
  Device id: 0xB87E
  Vendor id: 0x1A0E
  Serial number: 000000008197F765
  Firmware version: 2 (0x00000002)
  Firmware variant: 1 (0x00000001)
  PCI bus: 23, slot: 0
  Customer id: 302204
  Engineering change level: Rev 6.4
  Manufacture id: 04
  Production date: 2020.11
  Part number: DTA-2174
  Serial number: 2174220133

```

lelegard, 2021-04-27:
Hi @izharm,

Thanks for reporting this. I do not have such card to test. So, we must jointly work in this. I can update the code (unless you want to do it yourself and submit a pull request). But you will have to test the result.

This is not the first time a specific configuration issue on a specific Dektec device is reported. It usually takes one or two modification/test round trips to fix the issue.

What is your operating system, Linux or Windows, and which version?

Do you have the capabilities to rebuild from source (development environment)?

izharm, 2021-04-27:
Hi @lelegard 

Thanks for the quick response, I'm more than happy to test.

I am using Ubuntu 20.04.1 LTS, 

I can rebuild from sources, though it may take a bit longer to setup.

BTW, I think there is also a bug in tsdektec utility, failing to set the I/O direction, possibly it is the same issue.  I didn't verify so I didn't report...

Izhar

lelegard, 2021-04-27:
> BTW, I think there is also a bug in tsdektec utility, failing to set the I/O direction, possibly it is the same issue. 

Could you please post the command and the error?

izharm, 2021-04-28:
tsdektec -o1 1
* Error setting port 1 to output mode: DTAPI_E_INVALID_ARG (DTAPI status 4140)

lelegard, 2021-04-28:
Hi @izharm,

I have added option `--io-standard` to the two dektec plugins (input and output) to select one of the 55 available standards which are supported by the Dektec API (not all standards are supported on all devices of course).

In your case, that translates to `-O dektec -d 1 --io-standard asi`

Could you rebuild from the latest source and try? Alternatively, for an Ubuntu build, you may wait until tomorrow and grab the [next nightly build](https://tsduck.io/download/prerelease/). It must be version 3.27-2355 or higher.

The option  `--io-standard` forces a standard just before `SetTxMode` (output plugin) as indicated in the Dektec API doc.

Your error with `tsdektec -o1 1` is a bit more worrying. The DTA-2174B has 4 input/output ports. Each port must be configured either as input or output. This probably must be done anyway before setting any kind of mode. So, it seems unrelated to the other issue. I found no clue in the Dektec doc and it is complicated to test and fix without an actual device.

Note that since setting port 1 to output fails, the first output port is port 4 (seen on your display from `tsdektec -av`). So, in your `tsp -O dektec`, the output is on port 4. Just to make sure you plug the right port if you can make the plugin work with the new option.

From your display of `tsdektec -av`, you have the latest versions of the drivers, compatible with the DTAPI version. So, there is no mismatch.

lelegard, 2021-04-28:
If you decide to rebuild TSDuck for the first time, [have a look here](https://tsduck.io/doxy/building.html)

izharm, 2021-04-29:
using latest nightly build  - RX seems ot be working but Tx still fails:

```
$ tsp -v -d -I dektec -d1 --io-standard asi -O dektec -d 1 --io-standard asi
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -v -d
* Debug: debug level set to 1
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -I dektec -d1 --io-standard asi
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -O dektec -d 1 --io-standard asi
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* dektec: using Dektec device 1, input channel 0 (DTA-2174B port 1)
* Debug: dektec: setting IO config of port 1, group: 1, value: 33, subvalue: -1
* Debug: dektec: initial buffer load: 33,465 packets, 6,291,420 bytes
* dektec: initial input bitrate is 29,999,792 b/s
* dektec: using Dektec device 1, output channel 0 (DTA-2174B port 2)
* Debug: dektec: setting IO config of port 2, group: 1, value: 33, subvalue: -1
* dektec: output fifo size: 8,388,608 bytes, max: 8,388,608 bytes, typical: 8,388,608 bytes
* Error: dektec: output device set bitrate error: DTAPI_E_NOT_SUPPORTED (DTAPI status 4119)
* Debug: start() error in output plugin dektec
* Debug: Report logging thread terminated
```
I am not sure this is related but you may also want to set the I/O direction, I think this will negate the need for setting the I/O externally, which currently doesn't work...

the following works for input, we didn't try yet in output mode:

dr = Dvc.SetIoConfig(x2174_port, DTAPI_IOCONFIG_IODIR, DTAPI_IOCONFIG_INPUT, DTAPI_IOCONFIG_INPUT);

lelegard, 2021-04-29:
Hi @izharm,

Thanks for the new tests. It seems that we are progressing. Setting I/O standard, here ASI, seems to work.

On output, the new problem appears later on, while setting the output TS bitrate. The returned status is "not supported", which seems weird for ASI devices.

I can propose an possible explanation, to be confirmed by further tests. In TSDuck version 3.26, at a user's request, bitrates are now fractional (currently one decimal digit) because many modulation modes result in non-integral bitrates. The difference is not much (a matter of tenths of a bit per second) but it makes a difference on the long term. This modification of the internal bitrate representation is now consistent all over TSDuck. In most cases, the decimals only matter in intermediate results during calculations. When bitrate values are used (displayed, passed to external API's), the values are rounded to an integral value. In the Dektec API, the output TS bitrate can be set using two different calls: one using an integer value and one using a fractional representation. So, I naturally switched to the fractional version with the new fractional bitrate representation. It is _possible_ (this is my hypothesis), that the fractional version of `SetTsRateBps` is supported only for modulator devices. The DTAPI does not specify anything on this, this is just a guess.

As a test, I did the following modification: the fractional version of `SetTsRateBps` is called first because it is more precise. If the returned status is "not supported", the integer version of `SetTsRateBps` is then called with the rounded value. Let's see if it works that way.

> I am not sure this is related but you may also want to set the I/O direction, I think this will negate the need for setting the I/O externally, which currently doesn't work...

I cannot do that in the `dektec` input and output plugin for an unfortunate historial reason. When I started to use Dektec ASI devices, the very first Dektec product back in 2005, a device was single-port and either input or output, but not both. Later, when Dektec introduced multi-port devices, a port was also either input or output, but not both.

So, within TSDuck, for a given execution of `tsp`, "input ports" and "output ports" are considered separately. When starting `tsp`, the current input / output state of the devices is used as such. This is why you see this:
~~~
$ tsdektec -av
...
* Device 1: DTA-2174B (Quad 3G-SDI/ASI ports (1x12G) with genlock)
  Physical ports: 6
  Channels: input: 3, output: 1
  Input 0: DTA-2174B port 1 (top), ASI, SDI (DMA-rate test mode, Locked to a
...
  Input 1: DTA-2174B port 2, ASI, SDI (DMA-rate test mode, Locked to a genlock
...
  Input 2: DTA-2174B port 3, ASI, SDI (DMA-rate test mode, Locked to a genlock
...
  Output 0: DTA-2174B port 4, ASI, SDI (DMA-rate test mode, Locked to a genlock
...
~~~
The device is assumed to have 3 input channels and 1 output channel. But this is only the current configuration of the device.

The `dektec` output plugin options are:
~~~
-c value
--channel value
    Channel index on the output Dektec device. By default, use the first
    output channel on the device.
...
-d value
--device value
    Device index, from 0 to N-1 (with N being the number of Dektec devices in
    the system). Use the command "tsdektec -a [-v]" to have a complete list
    of devices in the system. By default, use the first output Dektec device.
~~~
So, you can only use ports which are already configured as output. You can say "use output channel #0" (here physical port 4 in the current configuration) but there is no way to say "use physical port #2 and configure it as output". This may change in the future, but there is some internal logic to change.

This is why the input/output direction is externally set by `tsdektec`. When `tsp` is started after using `tsdektec`, it uses the new input / output configuration of the ports.

So, we need to fix the `tsdektec` error.

> the following works for input, we didn't try yet in output mode:
> dr = Dvc.SetIoConfig(x2174_port, DTAPI_IOCONFIG_IODIR, DTAPI_IOCONFIG_INPUT, DTAPI_IOCONFIG_INPUT);

Your code line drew my attention in the repetition of `DTAPI_IOCONFIG_INPUT`, as `SetIoConfig` "value" and "subvalue". In the TSDuck code, the parameters are only `port, DTAPI_IOCONFIG_IODIR, DTAPI_IOCONFIG_INPUT` (only once). I investigated in the `CapList` file which comes with the DTAPI and it says that `SetIoConfig` on group `IODIR` with value `INPUT` shall take one subvalue, one of the possible subvalue being `INPUT` again.

At the beginning of Dektec bidirectional ports, there was only one configuration option: input or output (`SetIoConfig` value, no subvalue). This is where TSDuck was updated to handle input/output setup.

Later, more complex birectional ports were introduced and it seems that a subvalue was added to `SetIoConfig IODIR INPUT`: set input on what? physical port, internal loop back, antenna, etc. This is where the "subvalue" was added. By default, when omitted, the subvalue is -1. It is probable that the DTA-2174 needs a precise subvalue. So, I updated the input / output configuration to always provide the correct subvalue.

These two fixes (fractional vs integral TS output bitrate and input / input configuration with subvalue) have been pushed.

Could you please retest with the updated version? Either rebuilding from the current state of the repo or waiting for tomorrow's nightly build, 3.27-2356 or higher. Be sure to specify `tsp -d` to get the debug traces. And please also test setting input and output directions using `tsdektec`.

izharm, 2021-04-29:
Hi @lelegard 

I want to thank you again for quick and detailed explanation.

I have tried to rebuild but was not successful, I'll wait for the nightly build and retest both tsp and tsdektec.

lelegard, 2021-04-29:
> I have tried to rebuild but was not successful,

What's wrong? 

Maybe you do not have the prerequisites to build.

On an Ubuntu system, the first build is simply:
~~~
git clone https://github.com/tsduck/tsduck.git
cd tsduck
build/install-prerequisites.sh
make installer
~~~
And you get the `.deb` files in the `installers` subdirectory.

For all subsequent builds, from the `tsduck` directory:
~~~
git pull
make installer
~~~

izharm, 2021-05-02:
@lelegard 

I didn't get around to try to rebuild but I tested the nightly build (build 2358).  There is some progress.

it seems the first set Txbitrate works, but the next fails:

```
tsp -v -d -I dektec -d1 --io-standard asi -O dektec -d1 --io-standard asi
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -v -d
* Debug: debug level set to 1
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -I dektec -d1 --io-standard asi
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -O dektec -d1 --io-standard asi
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* dektec: using Dektec device 1, input channel 0 (DTA-2174B port 1)
* Debug: dektec: setting IO config of port 1, group: 1, value: 33, subvalue: -1
* Debug: dektec: initial buffer load: 33,465 packets, 6,291,420 bytes
* dektec: initial input bitrate is 29,999,793 b/s
* dektec: using Dektec device 1, output channel 0 (DTA-2174B port 2)
* Debug: dektec: attaching to device DTA-2174B serial 0x000000008197F765
* Debug: dektec: attaching to port 2
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting IO config of port 2, group: 1, value: 33, subvalue: -1
* Debug: dektec: setting TxMode, tx: 17, stuff: 0
* dektec: output fifo size: 8,388,608 bytes, max: 8,388,608 bytes, typical: 8,388,608 bytes
* Debug: dektec: setting TsRateBps using DtFractionInt: 29999793
* Debug: dektec: setting TsRateBps using DtFractionInt unsupported, using int: 29,999,793
* Debug: dektec: setting TxControl to 3
* dektec: initial output bitrate: 29,999,793 b/s
* dektec: Will start transmission immediately.
* Debug: dektec: input thread started
* Debug: dektec: output thread started
* Debug: control commands: found 0 packet processor plugins
* dektec: Got FIFO underflow.
* dektec: Got FIFO underflow.
* dektec: Got FIFO underflow.
* dektec: Got FIFO underflow.
* Debug: dektec: input: got bitrate 29,999,793 b/s
* Debug: dektec: input: got bitrate 29,999,793 b/s
* dektec: new input bitrate: 29,999,792 b/s
* Debug: dektec: input: got bitrate 29,999,792 b/s
* Error: dektec: error setting output bitrate on Dektec device: DTAPI_E_NOT_SUPPORTED (DTAPI status 4119)
* Error: dektec: error setting output bitrate on Dektec device: DTAPI_E_NOT_SUPPORTED (DTAPI status 4119)
* Error: dektec: error setting output bitrate on Dektec device: DTAPI_E_NOT_SUPPORTED (DTAPI status 4119)
* Error: dektec: error setting output bitrate on Dektec device: DTAPI_E_NOT_SUPPORTED (DTAPI status 4119)
* Error: dektec: error setting output bitrate on Dektec device: DTAPI_E_NOT_SUPPORTED (DTAPI status 4119)
* Error: dektec: error setting output bitrate on Dektec device: DTAPI_E_NOT_SUPPORTED (DTAPI status 4119)

```
a quick go/nogo test of tsdektec also worked!

one more point: setting the output bitrate manually (-b option) works but the configured bitrate is 1/10 of the requested one:

```
 tsp -v -d -I dektec -d1 --io-standard asi -O dektec -d1 --io-standard asi -b 299997930
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -v -d
* Debug: debug level set to 1
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -I dektec -d1 --io-standard asi
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -O dektec -d1 --io-standard asi -b 299997930
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* dektec: using Dektec device 1, input channel 0 (DTA-2174B port 1)
* Debug: dektec: setting IO config of port 1, group: 1, value: 33, subvalue: -1
* Debug: dektec: initial buffer load: 33,465 packets, 6,291,420 bytes
* dektec: initial input bitrate is 29,999,793 b/s
* dektec: using Dektec device 1, output channel 0 (DTA-2174B port 2)
* Debug: dektec: attaching to device DTA-2174B serial 0x000000008197F765
* Debug: dektec: attaching to port 2
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting IO config of port 2, group: 1, value: 33, subvalue: -1
* Debug: dektec: setting TxMode, tx: 17, stuff: 0
* dektec: output fifo size: 8,388,608 bytes, max: 8,388,608 bytes, typical: 8,388,608 bytes
* Debug: dektec: setting TsRateBps using DtFractionInt: 29999793
* Debug: dektec: setting TsRateBps using DtFractionInt unsupported, using int: 29,999,793
```

lelegard, 2021-05-02:
Hi @izharm 

> There is some progress.
> it seems the first set Txbitrate works, but the next fails:
> one more point: setting the output bitrate manually (-b option) works but the configured bitrate is 1/10 of the requested one

These two points are fixed now. 

Could you please try to rebuild and test (or next nightly build). Sorry for all those iterations but, without the corresponding hardware, it is difficult to get it right at once.

izharm, 2021-05-03:
Hi, @lelegard ,

It works|

thanks again.

Izhar

lelegard, 2021-05-03:
Great. Thanks for your patience. That was a good collaboration between developments and tests.

izharm, 2021-05-04:
I also noticed TSDuck doesn't support Dektec DTA-2162, if you want I can help debug this as well.

lelegard, 2021-05-04:
> I also noticed TSDuck doesn't support Dektec DTA-2162, if you want I can help debug this as well.

Thanks for proposing. Let's try.

I never had Dektec Ethernet cards and never had the opportunity to test them. These cards are expensive (€1000+) but they guarantee zero-jitter (do you confirm?). So, they are probably great for production. For me (and many users), TSDuck is a toolbox for tests, demos, debug, etc. In this context, we use Dektec cards when they provide unique features (modulations, ASI). But for network tests, the regular NIC of the system are good enough.

There are some programming specificities with Dektec Ethernet boards. I can implement them "as a guess", without test. So, there is a risk of several iterations. It would have been much simpler with a board to test: you can test various hypotheses, just immediately recompiling one source file between tests. We have experimented longer cycles with the DTA-2174B (although it ended up with a working solution). 

Perhaps, it would be useful for you to be able to recompile TSDuck. You mentioned that you had issues. Did you have the opportunity to try the commands I recommended in a previous post?

izharm, 2021-05-04:
@lelegard, I was able to compile successfully, not sure what went wrong before.
if you want, we can start with ip\rtp input into the 2162, we have such a setup up in the lab.

it is still in testing here, but it seems the timestamping of the input packets is very accurate, the data coming into the FIFO is extremely jittery.  This is also the case for the ASI as well.

you can see this in the following log:

```
tsp -v -I dektec -d1 -P bitrate_monitor -p 10 -t 10 -O drop
* dektec: using Dektec device 1, input channel 0 (DTA-2174B port 1)
* dektec: initial input bitrate is 4,199,958 b/s
* dektec: new input bitrate: 4,199,957 b/s
* bitrate_monitor: 2021/05/04 12:51:24, TS bitrate: 4,060,800 bits/s
* bitrate_monitor: 2021/05/04 12:51:34, TS bitrate: 4,327,760 bits/s
* dektec: new input bitrate: 4,199,958 b/s
* bitrate_monitor: 2021/05/04 12:51:44, TS bitrate: 4,211,200 bits/s
* dektec: new input bitrate: 4,199,957 b/s
* bitrate_monitor: 2021/05/04 12:51:54, TS bitrate: 4,060,800 bits/s
* dektec: new input bitrate: 4,199,958 b/s
* bitrate_monitor: 2021/05/04 12:52:04, TS bitrate: 4,247,296 bits/s
* dektec: new input bitrate: 4,199,957 b/s
* bitrate_monitor: 2021/05/04 12:52:14, TS bitrate: 4,211,200 bits/s
* dektec: new input bitrate: 4,199,958 b/s
* dektec: new input bitrate: 4,199,957 b/s
* bitrate_monitor: 2021/05/04 12:52:24, TS bitrate: 4,211,200 bits/s
* bitrate_monitor: 2021/05/04 12:52:34, TS bitrate: 4,247,296 bits/s
* dektec: new input bitrate: 4,199,958 b/s
* dektec: new input bitrate: 4,199,957 b/s
* bitrate_monitor: 2021/05/04 12:52:44, TS bitrate: 4,211,200 bits/s
* bitrate_monitor: 2021/05/04 12:52:54, TS bitrate: 4,211,200 bits/s
* bitrate_monitor: 2021/05/04 12:53:04, TS bitrate: 4,096,896 bits/s
* dektec: new input bitrate: 4,199,958 b/s
* bitrate_monitor: 2021/05/04 12:53:14, TS bitrate: 4,211,200 bits/s
* dektec: new input bitrate: 4,199,957 b/s
* bitrate_monitor: 2021/05/04 12:53:24, TS bitrate: 4,211,200 bits/s
* bitrate_monitor: 2021/05/04 12:53:34, TS bitrate: 4,247,296 bits/s
* bitrate_monitor: 2021/05/04 12:53:44, TS bitrate: 4,211,200 bits/s
* dektec: new input bitrate: 4,199,958 b/s
* dektec: new input bitrate: 4,199,957 b/s

```

lelegard, 2021-05-05:
Hi @izharm,

I have added support for Dektec Ethernet devices in the two plugins, `dektec` input and output. You will find the new options in the TSDuck user's guide (file `doc/tsduck.pdf` in the repo), reference sections for the two plugins, subsection _"TS-over-IP options"_.

You may pull the repo, rebuild and test.

As mentioned in a previous post, this is pure "blind coding", without device to test. So, we may expect some difficulties. Please report here the results of your tests. In a way, this was rather straightforward, just adding new command line options and directly mapping them into a `DtIpPars` Dektec structure. But a typo or misunderstanding of the DTAPI documentation is still possible.

izharm, 2021-05-06:
OK, moving forward....

output of tcpdump:

```
06:55:29.175538 IP 192.168.10.100.65059 > 226.1.2.3.50111: UDP, length 1316
06:55:29.176241 IP 192.168.10.100.65059 > 226.1.2.3.50111: UDP, length 1316
06:55:29.177364 IP 192.168.10.100.65059 > 226.1.2.3.50111: UDP, length 1316
06:55:29.178442 IP 192.168.10.100.65059 > 226.1.2.3.50111: UDP, length 1316
06:55:29.179447 IP 192.168.10.100.65059 > 226.1.2.3.50111: UDP, length 1316
06:55:29.180455 IP 192.168.10.100.65059 > 226.1.2.3.50111: UDP, length 1316
06:55:29.181558 IP 192.168.10.100.65059 > 226.1.2.3.50111: UDP, length 1316
06:55:29.182270 IP 192.168.10.100.65059 > 226.1.2.3.50111: UDP, length 1316
```

got this result:

```
$ tsp -d -v -I dektec --ip4 226.1.2.3:50111 -c0 -O drop
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d -v
* Debug: debug level set to 1
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -I dektec --ip4 226.1.2.3:50111 -c0
* Debug: drop: debug level set to 1
* Debug: drop: ====> tsp -O drop
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* dektec: using Dektec device 0, input channel 0 (DTA-2162 port 1)
* Debug: dektec: setting IP parameters
* Error: dektec: synchronization lost after 0 packets, got 0x00 instead of 0x47
* Debug: dektec: data at lost of synchronization:
    0000:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0020:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0030:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0040:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0050:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0060:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0070:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0080:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0090:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00A0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00B0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00C0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00D0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00E0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00F0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0100:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0110:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0120:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0130:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0140:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0150:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0160:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0170:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0180:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0190:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    01A0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    01B0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    01C0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    01D0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    01E0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    01F0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0200:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0210:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0220:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0230:  00 00 00 00                                      ....

* Debug: dektec: no initial packet read
* Debug: init buffer error
* Debug: Report logging thread terminated

```

lelegard, 2021-05-06:
Thanks for the tests. Clearly, the reception is "successfully" started, data are received but contain only zeroes. The reception code is common to all Dektec devices, so I don't think there is a problem here. The problem must be in the setup sequence. Something must be missing or incorrect.

I will add more debug traces. In the meantime, could you please perform two tests:
- In the same reception configuration, using tcpdump or wireshark, could make sure that the UDP packets actually contain TS packets and how are their layouts. The packet size is 1316, according to tcpdump, so that is exactly 7x188, the usual configuration of IP-TV streams. So, the content is probably correct, but we need to confirm.
- Second, could you try the output plugin? Since the setup sequence is similar for transmit and receive with Dektec Ethernet devices, it would be interesting to see if packets are correctly transmitted or if there is an error. You could try to send a TS file (or any other TS source), something like `tsp -d -I file in.tst -O dektec --bitrate XXXX --ip4 addr:port`.

izharm, 2021-05-06:
the input stream is OK.

the output also doesn't work:

 ```
tsp -v -d -I ip 51105 -O dektec --bitrate 4200000 --ip4 192.168.10.2:40000
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -v -d
* Debug: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: ip: ====> tsp -I ip 51105
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -O dektec --bitrate 4200000 --ip4 192.168.10.2:40000
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: ip: setting socket reuse address to 1
* Debug: ip: binding socket to 0.0.0.0:51105
* Debug: ip: initial buffer load: 7 packets, 1,316 bytes
* ip: unknown initial input bitrate
* dektec: using Dektec device 0, output channel 0 (DTA-2162 port 1)
* Debug: dektec: attaching to device DTA-2162 serial 0x0000000080DD8887
* Debug: dektec: attaching to port 1
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting TxMode, tx: 17, stuff: 0
* Debug: dektec: setting IP parameters
* Error: dektec: output device SetIpPars error: DTAPI_E_INVALID_VIDSTD (DTAPI status 4240)
* Debug: start() error in output plugin dektec
* Debug: ip: aborting IP input
* Debug: Report logging thread terminated
```

not related to the IP, there is another issue with the ASI to IP (standard interface, not dektec):

```
tsp -v -d -I ip 51105 -O dektec -d1 --io-standard asi
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -v -d
* Debug: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: ip: ====> tsp -I ip 51105
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -O dektec -d1 --io-standard asi
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: ip: setting socket reuse address to 1
* Debug: ip: binding socket to 0.0.0.0:51105
* Debug: ip: initial buffer load: 7 packets, 1,316 bytes
* ip: unknown initial input bitrate
* dektec: using Dektec device 1, output channel 0 (DTA-2174B port 2)
* Debug: dektec: attaching to device DTA-2174B serial 0x000000008197F765
* Debug: dektec: attaching to port 2
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting IO config of port 2, group: 1, value: 33, subvalue: -1
* Debug: dektec: setting TxMode, tx: 17, stuff: 0
* dektec: output fifo size: 8,388,608 bytes, max: 8,388,608 bytes, typical: 8,388,608 bytes
* Debug: dektec: setting TsRateBps using DtFractionInt: 0
* Debug: dektec: setting TsRateBps using DtFractionInt unsupported, using int: 0
* Error: dektec: output device set bitrate error: DTAPI_E_INVALID_RATE (DTAPI status 4109)
* Debug: start() error in output plugin dektec
* Debug: ip: aborting IP input
* Debug: Report logging thread terminated

```
with specific bitrate it works.
tsp -v -d -I ip 51105 -O dektec -d1 --io-standard asi -b 4200000

lelegard, 2021-05-06:
Hi @izharm,

Thanks for the new tests. I may have misinterpreted the DTAPI doc about the default value for the "video standard" field (`DTAPI_E_INVALID_VIDSTD`). I now explicitly specify it as TS. I have also added more debug traces. Could you pull again, rebuild and retry?

Concerning the ASI bitrate issue, this is the expected behaviour in that case. Let me explain why. The Dektec ASI output needs an output bitrate to be set. This bitrate is automatically detected when possible. When this is not possible, the `--bitrate` option is needed. To automatically get the bitrate, we use the "bitrate propagation system" (cf. user's guide on `tsp`).

In short, the bitrate is propagated this way:
- The initial input bitrate is fetched from the input plugin. If the input device is a tuner, a demodulator or ASI, the hardware returns a bitrate. In the case of the `ip` plugin, there is no intrinsic bitrate.
- When the input device does not report a bitrate, `tsp` tries to evaluate the bitrate from the PCR in the initial buffer load (option `--initial-input-packets`). In your case, there was apparently no or not enough PCR in the initial load.
- Each plugin propagates the bitrate downstream, transparently by default, or completely recomputing it (`pcrbitrate` plugin for instance).

In your case, it seems that no input source could provide a valid bitrate. See message _"ip: unknown initial input bitrate"_, probably due to the fact that you use `--initial-input-packets 7` (not seen in your command but obvious from the debug trace _"Debug: ip: initial buffer load: 7 packets, 1,316 bytes"_). So, the bitrate propagation results in zero (message _"Debug: dektec: setting TsRateBps ...: 0"_). So, this is why you need to explicitly specify a bitrate. Arguably, the `dektec`output plugin should report an error when the automatic bitrate is zero.

lelegard, 2021-05-06:
> Arguably, the dektecoutput plugin should report an error when the automatic bitrate is zero.

Warning message added....

izharm, 2021-05-06:
No change: 

$ tsp -d -v -I dektec --ip4 226.1.2.3:50111 -c0 -O drop
```
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d -v
* Debug: debug level set to 1
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -I dektec --ip4 226.1.2.3:50111 -c0
* Debug: drop: debug level set to 1
* Debug: drop: ====> tsp -O drop
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* dektec: using Dektec device 0, input channel 0 (DTA-2162 port 1)
* Debug: dektec: attaching to device DTA-2162 serial 0x0000000080DD8887
* Debug: dektec: attaching to port 1
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting RxControl, mode: 0
* Debug: dektec: clearing FIFO and flags
* Debug: dektec: setting IP parameters: DtIpPars2 = {
* Debug: dektec: m_Ip = E2 01 02 03 00 00 00 00 00 00 00 00 00 00 00 00
* Debug: dektec: m_Port = 50111
* Debug: dektec: m_Gateway = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
* Debug: dektec: m_VlanId = 0
* Debug: dektec: m_VlanPriority = 0
* Debug: dektec: m_Ip2 = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
* Debug: dektec: m_Port2 = 0
* Debug: dektec: m_Gateway2 = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
* Debug: dektec: m_VlanId2 = 0
* Debug: dektec: m_VlanPriority2 = 0
* Debug: dektec: m_TimeToLive = 0
* Debug: dektec: m_NumTpPerIp = 7
* Debug: dektec: m_Protocol = 2
* Debug: dektec: m_DiffServ = 0
* Debug: dektec: m_FecMode = 0
* Debug: dektec: m_FecNumRows = 0
* Debug: dektec: m_FecNumCols = 0
* Debug: dektec: m_Flags = 0x00000000
* Debug: dektec: m_Mode = 0
* Debug: dektec: m_IpProfile.m_Profile = 0
* Debug: dektec: m_IpProfile.m_MaxBitrate = 0
* Debug: dektec: m_IpProfile.m_MaxSkew = 0
* Debug: dektec: m_IpProfile.m_VideoStandard = 0
* Debug: dektec: }
* Debug: dektec: setting RxMode, mode: 17
* Debug: dektec: setting RxControl, mode: 1
* Error: dektec: synchronization lost after 0 packets, got 0x00 instead of 0x47
* Debug: dektec: data at loss of synchronization:
    0000:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0010:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0020:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0030:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0040:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0050:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0060:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0070:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0080:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0090:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00A0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00B0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00C0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00D0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00E0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    00F0:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0100:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0110:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0120:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0130:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0140:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0150:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
    0160:  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[... 25 lines cut ...]
tsp: Debug: ====> tsp -v -d --initial-input-packets 1000
* Debug: debug level set to 1
* Debug: ip: debug level set to 1
* Debug: ip: ====> tsp -I ip 51105
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -O dektec -d1 --io-standard asi
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: ip: setting socket reuse address to 1
* Debug: ip: binding socket to 0.0.0.0:51105
* Debug: ip: initial buffer load: 7 packets, 1,316 bytes
* ip: unknown initial input bitrate
* dektec: using Dektec device 1, output channel 0 (DTA-2174B port 2)
* Debug: dektec: attaching to device DTA-2174B serial 0x000000008197F765
* Debug: dektec: attaching to port 2
* Debug: dektec: resetting channel, mode: 1
* Debug: dektec: setting IO config of port 2, group: 1, value: 33, subvalue: -1
* Debug: dektec: setting TxMode, tx: 17, stuff: 0
* dektec: output fifo size: 8,388,608 bytes, max: 8,388,608 bytes, typical: 8,388,608 bytes
* Debug: dektec: setting TsRateBps using DtFractionInt: 0
* Debug: dektec: setting TsRateBps using DtFractionInt unsupported, using int: 0
* Error: dektec: output device set bitrate error: DTAPI_E_INVALID_RATE (DTAPI status 4109)
* Debug: start() error in output plugin dektec
* Debug: ip: aborting IP input
* Debug: Report logging thread terminated
```

lelegard, 2021-05-06:
Hi @izharm 

Thanks again for your tests. I don't know what's going on with reception. Maybe there is some other configuration to be set but this is not clear from the documentation. 

Could you try the output plugin on the DTA-2162 with the output traces? 

> as for ASI bitrate issue,I fugred it was an initail bit rate issue, however increasing initial packets didn't help:

Yes, I forgot the _exact_ behaviour of `--initial-input-packets`. In your case, you see the debug trace _"initial buffer load: 7 packets"_, so there are only 7 packets in the initial load (which will never be enough to evaluate the bitrate from PCR's). In fact, with `--initial-input-packets 1000`, `tsp` calls the input plugin with a buffer size of 1000 packets but will use what the input plugin returns. Input plugins such as `file` or `dvb` will indeed read as many packets as requested. The `ip` input plugin returns only whatever is immediately available (but at least one packet). In that case, the plugin only returns the 7 TS packets from the first UDP datagram. The input bitrate will rapidly evolves to some know value based on PCR's, for instance:

~~~
$ tsp -d --initial-input-packets 1000 -I ip 12345 -O drop
...
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
...
* Debug: ip: initial buffer load: 7 packets, 1,316 bytes
* ip: unknown initial input bitrate
...
* Debug: ip: input: got bitrate 4,771,098 b/s
* Debug: ip: input: got bitrate 4,658,353 b/s
* Debug: ip: input: got bitrate 4,489,739 b/s
...
~~~

But the configuration of the output ASI device occurs before outputting packets. So, it must use the very first input bitrate. In the case of an IP input, it will always be zero.

Anyway, an output ASI device must have a very precise bitrate. Even if the initial buffer load is sufficient to evaluate a bitrate from PCR's, it won't be precise enough for the ASI output and it is recommended to specify the intended output bitrate on the command line. Only precise hardware input devices such as ASI or demodulators can give a precise input bitrate, good enough to configure the ASI output.

izharm, 2021-05-06:
@lelegard 

the output seems ot be working, I didn't test the actual data but the application is running which looks promising.

as for the input, this is what we found out to be working when reading IP data:

first (only one time, outside the read loop) you need to wait until FIFO has enough data, then you read with timeout.
```

  int CurFifoLoad = 0;

  while (CurFifoLoad < BUFSIZE)
  {
    Inp.GetFifoLoad(CurFifoLoad);
	usleep(1000);
  }  

  for (int i=0; ; i++)
  {
	// read from dektec, with timeout
	dr = Inp.Read(pBuffer, m_bufsize, 100);
	if (dr != DTAPI_OK) 
		err_exit(dr);   

    // process data
	
  }

```

lelegard, 2021-05-06:
Thanks for the feedback on the output side. Let us when you can confirm that the transmitted data are correct.

On the input side, this is a bit strange.

That means that input can start only when the FIFO is full. This means that you have a latency which is equal to fifo-size x bitrate. This looks hardly compatible with high-precision / high-bitrate / low-latency objective of such devices.

Moreover, do we need to maintain a full FIFO? Typically, when the FIFO is initially full and we start reading, we will empty the FIFO quite rapidly since our processing is likely much faster than the input rate. So, once we have an empty FIFO, we are back in the original state. Do we have to wait again for a full FIFO? If not, why do we have to wait for a full FIFO the first time and not all subsequent times? It does not sound logical to me.

Did you verify that the code works that way? I can update the dektec input plugin to work that way but I have the feeling that their is some hidden issue or alternative more satistactory solution.

As a first short, I will add some option (not set by default) to preload the FIFO before receiving the first time. I will let you know when you can get the code. But I wanted to share my first impressions right now.

izharm, 2021-05-06:
this sequence works, I don't think the FIFO needs to be full, just contain enough data for the first read (with timeout) to return without time_out_error.  In fact it worked even when BUFSIZE is only 188,

lelegard, 2021-05-06:
I understand, this is due to the timeout. I did not pay attention to the timeout in your code sample. I agree that a careless reader which specifies a timeout and does not check the returned size from the Read() call may get the impression of having received zeroes (or garbage from the memory).

But this is not the case here. There is no timeout in the `dektec` input plugin. In fact, if the option `--receive-timeout` is used, there is a specified timeout. But this is an error timeout. The timeout condition is checked and the reception stops on error in case of timeout. So, this is quite different.

In your command, there is no option `--receive-timeout` and Read() is called without timeout, using the 2-parameter version of Read(), not the 3-parameter version as in your sample code. [See code here](https://github.com/tsduck/tsduck/blob/fc82531fd24a6e763765f96d717a11774a414fca/src/libtsduck/plugin/tsDektecInputPlugin.cpp#L932).

So, I do not think that waiting for the FIFO to be loaded is the reason of the problem here.

lelegard, 2021-05-06:
I have added two options in the `dektec` input plugin: `--preload-fifo` and `--fifo-size NNN`.

Given the previous posts, I do not think that this will change anything in this issue. But let's try. They could be useful in other situations anyway, who knows.

Could you please pull, rebuild and make two reception tests with debug traces, one without `--preload-fifo` and one with `--preload-fifo`. Please post the traces here.

I still doubt that `--preload-fifo` will make it work (no timeout is used). If it doesn't, I will ask Dektec support.

lelegard, 2021-05-09:
Hi @izharm 

Thanks again for your tests. This is weird. I acknowledge your test results but I don't understand the logic behind that behaviour.

I wonder what would happen if the source stops sending packets. At some point, a Read() operation is issued while the FIFO is empty. Does it fail like the initial read without `--preload-fifo` or does it wait for packets?

Generally speaking, without using a Read() timeout, it seems completely bizarre to me that the Read() operation on an empty FIFO returns zeroes. It should simply wait for packets. Calling Read() on an ASI or demodulator waits for packets, it does not return zeroes. 

I will ask Dektec support. I don't like the idea of leaving irrational code inside TSDuck without having tested the situation myself.

izharm, 2021-05-10:
I agree it is bizarre behavior, the only reason we implemented it this way (one wait for FIFO load, then read with timeout) is after a suggestion from DekTec to do it this way.

izharm, 2021-05-20:
@lelegard,

I think that when adding the support for the IP, something was broken in ASI support:

```
tsp -v -d -I dektec -d1 --io-standard asi -O drop
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -v -d
* Debug: debug level set to 1
* Debug: dektec: debug level set to 1
* Debug: dektec: ====> tsp -I dektec -d1 --io-standard asi
* Debug: drop: debug level set to 1
* Debug: drop: ====> tsp -O drop
* Error: dektec: inconsistent IP parameters, check IPv4 vs. IPv6 and number of links (single vs. redundant)
* Debug: getOptions() error in plugin dektec
* Debug: Report logging thread terminated

```
d1 is ASI card,
using ASI as output works.
tried with latest build (2393).

I'm sure this worked before,

lelegard, 2021-05-20:
Hi @izharm, you are right, sorry about that. The IP parameters were always checked, even if the device was not an IP one. This is the problem with working from home because of the pandemic and not having any Dektec device at home to test.

This is fixed now. You can pull the repo and rebuild (or wait for next nightly build, 3.27-2396 or higher).

izharm, 2021-05-23:
@lelegard,
No worries, I can confirm that build 2397 fixed the problem.  

on the down side, I needed to get the nightly build as I was unable rebuild myself:

```

root@XStream:/home/giorac/tsduck# make installer
make -C build installer
make -j32 NOTEST=true -C /home/giorac/tsduck
make[4]: *** No rule to make target '/home/giorac/tsduck/src/libtsduck/base/tsGuard.h', needed by '/home/giorac/tsduck/bin/release-x86_64-XStream/objs-libtsduck/tsAACDescriptor.o'.  Stop.
make[3]: *** [Makefile:43: default] Error 2
make[2]: *** [Makefile:59: default] Error 2
make[1]: *** [Makefile:142: deb-tools] Error 2
make: *** [Makefile:106: installer] Error 2

```

izharm, 2021-05-24:
> I agree it is bizarre behavior, the only reason we implemented it this way (one wait for FIFO load, then read with timeout) is after a suggestion from DekTec to do it this way.

any updates on this issue ?