tsduck/tsduck #772: [Q] Illegal descriptor at line X

przemo1983, 2021-04-28:
I am trying to write my own descriptor, but not as extension.
I am directly modifying the source code, and after the compilations I still get "illegal descriptor at line X" error when using tstabcomp tool with my descriptor.
I added my descriptor and related data to following files:
- tsduck.tables.model.xml
- tsduck.h
- tsduck.names
- tsRefType.h
- tsPSI.h
- and of course created descriptor.h and descriptor.c files according to other descriptors.

Could you please give me any advice when this error occurs?
Could you please provide a simple tutorial only for adding own descriptors?

BTW. Code from https://github.com/tsduck/tsduck/tree/master/sample/sample-extension does not work on windows with latest version. Does not compile.

lelegard, 2021-04-28:
Hi @przemo1983 

It is difficult to diagnose a problem without more information. Did you compile everything and built an installer and installed it? Are you sure that all configuration files were properly installed? Did you check their content?

Is your descriptor a "private descriptor" in the DVB sense, using a private data specifier (PDS)? Did you properly use the PDS, based on other private descriptors implementation?

You may of course modify TSDuck the way you want, this is the core of open source. However, if you add your own code or descriptors in the main source tree and do not want to contribute them back to the open source project, you will have to redo the merge with each new version of TSDuck. This can be your choice of course. But the extension mechanism was done to avoid that, precisely. Either your additions can be open sourced and included in the main project or you want to keep them private and develop an extension.

Note that tsduck.h and tsRefType.h are auto-generated files. They are automatically updated by `make` on Linux and macOS. On Windows, `build-project-files.ps1` must be run manually.

> Code from https://github.com/tsduck/tsduck/tree/master/sample/sample-extension does not work on windows with latest version. Does not compile.

I just tested it and it compiles. What is your issue? Which version are you using? Did you install a rebuilt TSDuck or a pre-built binary installer? Which version? Which version of the sample extension source tree with the which installed binary version?

przemo1983, 2021-04-28:
Sorry, I was wrong. Installed version without development extensions.
Thank you for your answer. It helped as lot.

closed 2021-04-28 by przemo1983