tsduck/tsduck #773: Emulation of a Conditional Access head-end

nikpakar, 2021-04-28:
Dear all,

Im trying emulate the CAS head-end as described in page 373 of the user guide.

Compiled without any issues
started tsecmg as below,
./tsecmg -v
* TCP server listening on 0.0.0.0:2222, using ECMG <=> SCS protocol version 2
* 127.0.0.1:36092: 2021/04/28 15:01:03: session started

And running the tsp command in a separate window as below,
./tsp -v -I file sample.ts -P scrambler Service01 --ecmg localhost:2222 --super-cas-id 0xDEADBEEF --access-criteria 0123456789 -P analyze --interval 30 -o cas_>
* file: initial input bitrate is 2,280,143 b/s
* play: using media player command: "/usr/bin/vlc" - --play-and-exit
* scrambler: found service "Service01", service id is 0x0001 (1)
* scrambler: found service id 0x0001 (1), PMT PID is 0x1000 (4096)
* scrambler: starting scrambling PID 0x0100
* scrambler: starting scrambling PID 0x0101
* scrambler: using PID 4097 (0x1001) for ECM
VLC media player 3.0.9.2 Vetinari (revision 3.0.9.2-0-gd4c1aefe4d)
[0000560e665ea630] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
* descrambler: found service "Service01", service id is 0x0001 (1)
* descrambler: found service id 0x0001 (1), PMT PID is 0x1000 (4096)
* descrambler: using ECM PID 4097 (0x1001)
* descrambler: using scrambling mode: DVB-CSA2
* zap: found service "Service01", 0x0001 (1)
* zap: found service id 0x0001, PMT PID is 0x1000
[00007f7280026360] main decoder error: buffer deadlock prevented
[00007f7280044520] main decoder error: buffer deadlock prevented

Results
vlc player opens, counter keeps running but video is black
tsecmg screen shows a session connected, but no further ECM activities

./tsecmg -v
* TCP server listening on 0.0.0.0:2222, using ECMG <=> SCS protocol version 2
* 127.0.0.1:36092: 2021/04/28 15:01:03: session started
* 127.0.0.1:36092: 2021/04/28 15:01:45: session completed

On cas_scrambler.txt i can see the ECM pid as DEAD
===============================================================================
|  SERVICES ANALYSIS REPORT                                                   |
|=============================================================================|
|  Global PID's                                                               |
|  TS packets: 248, PID's: 2 (clear: 2, scrambled: 0)                         |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Global PID's ................................. C       23,325 b/s  |
|   Subt.  Global PSI/SI PID's (0x00-0x1F) .............. C       23,325 b/s  |
|  0x0000  PAT .......................................... C       18,717 b/s  |
|  0x0011  SDT/BAT ...................................... C        4,608 b/s  |
|=============================================================================|
|  Service: 0x0001 (1), TS: 0x0001 (1), Original Netw: 0xFF01 (65281)         |
|  Service name: Service01, provider: FFmpeg                                  |
|  Service type: 0x01 (Digital television service)                            |
|  TS packets: 20,752, PID's: 4 (clear: 2, scrambled: 2)                      |
|  PMT PID: 0x1000 (4096), PCR PID: 0x0100 (256)                              |
|-----------------------------------------------------------------------------|
|     PID  Usage                                     Access          Bitrate  |
|   Total  Digital television service ................... S    1,951,842 b/s  |
|  0x0100  MPEG-2 Video ................................. S    1,303,987 b/s  |
|  0x0101  MPEG-1 Audio (eng) ........................... S      629,138 b/s  |
|  0x1000  PMT .......................................... C       18,717 b/s  |
|  0x1001  unknown (0xDEAD) ECM ......................... C            0 b/s  |
|          (C=Clear, S=Scrambled, +=Shared)                                   |
===============================================================================

** If i avoid the scramber plugin from tsp command, i can obviously see the video playing on vlc.

Would highly appreciate if some one help me to identify the issue here.

Thanks in advance
-Nik

lelegard, 2021-04-28:
Did you check the TSDuck user's guide, chapter 6 "Troubleshooting", section 6.2 "Typical issues with tsp", topic "A plugin which is supposed to insert some data in the stream does not insert anything"?

nikpakar, 2021-04-28:
Hi Thierry, many thanks for the reply. I tried to go through and understand that section and tried to add input stuffing as a global parameter to the tsp. But gives an error

./tsp -v -a nullpkt -I file sample.ts -P scrambler Service01 --ecmg localhost:2222 --super-cas-id 0xDEADBEEF --access-criteria 0123456789 -P analyze --interval 30 -o cas_scrambled.txt -P descrambler Service01 -P analyze --interval 30 -o cas_descrambled.txt -P zap Service01 -O play
tsp: invalid value for --add-input-stuffing, use "nullpkt/inpkt" format

nikpakar, 2021-04-28:
Sorry about the above reply. My fault in stuffing option. it works when i add 14/24. Thanks a lot.

Now i can see the video
activity in tsp windows
but there is no activity in tsecmg.
 
i have started it with -v as i wanted to see ECMs.

lelegard, 2021-04-28:
> Now i can see the video

So, the ECM are there. And if they are there, they must have been generated, meaning that there were actitity in the ECMG.

> but there is no activity in tsecmg.
> i have started it with -v as i wanted to see ECMs.

Honestly, do you _ever_ read the documentation? Do you try the various proposed options? Specifically `--log-protocol` and `--log-data`.

People developing open source software spend hours developing code and writing documentation _for free_. The least we expect from users taking advantage of this free work is to show some basic respect, starting by reading the documentation, making the effort to understand it and the associated standards before asking even more free time from the developers. I mean it.

nikpakar, 2021-04-29:
Hi Thierry, im really sorry for been blunt on this. Honestly i was following the documentation but since --verbose was on never thought i should further enable options. And it works. Fantastic and thanks a lot for the great work you have done.

Extremely sorry again.

closed 2021-04-29 by nikpakar