tsduck/tsduck #774: [Q] "tsp --verbose -I ip" returns "ip: unknown initial input bitrate": Is there a way to force the initial input bitrate if/when unknown?

moving-digital, 2021-04-30:
Hello @lelegard 

I am using the following command: `tsp --verbose -I ip` followed by the multicast IP,

But it returns `ip: unknown initial input bitrate`. 

Is there a way to force the initial input bitrate if/when unknown?

lelegard, 2021-04-30:
The answers to many questions can be summarised in four letters: RTFM 😉 
The option `tsp --bitrate NNN` does this:
> __-b__  _value_
> __--bitrate__ _value_
>Specify the input bitrate, in bits/seconds. By default, the input bitrate is provided by the input plugin or by analysis of the PCR’s at the beginning of the input stream. If no or not enough PCR are found, the DTS from video PID’s are used.
> Bitrate values may include up to 1 meaningful decimal digits.
> Use option --bitrate when you know precisely the input bitrate and you do not trust the input device, the PCR’s or the DTS.

moving-digital, 2021-05-01:
My bad... Thanks !

closed 2021-05-01 by moving-digital