tsduck/tsduck #775: [BUG] Unable to inject a SCTE-35 splice when not creating a TS from scratch

moving-digital, 2021-05-01:
Hi @lelegard 

#### Bug description:
I'm unable to inject a SCTE-35 SpliceInsert() Normal (= not immediate) when extracting the current PTS of a TS stream containing a video+audio.
However, I know that the PTS value I am using for the splice is correct, and is in the future. I tried with different PTS value in the future: +4s, +6s, +10s, +20s, +30s. 

However, the result is still the same: TSDuck will never inject the splice in the TS, and I will never be able to get the below message (Example of command, please disregard the values here):
`spliceinject: injecting SpliceInsert out @0xFFFFFFFFFFFFFFFF, immediate, current PTS: 0x000045CB8`

#### How to reproduce:
I am using 2 TSP commands:

1. The first one is used to set the TSP command generating the UDP TS for SCTE-35, while re-using the existing PTS value from the existing TS stream (And throwing the audio+video to only keep the SCTE-35 PID)
```
tsp --verbose --bitrate 1000000 --max-flushed-packets 7 " +
"-I ip 239.111.111.111:50011 " +
"-P inject " + msg.watchfolder + "pat.xml --pid 0 --replace --bitrate 15000 " + 
"-P inject " + msg.watchfolder + "pmt.xml --pid 100 --replace --bitrate 15000 " + 
"-P spliceinject --udp " + msg.scte35_udp_port + " --service 101 --delete-files " + 
"-O ip 225.5.5.5:12345 --enforce-burst"
```

2. The second command is used only when the injection of the SCTE-35 spliceInsert() Normal is requested, in order to extract the current PTS value:
tsp --bitrate 1000000 " +
```
"-I ip 239.111.111.111:50011 " +
"-P pcrextract " +
"-O drop";
```

You'll find below two use-cases:

Use-case #1: PTS value in the future

When extracting the PTS value from the existing TS stream (Either the "original" one containing the video+audio, or the one generated by TSDuck), I am getting the same PTS value, which is the following:
```
2110,145,63,PCR,1,1111971857085,0,
2110,147,64,PTS,1,3706722096,0,
2110,147,64,DTS,1,3706704096,0,
2111,169,3,PTS,1,3706578141,0,
2110,571,252,PCR,2,1111972722056,864971,
2111,604,12,PTS,2,3706581021,2880,
2110,997,440,PCR,3,1111973587022,1729937,
2111,1024,21,PTS,3,3706583901,5760,
2113,1248,6,PTS,1,3706590806,0,
2110,1307,575,PCR,4,1111974216461,2359376,
2110,1309,576,PTS,2,3706714896,18446744073709544416,
```

The generated SCTE-35 XML contains the below information:
```
splice_event_id: "0x00000001"
splice_event_cancel: "false"
out_of_network: "true"
splice_immediate: "false"
pts_time: 3710837661
unique_program_id: "0x0001"
avail_num: "0"
avails_expected: "0"
```
As we can see, the PTS value of the XML is in the future:

3710837661-3706714896=4,122,765
4,122,765 at 90Khz = 45sec in the future

However, TSDuck will not inject the SCTE-35 Splice. It will receive, and enqueue, but that's it:
```
spliceinject: received message, 474 bytes, from 127.0.0.1:36754
spliceinject: enqueuing SpliceInsert out @0x0DD2EE39D, start: 0x0DD2C247D, end: 0x0DD2EE39D, 2 times
```

I have performed another test with PTS value in the past on purpose (-4s, -6s, -10s, -20s, -30s), and the result is the same. On top of it, TSDuck does not throw the error that the PTS is in the past. 

The current TS stream (Either the one containing the video+audio, or the one generated by TSDuck) has the following PTS value:
```
2111,138,2,PTS,1,3723184221,0,
2110,295,130,PCR,1,1116954059452,0,
2110,390,174,PCR,2,1116954252345,192893,
2110,390,174,PTS,1,3723339696,0,158855
2110,390,174,DTS,1,3723321696,0,140855
2111,560,11,PTS,2,3723187101,2880,
2110,718,318,PCR,3,1116954918327,858875,
```

And the generated SCTE-35 XML file has the below PTS value (in the past):
splice_event_id: "0x00000001"
splice_event_cancel: "false"
out_of_network: "true"
splice_immediate: "false"
pts_time: 3714054621
unique_program_id: "0x0001"
avail_num: "0"

The PTS time I am trying to use is in the past (on purpose), yet TSDuck does not throw any error, and stays stuck at "enqueuing":

* spliceinject: received message, 474 bytes, from 127.0.0.1:58706
* spliceinject: enqueuing SpliceInsert out @0x0DD5FF9DD, start: 0x0DD5D3ABD, end: 0x0DD5FF9DD, 2 times

TSDuck should throw the error that it is throwing the SCTE-35 since the PTS value is obsolete.

The PAT XML is:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
	<PAT version="0" current="true" transport_stream_id="0x0000">
		<metadata PID="0"/>
		<service service_id="101" program_map_PID="100"/>
	</PAT>
</tsduck>
```

The PMT XML is:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <PMT version="0" current="true" service_id="101" PCR_PID="102">
    <metadata PID="100"/>
    <registration_descriptor format_identifier="0x43554549"/>
    <component elementary_PID="102" stream_type="0x86">
      <registration_descriptor format_identifier="0x43554549"/>
    </component>
  </PMT>
</tsduck>
```

#### Expected behavior:
TSDuck shall be able to inject the SCTE-35 SpliceInsert() Normal (I.e. with PTS value, since not Immediate) since the PTS value is valid, and is in the future.

#### Environment:
 - OS: Linux
 - OS version: Raspbian 10
 - TSDuck full version: tsduck_3.26-2349.raspbian10_armhf.deb
 - Installation type: Official binary

lelegard, 2021-05-01:
Did you check the TSDuck user's guide, chapter 6 _"Troubleshooting"_, section 6.2 _"Typical issues with tsp"_, topic _"A plugin which is supposed to insert some data in the stream does not insert anything"_?

Which can be translated into: does your input TS have some stuffing to fill?

A TS is like a file system, you need free space to overwrite with new files. Free space in a TS is made of null packets. Creating a TS from scratch is like formatting a disk, you create a lot of free space. At the other end, trying to create new files in a file system which is already full fails (or inserting sections in a TS). With TSDuck, you can "enlarge" a full TS with null packets (see the topic above), thus creating free space at regular intervals.

moving-digital, 2021-05-01:
Hi @lelegard 

I have tried with the below command setting TSP command generating the UDP TS for SCTE-35:

```
msg.payload=" --verbose --bitrate " + bitrate + "  --add-input-stuffing 1/50 --max-flushed-packets 7 " +
"-I ip 239.111.111.111:50011 " +
"-P inject " + msg.watchfolder + "pat.xml --pid 0 --replace --bitrate 15000 " + 
"-P inject " + msg.watchfolder + "pmt.xml --pid 100 --replace --bitrate 15000 " + 
"-P spliceinject --udp " + msg.scte35_udp_port + " --service 101 --delete-files " + 
"-O ip " + msg.ts_udp_out + " --enforce-burst";
```

As you can see, I have added `--add-input-stuffing 1/50` and I now have 10 Mbps of null packets (0x1fff) and yet, the SCTE-35 SpliceInsert() Normal (Type 5 with PTS) does not work.

I even tried with `--add-input-stuffing 1/10` which now gives 11 Mbps of null packets. Yet the SCTE-35 cannot be inserted.

I ended with `--add-input-stuffing 14/24` which now gives 13 Mbps of null packets. Yet the SCTE-35 cannot be inserted either.

PS: I know a single 1 Mbps should work so, I believe something is not working as expected.

I've tried many things but to be honest, I don't know how to fix this. Do you have any idea?

lelegard, 2021-05-01:
There is something I do not understand in your command. Why do you replace the PAT and PMT of the existing input TS?

When you tested empty TS, created from scratch using a stream of null packets, there was no PSI/SI and you needed to create the PAT and PMT using XML files. But in the present case, the PAT and PMT already exist. You simply need to add one component in the PMT of the service (see the example in section 5.2.24 of the user's guide). At best, completely overwriting the PAT and PMT may work if you are 100% sure that the structure and PIDs of the PAT and PMT are identical to your XML file. At worst, in case of mismatch, the structure of the TS is completely corrupted.

Additionally, when you add input stuffing to get free space, don't forget to remove the unused null packets. This is not required but recommended in IP VBR TS. Add `-P filter -n -p 0x1FFF` at the end of the chain, just before `-O`.

> If you want, I can send you the original TS capture I am working with, for troubleshoot purpose.

Yes, that may help. Please also provide a couple of SCTE 35 that you expect to be inserted here.

moving-digital, 2021-05-01:
Hi @lelegard 

I can't share drawings here, but the idea is the following:
- I'm trying to find a but to insert frame accurate OOB (Out of Band) SCTE-35 SpliceInsert() markers. It could be SpliceInsert() Normal (Based on PTS time) or Schedule (Based on UTC time).
- However, "Schedule" only provides 1 second accuracy, which is far from the expected frame accuracy. So, "Normal" is the way to go.
- The encoder I'm using can remux external data PID, meaning there can be 2 inputs: "Main" one being video+audio, and the second one that I could use to generate the SCTE-35 message, based on the PTS on the Main input (aka input 1).
- However this main input #1 does not contain any SCTE-35 markers. I need to create a new PAT/PMT containing it, but still having the PTS on the main input #1.

My idea was to make the input #2 only containing a basic PAT/PMT + SCTE-35 PID, in order to to duplicate unnecessary info (video+audio). 
I don't have any bitrate constraint so I don't need to delete data in order to insert SCTE-35 info. Do you think it is wiser/easier/better to simply add an SCTE-35 PID & descriptor in the existing PMT?

lelegard, 2021-05-01:
OK, I understand now. I cannot work that way. Neither with TSDuck nor any other SCTE 35 inserter, most probably.

The `spliceinject` plugin is based on the PTS of the audio/video PID's. You cannot inject `SpliceInsert()` based on PTS time when there is no PTS in the stream. A `SpliceInsert()` shall be inserted, let's say, 6, 4 and 2 seconds before the PTS of the event (depending on your options). How could the `spliceinject` plugin know when to insert the tables if there is no clue about the PTS reference? The `spliceinject` plugin needs the presence of the audio/video PID's of the service to be constantly aware of the PTS clock reference.

Generating the SCTE 35 stream independently of the audio/video of the service and remuxing them later is impossible to achieve (unless the encoder and the SCTE 35 inserter are provided by the same vendor and fully synchronised in some way, which is not the case here).

You must encode audio/video first and then run the output of the encoder into `tsp` where the SCTE 35 will be added: encoder --> tsp --> output

Inside tsp: add a few kb/s of stuffing --> update the PMT to add the SCTE 35 PID -> inject SCTE 35 -> remove unused stuffing.

There is no need to "duplicate" the audio video. The actions shall be serialised, not parallelised.

moving-digital, 2021-05-01:
Hi @lelegard 

I've just followed your advice on page 389 on the user guide, section **5.2.24.1 Real-time live stream**. The newly created command is:

```
msg.payload=" --verbose --bitrate " + bitrate + " --max-flushed-packets 7 " +
"-I ip 239.111.111.111:50011 " +
"-P pmt --service 111 --add-programinfo-id 0x43554549 --add-pid 600/0x86 " + 
"-P spliceinject --service 111 --udp " + msg.scte35_udp_port + " --delete-files " + 
"-O ip " + msg.ts_udp_out + " --enforce-burst";
```

I'm not modifying anything anymore. Simply inserting cue information in a real-time live transport stream.

Yet, when I try to inject SCTE-35 spliceInsert() "Normal" based on future PTS, it does not work, even if the action is serialized.

I'm still only getting the below verbose:
```
spliceinject: received message, 474 bytes, from 127.0.0.1:46437
spliceinject: enqueuing SpliceInsert out @0x0DE1280FD, start: 0x0DE0FC1DD, end: 0x0DE1280FD, 2 times
```

Anything I am doing wrong?

lelegard, 2021-05-01:
Your command no longer add input stuffing.

moving-digital, 2021-05-01:
Hi @lelegard 

Just added the stuffing, same issue:

```
msg.payload=" --verbose --bitrate " + bitrate + " --add-input-stuffing 1/50 --max-flushed-packets 7 " +
"-I ip 239.111.111.111:50011 " +
"-P pmt --service 111 --add-programinfo-id 0x43554549 --add-pid 600/0x86 " + 
"-P spliceinject --service 111 --udp " + msg.scte35_udp_port + " --delete-files " + 
"-O ip " + msg.ts_udp_out + " --enforce-burst";
```

I've also tried with:
--add-input-stuffing 1/10
--add-input-stuffing 14/24

Even without adding stuffing with TSDuck, my original TS got 8 Mbps of stuffing, so adding stuffing with TSDuck may not be the solution.

Any other idea?

lelegard, 2021-05-01:
> Even without adding stuffing with TSDuck, my original TS got 8 Mbps of stuffing, so adding stuffing with TSDuck may not be the solution.

OK, if there is already stuffing in the input stream, there is no need to add any. I thought it was a VBR SPTS. SCTE 35 requires only a ridiculously low bitrate. It just needs a few null packets per second to be safe. In fact, it just needs one null packet each time a SCTE 35 table shall be inserted.

Please provide a sample input TS and one splice insert that you expect to be inserted inside it. It works for everyone, there must be something silly here.

moving-digital, 2021-05-01:
Hi @lelegard 

The PTS was too much in the future, and the injecting only happened 30sec after the enqueuing... Seems I'm too impatient, haha!

Your software rocks ;-)

closed 2021-05-01 by moving-digital