tsduck/tsduck #776: [Q] Linux DVB API: Why use 'dvrX' devices for Capturing?

SS-FF, 2021-05-03:
### Background

As someone who hasn't been very familiar with the Linux DVB API, I've always been wondering about the purpose of those 'dvrX' devices. Just now I had taken a closer look at the Linux side.
After all the praise for Linux/DVB over Windows/BDA, I had really expected some enlightening insights from the documentation, but even though BDA docs are questionable, the Linux docs are much worse than that: Zero answers to elementary questions and content half up- half-outdated - that didn't get me very far.
What's surely better at the Linux side is the availability of source code and looking at the code from various projects created at different points in time, helped answer many questions - one of them is this.

### The dvrX Device

Now - at least from my understanding - the primary purpose of those 'dvrX' devices is this:

- You want to play or analyze a TS recording on disk
- You want this to be accelerated and handled by your DVB hardware
   using one or more of the hw devices
  - hardware demuxer
  - hardware audio decoder
  - hardware video decoder
    e.g. `/dev/dvb/adapter0/video0`
- And that works in a way that you
  - Open a dvr device
  - Open a demux device
  - Set the input of the demux device to the dvr device
  - Configure multiple outputs on the demux to the appropriate targets
    (by calling `::open` multiple times on the demux and set different different PES filters and targets)
- Finally: You read the recorded TS data from disk and write it to the dvr device

That makes a lot of sense to me. The dvr device is an entry point for feeding TS data into your hardware that is not coming "live" from the tuner (I mean, it makes sense in a logical way, obviously that's more or less pointless these days).

Secondly, the dvr device can also work in the other direction:

- You tune to a frequency
- Set a demux to get input from a frontend device and output to a dvr device
  what doesn't make sense to me here: you can only filter for a single PID or all PIDs with that setup
  I'm not sure how this is intended to be used for recording a single channel from a mux..
- You read from the dvr device and save it to a file (or do whatever)

### Question

For more than 13 years ago - since Linux kernel 2.6.26 - it is possible to read directly from a demux device without needing to open a corresponding dvr device and let the demux copy it over there before retrieving.

Is there any reason or advantage of going through that additional device?
Or: why does tsDuck do it like this?

lelegard, 2021-05-03:
Hi @SS-FF 

> For more than 13 years ago - since Linux kernel 2.6.26 - it is possible to read directly from a demux device without needing to open a corresponding dvr device and let the demux copy it over there [...]

In the case of TSDuck, this is purely historical. The very first version of what later became TSDuck started in 2005 (16 years ago), was written in C (instead of C++) and was Linux only. The first input/output plugins I needed were TS files, Linux DVB tuners and Dektec ASI cards. At that time, going through the dvr devices was the way to go. I admit that I never challenged this later.

The only fundamental changes in Linux DVB I was involved in later was the "S2 API war", what later became the DVB API V5, and "new" modulations such as ISDB. Otherwise, apart from code structure refactoring, the basic code which calls Linux DVB did not change much in TSDuck.

It is true that, at the beginning, I was a bit surprised by the dvr device. I understood it, as you just did, in a more general context of hardware demux. I do not know if there is a good reason for changing the TSDuck code now. Does it bring any value? Isn't there a risk of breaking the support for some tuners?

SS-FF, 2021-05-03:
> I do not know if there is a good reason for changing the TSDuck code now. Does it bring any value? Isn't there a risk of breaking the support for some tuners?

From the very first glance - I wouldn't expect that. The option to read ts packets from demux devices exists for along time and is something pretty basic and it's more direct. But there's always a risk..

That's why I have added another option `--read-from-demux` which allows to switch between both variants, so I'll be able to switch back quickly in case it would cause any problems.

My motivations for changing this are two-fold:

1. The same relation-problem that I've described here https://github.com/tsduck/tsduck/issues/777 also exists between demux and dvr and getting rid of the dvr device simplifies the whole story of course
2. I have implemented hardware PID filtering for both Windows BDA and Linux and in case of Linux, this only works when reading from the demux (setting filter with `DMX_OUT_TSDEMUX_TAP`) but not when going through the dvr device (`DMX_OUT_TSTAP`)

(Regarding the hw PID filtering, you had said earlier that you can't see any legitimate use cases and that tsDuck is intended to work on a full mux. For me, there's nothing to argue about the latter, but I do have a list of serious and legitimate uses cases...)

SS-FF, 2021-05-04:
Here's my list of use cases for 

### Hardware PID filtering Use Cases

- Network Tuners
  We have quite some users with tuners from Silicon Dust which are network tuners that come with a BDA driver (for Windows).  Some have two or three of them, each with 4 tuners. Now imagine you need to stream 3x4 = 12 full muxes DVB-C/T/ATSC over the (local) network via UDP just to arrive at the BDA driver's interface. 
  This is where it's very useful, when you're able to choose at the hardware level what is being sent and what can be filtered out in the first place
- High Density Configurations
  Building streaming or testing boxes where you'd use more than just a handful of tuners (assume a DVB-S2 mux for each), this could even get critical to handle for PCIex or other buses. So, even there, one might be thankful when it would be possible to reduce the volume of data that needs to be processed before it even hits the ground
- Low Profile Systems
  No matter how fast and effective tsDuck actually is - data that doesn't even happen to exist is always more efficient than otherwise
- Legacy DVB Devices that don't support 0x2000
  Those devices exist, even though it's a minority without doubt

PID filtering at the hardware level is something that I had on my list for quite some time due to the first bullet (Network Tuners). Transferring several full muxes over the network is not very efficient when you actually just need a small subset of the data.
I wanted to do this for Windows/BDA only which is where I know how it works. The case about those Linux tuners not supporting unfiltered muxes just came by accident, but it was a good occasion to get implemented.

Same like in the other conversation: This is not a feature request, just an offer. If you would want it, I can share it with you.

xenpac, 2022-01-26:
i am also frustrated with the DVB docu to understand.

well each APADPTER has:
FRONTEND-> reads/writes to the frontend, tune etc, get status
DVR-> should be the TS output from the demodulator to be fed into the soft-demuxer.
DEMUX->? ja, is that a hardware demuxer, or a kernel software demuxer?
NET->network io, to stream a TS to the network?

lelegard, 2022-01-26:
Hi @xenpac,

Is this a general Linux DVB question or a question about TSDuck accessing tuners on Linux?

TSDuck processes a complete TS. Its way of getting a TS is always grabbing the full TS from the tuner. So, the really important stuff is the frontend (tuner and demod). The demux is always used transparently (all PID's). The DVR is just a route to extract the TS from the kernel. The NET is unused.

- In the vast majority of tuners, PCI or USB on PC, the frontend, demux and dvb numbers are identical. Just specify the frontend number.
- In some very specific cases, typically massively parallel tuners in embedded systems, having a combination of different numbers is necessary. This was not possible in older versions of TSDuck but was fixed to address those devices.
- In a very limited number of cases, typically some low-end embedded tuners in set-top boxes, getting the complete TS is not possible (demux 0x0200 not working). The tuner is only able to extract a limited number of PID's. These devices are not supported by TSDuck and will never be. This type of device is not the TSDuck target.

lars18th, 2022-01-28:
Hi @lelegard ,

> * In a very limited number of cases, typically some low-end embedded tuners in set-top boxes, getting the complete TS is not possible (demux 0x0200 not working). The tuner is only able to extract a [...]

With all respect, please consider a fourth case:
- Network tuners (HDHomeRuns and SAT>IP servers). Typically they could provide the full transport stream. But in some cases this is not true: because the source has low performance (SAT>IP LNB's and STB's) or becase the network bandwith is limited.

In my particullar use case I use (for years) **only** external network tuners. I hate the use of PCIe or USB tuners. I prefer to use _only_ network sources that are platform agnostic (RTP multicast, RTSP o any other protocol). And mixing such tuners with TSDuck it's easy, if you can enable the Full Transport Stream. However, even when you can't do that, the problem can be overcomed. For example, you can convert the received stream to CBR with null padding. So to do transformations you have the necessary NULL stuff and all work as expected. Except if you need to add some missing pid.

So anyway, a partial support for sources with internal pid filtering but lack of full transport stream, seems to have sense.
I say this just to share my point of view in a constructive spirit.
Regards.