tsduck/tsduck #777: [Q] Linux DVB API: How to Relate frontend/demux/dvr Devices?

SS-FF, 2021-05-03:
same background as this question: https://github.com/tsduck/tsduck/issues/776

There are Linux DVB devices that don't have an equal number of the primary three device types:

- frontend
- demux
- dvr

I've posted an example for this a while ago: https://github.com/tsduck/tsduck/issues/596#issuecomment-643098731

And here's another example I recently came across:

```
drwxr-xr-x    2 root     root           620 Jan  1  1970 .
drwxr-xr-x    3 root     root            60 Jan  1  1970 ..
crw-rw----    1 root     root      212,  28 Jan  1  1970 audio0
crw-rw----    1 root     root      212,   5 Jan  1  1970 ca0
crw-rw----    1 root     root      212,   8 Jan  1  1970 ca1
crw-rw----    1 root     root      212,  11 Jan  1  1970 ca2
crw-rw----    1 root     root      212,  14 Jan  1  1970 ca3
crw-rw----    1 root     root      212,  17 Jan  1  1970 ca4
crw-rw----    1 root     root      212,  20 Jan  1  1970 ca5
crw-rw----    1 root     root      212,  23 Jan  1  1970 ca6
crw-rw----    1 root     root      212,  26 Jan  1  1970 ca7
crw-rw----    1 root     root      212,   3 Jan  1  1970 demux0
crw-rw----    1 root     root      212,   6 Jan  1  1970 demux1
crw-rw----    1 root     root      212,   9 Jan  1  1970 demux2
crw-rw----    1 root     root      212,  12 Jan  1  1970 demux3
crw-rw----    1 root     root      212,  15 Jan  1  1970 demux4
crw-rw----    1 root     root      212,  18 Jan  1  1970 demux5
crw-rw----    1 root     root      212,  21 Jan  1  1970 demux6
crw-rw----    1 root     root      212,  24 Jan  1  1970 demux7
crw-rw----    1 root     root      212,   4 Jan  1  1970 dvr0
crw-rw----    1 root     root      212,   7 Jan  1  1970 dvr1
crw-rw----    1 root     root      212,  10 Jan  1  1970 dvr2
crw-rw----    1 root     root      212,  13 Jan  1  1970 dvr3
crw-rw----    1 root     root      212,  16 Jan  1  1970 dvr4
crw-rw----    1 root     root      212,  19 Jan  1  1970 dvr5
crw-rw----    1 root     root      212,  22 Jan  1  1970 dvr6
crw-rw----    1 root     root      212,  25 Jan  1  1970 dvr7
crw-rw----    1 root     root      212,   0 Jan  1  1970 frontend0
crw-rw----    1 root     root      212,   1 Jan  1  1970 frontend1
crw-rw----    1 root     root      212,   2 Jan  1  1970 frontend2
crw-rw----    1 root     root      212,  27 Jan  1  1970 video0
```

I assume there is no way - at least I couldn't find any - to "detect" which demux is associated with which frontend?

Actually, it seems that on the hardware above, the demux devices can be arbitrarily assigned to the frontends.
I came to that, because the current logic  assumes: 

**frontendN => demuxN => dvrN** 

but that (e.g. `/dev/dvb/adapter0:1:1:1`) didn't work. 

I found an ::ioctl that had recently been removed from the kernel: `DMX_SET_SOURCE`. I tried that and it works in a way that you can set the frontend number for a specific demux device.

I've added an option `--set-demux-source` that executed that when specified.
If you wish, I could provide a PR with that.

lelegard, 2021-05-03:
Hi @SS-FF,

This ioctl disappeared in kernel 4.14. The current kernel is 5.12. It seems hard to start using this ioctl in 2021.

In 2017, there was a discussion here about `DMX_SET_SOURCE`:
https://www.mail-archive.com/linux-media@vger.kernel.org/msg118071.html

Mauro Carvalho Chehab, the Linux DVB project leader, wrote:
> No driver uses this ioctl, nor it is documented anywhere.
> So, get rid of it.

Reading the discussion thread, I guess that:
- The default is frontendX -> dmxX -> dvrX (same X)
- This ioctl was maybe defined to tell a tuner driver to change the routing between frontend, dmx and dvr but noone is assertive about it since it has never been documented or used.
- No tuner driver in the mainstream kernel tree uses that ioctl.

Given that this ioctl was undocumented, uncleared and unused, Mauro decided to remove it.

It is still possible that some third-party tuner driver, _not present in the mainstream kernel_, uses it, relies on it and does not implement the default frontendX -> dmxX -> dvrX routing. Could that be the case for your tuner?

If this is the case, then this driver must have been changed when `DMX_SET_SOURCE` disappeared. It could probably not compile anymore in kernel 4.14 and later.

SS-FF, 2021-05-03:
> In 2017, there was a discussion here about `DMX_SET_SOURCE`:
> https://www.mail-archive.com/linux-media@vger.kernel.org/msg118071.html

Thanks for the link, I hadn't seen that discussion.

I mean, the discussion ends with Mauro saying 

> Yeah, I agree, but it is too late to change it ;)

But they haven't actually hit the point. They talked about it being not nice to use the DMX_SET_PES_FILTER ioctl for choosing the source, but they didn't even cover the case where multiple demux devices exist that do not have a fixed association to frontend devices.

> It is still possible that some third-party tuner driver, not present in the mainstream kernel, uses it, relies on it and does not implement the default frontendX -> dmxX -> dvrX routing. Could that be the case for your tuner?

Yes, that exactly the case (as shown above, they have 8 demux devices for 3 tuners). 

I have added a define for the removed ioctl in tsTunerGuts:

```cpp
#ifndef DMX_SET_SOURCE
    #define DMX_SET_SOURCE _IOW('o', 49, int)
#endif
```

and added a tuner option --set-demux-source which does this after opening the demux device:

```cpp
    if (_guts->set_demux_source) {
        if (::ioctl(_guts->demux_fd, ioctl_request_t(DMX_SET_SOURCE), &frontend_nb) < 0) {
            report.error(u"error executing DMX_SET_SOURCE to %d on %s: %s", {frontend_nb, _guts->demux_name, SysErrorCodeMessage()});
        }
    }
```

I'm not sure whether that's too "edgy" for tsDuck. It's just an offer - I won't feel bad when you don't want it ;-)

lelegard, 2021-05-03:
> they didn't even cover the case where multiple demux devices exist that do not have a fixed association to frontend devices.

Yes, this seems to be a basic assumption of all Linux DVB drivers. So far, for multi-frontend devices, the route frontendX -> dmxX -> dvrX has always worked. Or at least I did not receive any negative feedback until yours.

The basic constraint is the number of frontends, they are always hardware resources. Then, for test and demos systems (the target for the TSDuck toolbox), one demux per frontend is sufficient.

Having more demux than frontends is typically a set-top box configuration (live plus multiple recordings from the same source). This is not a TSDuck target. Having _much more_ demux than frontends is a usual characteristics of devices with suspicious purposes...

> Yes, that exactly the case (as shown above, they have 8 demux devices for 3 tuners).

How does this driver integrate in kernel 4.14 and later?

SS-FF, 2021-05-03:
> How does this driver integrate in kernel 4.14 and later?

I just checked: the testing device I have access to is 4.10.12

> Having _much more_ demux than frontends is a usual characteristics of devices with suspicious purposes...

That's not necessarily a correct assumption. The device I have access to is from a UK user, configured for DVB-T reception (which is not encrypted: FreeView UK) and he wants to use this like a hardware tuner with our software. From the point where I'm grabbing the data from the DVB API, it's all under my control, so nothing 'suspicious' is happening there. 😉

SS-FF, 2021-05-03:
> Having more demux than frontends is typically a set-top box configuration (live plus multiple recordings from the same source).

Doing multiple recordings from the same source is a core expertise of tsDuck - so I wouldn't know what's wrong about that.

For me, those many demux devices are rather an indication that this is a pretty outdated design from times, where you needed to have hardware acceleration for demuxing, which is kind of ridiculous nowadays, even for devices with limited processing capabilities..

lelegard, 2021-05-03:
> I just checked: the testing device I have access to is 4.10.12

OK, so this is a very specific tuner, using an old kernel, with a third-party Linux driver (i.e. not in the mainstream kernel), not even compatible with recent versions of the kernel. This does not make the proposed patch an attractive candidate for TSDuck code 😉 

If this kind of device wants to survive after 4.13, the driver should at least integrate modifications such as:
- By default, associate frontendX -> dmxX -> dvrX (for the same X) so that regular Linux applications can use it.
- Implement a "proprietary" ioctl (perhaps just the old `DMX_SET_SOURCE`) so that dedicated applications can use all demux.

Do you know who is the maintainer of this driver?

SS-FF, 2021-05-03:
I don't even know the name of the device or its type. I couldn't find anything useful in the `/sys/` file system.

But I found out how the firmware is build. It's from the following repository:  https://github.com/oe-alliance/oe-alliance-core
And interestingly, the DMX_SET_SOURCE feature seems to be a big thing over there: https://github.com/oe-alliance/oe-alliance-core/search?q=DMX_SET_SOURCE

This is not my world, so I'm able to make sense of all this just slowly..

lelegard, 2021-05-03:
Following your links, we can see that they even _maintain_ patches for all versions of the kernel after 4.13 to revert the official modifications in the kernel. This is crazy. They swim against the tide. They will always lose.

Root : https://github.com/oe-alliance/oe-alliance-core/tree/4.4/meta-oe/recipes-kernel/linux-libc-headers
Example : https://github.com/oe-alliance/oe-alliance-core/blob/4.4/meta-oe/recipes-kernel/linux-libc-headers/linux-libc-headers-5.2/dmx_set_source.patch

The internal structure of the kernel evolves over time. This is normal. All drivers have to adapt. The mainstream drivers are maintained with the kernel. Third-party drivers need to adapt. I see this from time to time with the Dektec drivers. They don't compile with the most recent kernel version, I report the issue to Dektec, they analyse the evolution and they adapt their drivers. This is the way everybody work. Trying to fight the evolution of the kernel is a recipe for failure.

The normal way should be what I proposed in my previous post: have a default behaviour which is compatible with the regular Linux way and add driver-specific ioctl's for additional behaviours. But don't patch the kernel against the will of the kernel maintainers. Trying to follow the same dead end as them by adding obsolete patches all the time will drive you in an endless maintenance cycle. You must have very good business reasons to support that device.

Anyway, if you follow the links of that software (Google), you find OpenEmbedded, Enigma2, Dreambox, Oscam, Kodi, softcam, NCAM, etc. The complete illegal hacker's toolbox. If you do legal business, stay out of this world. But don't even ask TSDuck to support that crap.

SS-FF, 2021-05-03:
> OpenEmbedded, Enigma2, Dreambox, Oscam, Kodi, softcam, NCAM, etc. The complete illegal hacker's toolbox. If you do legal business, stay out of this world.

I'm not much familiar with the STB side but I still know many of these things and how they work. 

What I find remarkable is that **almost all** TV software applications , even the most prominent and renowned ones, have some kind of "plugin" mechanism or silent automatic BDA-filter-insertion-when-present which allow to use cam emulations, cs integration or other kind of things. Ours has no such extension points and abilities - following a strict policy. 

But all that is really not a direction I want to take. It's just about some devices with hw tuners and accessing them for legitimate use. One could have hundreds of arguments, on both sides, but I only take care and responsibility for what _we_ are doing.
The dooming of STBs might surely have its reasons and history - but all the same can be done with most other tuners as well, so I'm not sure how to assess these things.

> But don't even ask TSDuck to support that crap.

I'm not sure at what time I would have done that. I had reported about my experience with the Linux DVB API, asked a few questions and then I have offered to share my implementation in case you would like to include it. I don't need this to be included, I already have it. I'm just trying to give some return in exchange for what I got from tsDuck, whenever I have something special to share.
A "no thanks" is totally acceptable. I still enjoyed the conversation. It's those edge cases from which you learn the most..

lelegard, 2021-05-04:
Sorry if I appear a bit harsh. After 25 years in the PayTV business, I have become quite familiar with the usual hacker's arsenal. I have always been irritated by some form of hypocrisy around those tools. Yes, an open STB is a "just" a technical tool which can be used for legitimate technical purposes. But the fact is that the vast majority of sales of those devices are for illegal purposes. After all those years in this business, _it is a fact_ that the legitimate market share of those devices is so small that, for obvious business reasons, these devices would have never existed without the market share of the illegal usage.

So, even though legitimate usages may exist (let's say 10%, being generous), the presence of such a device somewhere is a clear symptom of some form of piracy with a probability of 90% or more.

Whatever business you do, when one of your customers uses such a device, and unless you are 100% sure that he is a TV engineer or some form of cheap hospitality usage, you still have a 90+% probability that your software may be used in illegal activities. If this is a non-free software from which you get revenues with identified customers, you need to get a lawyer's advice on your responsibilities. I am just an engineer, not a lawyer...

SS-FF, 2021-05-04:
> _t is a fact_ that the legitimate market share of those devices is so small that, for obvious business reasons, these devices would have never existed without the market share of the illegal usage. [...]

Isn't it the other way round, that the biggest  hardware market share is made up from those boxes that the PayTV providers are handing out?  (closed systems of course with different software)

Anyway - aren't the high times of illegal viewing a thing of the past? I'm not really up-to-date. I just remember that few years ago, suddenly everybody "knew someone...". But then there was some technical change and the subject disappeared as quickly as it had emerged. At least here in DE. 
This is one part of the motivation actually, I think many users still have some of those boxes lying around and it will be nice to be able to use these as tuner devices.

> Whatever business you do, when one of your customers uses such a device, and unless you are 100% sure that he is a TV engineer

An engineer? Why that? We're targeting home users..

> you still have a 90+% probability that your software may be used in illegal activities.

Well - first of all, **our** software doesn't run on those devices. What's running there is a minimized version of tsDuck that does nothing more than to open a tuner, tune, receive and send the mux over the network to our software, which can do a numnber of things with it, but also basic things like recording, re-streaming, etc. 

> you need to get a lawyer's advice on your responsibilities

Well - **_our_** software doesn't run on those devices...hehe.. ;-) 

But seriously: Where do you see a potential for illegal use in that context?  I read the data from the DVB driver and send it over the network to our software. When channels are encrypted they won't work and we have no hooks or interfaces (like most other DVB applications) that would allow to do anything about it

Where (and how) do you think would the "illegal" part happen in the scenario I have described?
As long as "the illegal" wouldn't happen at the hardware or driver level, I'm not sure what your are concerned about..?

lelegard, 2021-05-04:
> Isn't it the other way round, that the biggest hardware market share is made up from those boxes that the PayTV providers are handing out? (closed systems of course with different software)

I meant _"market share for those specific open STB devices"_, i.e. legal vs. illegal usage share of them. Without illegal usages, devices such as Dreambox and all subsequent equivalent devices would have never existed because it wouldn't have been a profitable market.

Of course, in the global STB market, closed STB for operators outnumber open STB devices. Otherwise, Pay TV operators would have gone bankrupt for a while...

> Anyway - aren't the high times of illegal viewing a thing of the past? [...] But then there was some technical change and the subject disappeared as quickly as it had emerged. At least here in DE.

Nowadays, piracy has mostly moved to illegal content restreaming over IP. Thanks to high-bandwidth home connections, lousy retail DRM's on OTT devices, lousy HDCP security, etc... But, as you wrote, technology changes rapidly and so does piracy and so do counter-measures...

> This is one part of the motivation actually, I think many users still have some of those boxes lying around and it will be nice to be able to use these as tuner devices.

This is a bit opportunistic, but why not 😞 Or, positively, you act as a redeemer for the sins of former piracy devices, bringing them back on the right track 😃

SS-FF, 2021-05-04:
> This is a bit opportunistic, but why not 😞 Or, positively, you act as a redeemer for the sins of former piracy devices, bringing them back on the right track 😃

LOL, sure: mission in the name of the lord..

And that's not all. Being able to use some old boxes as if they were local hardware tuners is a nice bit and something that no other software can do normally. Also, it's about active recycling, it's a "green" feature, CO2 neutral and of course vegan, lactose- and gluten-free 😉

closed 2026-04-26 by github-actions[bot]