tsduck/tsduck #778: Add (copy) PCR to a PID

tipexfm, 2021-05-04:
Hi!

I'm trying to generate a PCR for a PID, but I'd no luck so far. 

What I'd like to do is to read the PCR values from a 'master' PID (a video PID with a valid PCR) and insert them on another PID (audio without PCR).
Something like what the 'encap' plugin does, where you can specify the output PID, the input PID, and the PCR PID. But without the encapsulation. I assume it'd need to 'repacketize' every packet in order to fit the payload of the original PID into the PCRized PID (as PCR takes someplace in the ad. field).
I also tried with 'craft', generating packets with a payload lower than 188, specifying the output PID with a dummy PCR value (later fixed with 'continuity' and 'pcradjust') , but it doesn't work.

Is it actually possible to achieve this with TSDuck?

Thank you and keep up the good work, TSDuck is a great set of tools!!

lelegard, 2021-05-05:
It would require a specific plugin. You clearly explained the spec of such a plugin: copy a PID, insert a PCR from time to time (where? on payload unit start only?), shift a part of the payload from packet to packet until a full  payload is needed or a PUSI is found, in which case we need to steal a null packet.

This is technically quite possible, it's just work...

> I also tried with 'craft', generating packets with a payload lower than 188, specifying the output PID with a dummy PCR value (later fixed with 'continuity' and 'pcradjust') , but it doesn't work.

That should create a valid PCR PID but it does not match your expectations in two ways: 1) it won't be synchronized with the video PID, 2) the audio payload is not copied into it.

lars18th, 2021-05-08:
Hi @lelegard ,

I see the implementation of the new `pcrcopy` plugin and I feel it's interesting. Now, with it we can generate PCR timestamps in a different PID. So it will possible to "move" the PCR from one pid to another (just need to first copy it and then remove the PCR from the ancient). So, great but I have serveral questions:
1. How it's the best option to "remove" a PCR from a pid with the current implementation?
2. It will be possible to create from scratch PCR marks? This will be the case when none of the pids have PCR marks but they have PTS/DTS.
3. You will provide something similar (`ptscopy`) for PTS/DTS marks? Remember that I'm waiting for a method to change PTS/DTS marks of one pid based on the difference with the PTS/DTS marks on another pid. So the use case is when you add a new _timed_ stream to one service and then you require to adjust the clock (just adding or substrating an offset).

Regards.

lelegard, 2021-05-08:
> How it's the best option to "remove" a PCR from a pid with the current implementation?

~~~
tsp ... -P filter --pid XXX --set-label 0 -P craft --only-label 0 --no-pcr -P ...
~~~

> It will be possible to create from scratch PCR marks? This will be the case when none of the pids have PCR marks but they have PTS/DTS.

In short, you can't. There is always a misunderstanding between PCR and PTS/DTS. Their origin are different. PCR is a "mechanical" system clock of the transport stream. It is easy to manipulate by TS-level tools such as TSDuck. It is trivial with CBR, a bit more fuzzy with VBR. PTS/DTS are created by codecs. Only the codec knows the relationship between the PCR and the PTS/DTS for each frame. You cannot recreate a PCR from a PTS/DTS. Similarly, PCR and PTS/DTS do not change together. When you move a TS packet slightly back and forth in the TS, you must change the PCR because the TS packet placement has changed but you shall not change the PTS/DTS because there is no reason to change the decoding and presentation time of the video frame.

I think I should adopt stricter policy on requests for features about PCR, PTS, DTS: In addition to the high-level description of the request, please propose a precise algorithm of the operations on those time stamps. This is a very good exercise to assess the understanding and feasibility of the request.

lelegard, 2021-05-08:
@tipexfm, could you test the new plugin `pcrcopy` and report if it works correctly in your case?

This is new, so you have to [rebuild TSDuck for your platform](https://tsduck.io/doxy/building.html). If you use Windows or Ubuntu Linux, nightly builds are [available here](https://tsduck.io/download/prerelease/). In that case, you have to wait until tomorrow for version 3.217-2373.

lars18th, 2021-05-09:
Hi @lelegard ,

> > It will be possible to create from scratch PCR marks? This will be the case when none of the pids have PCR marks but they have PTS/DTS.
> 
> In short, you can't. 

But then one question about an specific use case: If we have a CBR Transport Stream (from an external source, or enforced with TSDuck), we can create the PCR marks from scratch? For example, using the `pcradjust` plugin with a TS created with the `null` input indicating an specific bitrate, we can generate an empty pid with PCR timestamps?

> I think I should adopt stricter policy on requests for features about PCR, PTS, DTS: In addition to the high-level description of the request, please propose a precise algorithm of the operations on [...]

OK, I'll try to define it with more precision.

REQUEST: We have a MPTS with one (or more) services and free space. And we want to mux (inject) a new stream with timestamps (PTS) inside this MPTS. We know that the PTS value X in this added stream corresponds to the Y value of the PTS in one already stream in the service. Then we want to "resync" the PTS values of the added stream. Example of the syntax:
- `-P resync --new-pid 110 --base-pid 100 --target-pts 18402 --base-pts 986108`

In this case the operations what we want to be done are: Calculate the offset between "base-pts" and "target-pts". Search for PTS marks in the pids 110 (target) and 100 (base). For every PTS in the pid 110 apply the offset calculated using the last PTS value of the pid 100.

USE CASE: I have a MPTS with one service. I want to add some subtitles for impaired people. And the idea is to create this new stream on-the-fly as PES packets (clock starting from 0). So using the TSDuck toolkit I want to inject this PES in a "null" SPTS with CBR, create the PCR of this new SPTS and then MUX it in the MPTS. However, after I need to "resync" the PTS clock with the target service in order to play it without troubles in the updated service (for patching the PSI tables we have already all what I need).

You feel this will can be done using TSDuck?
Regards.

lelegard, 2021-05-09:
> But then one question about an specific use case: If we have a CBR Transport Stream (from an external source, or enforced with TSDuck), we can create the PCR marks from scratch? For example, using [...]

The problem is not to create PCR's, this is easy as long as you have an initial value (just update according to bitrate and packet count). When a PID has initially no PCR and we have no other PCR reference (as with `pcrcopy`), what will be the initial value? If you don't care about audio/video, then just pick any arbitary value, typically zero.

The problem is to create PCR in a PID which already has PTS/DTS (typically audio/video). These PTS/DTS were not created by the encoder from nowhere. They were created from a reference PCR. This reference PCR should be in another PID (referenced by the PCR PID field in the PMT). There are only two cases: 1) the PCR PID is still there, use `pcrcopy`, 2) the original PCR PID is lost, the time reference is lost and you can't recreate it by magic.

> REQUEST: 
> USE CASE: 

This is a request and a use case, not an algorithm.

The problem to be solved (by an algorithm):
- Streams from different services have different time references. An elementary stream of a service has PTS/DTS which are defined from the PCR of _this_ service. PCR's from other services are completely unrelated and have different time reference.
- A PTS is not a timestamp of the transport stream. This is a reference _in the future from the PCR_ for the action of presenting the frame to the display engine.
- The offset between two PTS of two elementary streams in the same service has no reason to be constant. Each codec decides how and when a frame shall be presented.

lars18th, 2021-05-10:
Hi @lelegard ,

I appreciate a lot your information. And you're right: it doesn't have sense if we want to generate PCR marks from pure scratch. However, I feel you'll agree that it has sense that _regenerate_ them from PTS marks has sense. Or not? In this case, almost you want to add the option to enforce to start from 0 for the `pcradjust` plugin (or any given value) when using with the `--min-ms-interval`. This will provide the option to generate "fake" PCR marks in one empty pid, that for example will have sense if we create an empty MPTS from the NULL input with CBR.

Anyway, regarding the use case and the algorithm description, let me to explain it another time:

- USE CASE: This is _only_ an example. You have a MPTS with two streams. One stream is a regular video service with one video stream and one audio track. The second stream is a radio service (audio only). Both streams are in the same MPTS (in fact previously muxed) and it plays without troubles. However, now you want to "add" (or "move", or "transplanting") the audio pid from the radio service to the video service. The tools to change the PSI are here in TSDuck, so you can do the "transfer" of the pid. However, the clock used for the radio service (PCR and PTS timestamps in the audio stream) are not synchronized (obviosly) with the PCR and PTS timestamps in the video service. So, you need a new tool.
- ALGORITHM: With the assumption that both services (video and radio) have a **fixed** time deviation (because even using different clocks, if the drift is very small then you'll not be in troubles), in this case you need the new `resync` plugin. This plugin calculates the difference between _target_ and _master_ PCR clocks (difference is not a drift, it's the integer value difference between two PCR timestamps, one from target and one from master, that the user indicates that is the same exact time). Then using this difference for every PTS/DTS timestamp in the _target_ stream a new value is recalculated. Futhermore, after that processing the PCR timestamps of the secondary stream will be dropped (this action can be completed with current tools).

It's now well described? You see why we need a new `resync` plugin?
I hope you want to consider it.
Regards.

tipexfm, 2021-05-10:
> @tipexfm, could you test the new plugin `pcrcopy` and report if it works correctly in your case?
>
> This is new, so you have to [rebuild TSDuck for your [...]

Hi lelegard!
Thank you for the response, I didn't expect that so quickly! :)

I'll test the new plugin pcrcopy tomorrow. I'm using Raspbian so I guess I'll need to uninstall TSDuck and install everything again. 
I never 'rebuild' anything before.

I'll let you know.

Thanks!

lars18th, 2021-05-11:
Hi @lelegard and @tipexfm ,

First of all: please excuse me to make noise on this issue regarding my request of a `resync` plugin. If you found it annoying then please let me know to continue in another different. If not, then I'll continue here.

Waiting for your response. :smiley:

lelegard, 2021-05-11:
@tipexfm, you will find binaries for Raspberry Pi (pre-release version 3.27-2377) here: https://tsduck.io/download/prerelease/

tipexfm, 2021-05-12:
Hi @lars18th, no problem for me!

@lelegard thank you for the binaries! I've successfully updated TSDuck on my system. I tested the `pcrcopy` plugin, with the reference PID and the target PID, it works very well ! 
Now I can see a valid PCR in my PID audio stream with DVBInspector :) 
I've let it run for some hours, without any problem so far...

Thank you very much for this !!! :)

lars18th, 2021-05-12:
Hi @tipexfm ,

I'm glad if the `pcrcopy` plugin functions for you. And thank you to leave me to space to continue discussing about an hypotetical `resync` plugin.

About this `resync` plugin, please @lelegard what's your opinion about this?
- If we leave out-of-the-scope the case of generating PCR marks from scratch (as this in fact the problem of how to encapsulate PES packets inside a MPEG-TS), regarding the idea of providing a method to _change/move/resync_ a pid with another pid, this is my proposal: The starting point is a MPTS with two different services. The services have time-based pids and PCR marks, and of course they are independent. Typical example is: one video service and one radio service. The challenge is to add the audio track of the radio service as an additional audio track of the video service. This implies (overall) in updating the PTS/DTS marks of the radio audio pid to be synchronized with the time clock of the video service. An example of the syntax could be: `-P resync --master-pid 1001 --target-pid 2001 --master-clock-pts 18402 --target-clock-pts 986108`. In this example the `master-pid` is the video pid of the video service, and the `target-pid` is the audio pid of the radio service. The algorithm to execute will be (more or less): identify the services of the pids to kown where are the PCR marks. After that compute the difference (offset) of the two clocks based on the pointed pts marks assuming that both are the same instant in time. Then using this difference in clock restamp all PTS/DTS marks in the `target-pid`. This processing can include optionally to remove the PCR marks of the `target-pid` (it has sense if they're inserted into this pid, and it doesn't have sense if they're in a different pid); or to optionally restamping these target PCR marks as well (in case you want to maintain the radio service alone). And regarding the PSI tables, the changes can be done after with other filters (to remove the old service, if you want to do it; to add the transplantated pid in the master service; etc.). The outcome after this processing is that the video service will have a new audio track that plays without troubles and corresponds to the audio of the radio.

Please, let me to know if this has sense from your point of view.
Thank you, as usual, for listening to us.
Regards.

lelegard, 2021-05-12:
A more consistent and complete method would be to synchronize the clock of a target service (all PCR, PTS, DTS) from a reference service (using its PCR).

> --master-clock-pts 18402 --target-clock-pts 986108

What's the use of these `-pts` parameters?

lars18th, 2021-05-12:
> A more consistent and complete method would be to synchronize the clock of a target service (all PCR, PTS, DTS) from a reference service (using its PCR).

OK. That has sense. However, only one particular case to almost think on it: if the PCR marks of the target service are inside the same pid to resynchronize, and you want to _remove_ these PCR marks (because you will remove the "service"), then it's a waste of resources to recalculate the new value of this PCR and it's preferable to remove it before. Don't you think so? For sure, this has sense only if it's optional as perhaps you want to maintain the target service.

> > --master-clock-pts 18402 --target-clock-pts 986108
> 
> What's the use of these `-pts` parameters?

It's necessary that the tool provides a method to indicate at wich point the two clocks correspond to the same point in time, no? How you want to calculate the difference then? For sure, different alternatives can be used: the most closed PUSI marks of the two pids, if the user provides the offset, etc. Perhaps with your experience you have a more simple solution. Is this the case? In my example the `--master-clock-pts` is the value of the PTS in the master pid (the video service) and the `--target-clock-pts` is the value in the target pid (the radio service, where the values will be recalculated). Thse two values represents the same instant time.

lelegard, 2021-05-12:
> It's necessary that the tool provides a method to indicate at wich point the two clocks correspond to the same point in time, no? How you want to calculate the difference then? 

1. At packet P1, get REF-PCR-1 from reference service.
2. At packet P2, get TG-PCR-2 from target service. Compute extrapolated REF-PCR-2 = (P2-P1)/bitrate. Keep DELTA-PCR = REF-PCR-2 - REF-PCR-1. Replace PCR in P2 with REF-PCR-2.
3. From now on, replace all PTS/DTS in target service with PTS/DTS + DELTA-PCR/300.

Redo steps 1 or 2 each time you get a PCR in reference or target service.

Then, the target service is synchronized with the reference service. With VBR streams, the extrapolated REF-PCR-2 will be inaccurate, possible causing playout troubles.

Note that it is essential that the PCR exists in both services. Without a PCR in the target service, it would be impossible to synchronize a PCR-less PID into a reference service.

lars18th, 2021-05-12:
> Then, the target service is synchronized with the reference service. With VBR streams, the extrapolated REF-PCR-2 will be inaccurate, possible causing playout troubles.

Sure! The idea is to work only with CBR, almost during all the processing of the "transplantation" of the new pid (that's when muxing and  resynching). So, if you want a VBR at the end then it will be preferable to remove the NULL padding after completing the task. And in this case you'll need to recompute the PCR marks after.

Anyway, I suggest to include this in the algorithm: print a warning if some recaltulated PTS or DTS values are after the PCR clock. This case can be produced if the muxing has done well but it exists a long delay between the P1 and P2 packets that it's longer than the muxing delay. In this case the process could work, but the outcoming TS will produce troubles when playing it. A simple remultiplexing should fix it, but this out-the-scope of this tool.

Regards.

lelegard, 2021-05-12:
> print a warning if some recaltulated PTS or DTS values are after the PCR clock. 

PTS and DTS are always _after_ the PCR.

If you meant _before_, it can't happen. Inside the target service, the same offset is consistently applied to all PCR, PTS and DTS. So, if a recomputed PTS/DTS is before its PCR, the original value had the same defect. The plugin would shift the clock of a service to align it to another service, it would not fix the existing inconsistencies of the service.

lars18th, 2021-05-12:
Hi @lelegard ,

Yes, you're right. Sorry! But the case that I want to point is when the POSITION of the packets are far in comparision with the TIME offset. See this example:
```
MPTS:  vvvvvaaaaavavabbbbvvvvvvvvvvaaaaavavabbbbvvvvvvvvvvaaaaavavabbbbvvvvv
            | t1     |t1x                                          |t1y
```
Here the `v` is a video packet, `a` is audio packet (of the video service) and `b` is the audio packet of the radio service. Imagine that this represents the sequence of the packets inside the MPTS. Both services have PCR marks (video inside v packets, and radio inside b packets). And all is right, so no play troubles. However, when you want to use the `resync` plugin to "incorporate" the b track in the video service, in this case if you use the packets "t1" and "t1x" as the same time point to calculate the offset, then you will don't have troubles (as the packets are close in space). But, what happens in case of using "t1" and "t1y" (imagine a gap for around 2 seconds)? In this case if you change the PTS of the "t1y" packet to be in synch with the "t1" video packet then the packet is very far _from the muxing point of view_.

That is the case that I suggest to print a warning.

lelegard, 2021-05-12:
> But, what happens in case of using "t1" and "t1y" (imagine a gap for around 2 seconds)?

This is addressed in the generic simple algo. Please review it.

lars18th, 2021-05-13:
Hi @lelegard ,

> > But, what happens in case of using "t1" and "t1y" (imagine a gap for around 2 seconds)?
> 
> This is addressed in the generic simple algo. Please review it.

Perhaps I'm stupid, but I don't see clearly how it solves the problem.

If a TS packet from the new pid (donor service) is at position "t1y" and it correspond _in time_ to a packet in the reference pid (receiver service) in the position "t1", then you will "move" the packets? If not, then it's mandatory that the _position_ in the stream are close, or the "new" service will have troubles when playing.

However, if you don't leave the option to _mark_ "t1" and "t1y", and you only use the DELTA between PCRs then it can work. But, what happens when some PCR resets?

lelegard, 2021-05-13:
> what happens when some PCR resets?

All PCR's, in reference and target PID's, are synchronisation points. If the reference PCR resets, the target will reset at the same point. If the target resets, it will be realigned at the next reference PCR. Of course, when either PCR resets, some packets between the two PCR's are completely out of sync. But a PCR reset always results in some glitch.

lars18th, 2021-05-14:
Hi @lelegard ,

Thank you for the implementation of the new `svresync` plugin! :+1: 

Regarding about how to use it, when I want is to fine-tune the synchronization... what you think it's then the best option to achieve it?
My current idea is this (thinking in a off-line processing):
- Search for one packet in each service with a PUSI mark that represents the same _instant time_.
- Mux/Join/Inject the services with the restriction that the muxing starts when the selected packets comes, and both are placed very close (ideally one after the other).
- After the muxing use the new `svresync` over the slave service.

This will work? If yes, then any idea to achieve the part of muxing the slave service using this technique?

I appreciate a lot your experience and excellent working methods. So please let me know what you think.
Regards.