tsduck/tsduck #779: Streaming to 127.0.0.1

Sefy76, 2021-05-06:
In previous versions of TSDuck, (3.24 and before), i had this command:  -O ip -l 127.0.0.1  234.250.1.1:6003
However, ever since 3.25 (and just tested 3.26) it gives error:
**Error: ip: error sending UDP message: A socket operation was attempted to an unreachable network.**

Has this option been removed or changed?
in the TSDuck PDF, i'm seeing the -l option still exists, so why is it not working with the 127.0.0.1 ?

lelegard, 2021-05-06:
Hi @Sefy76 

This is a Windows issue, not a TSDuck issue. 

The message is a Windows one, so I assume you are using Windows. The same command works fine on Linux.

I indeed get the same message when running that command. On Windows, it seems that you cannot send a multicast message to the loopback interface as long as no other program _on the same machine_ has joined the corresponding multicast group. Windows considers that the loopback interface is a local sinkhole and if noone has joined the group on this interface, your packets can't go anywhere and reports this error.

In fact, I tested the following scenario:
- Run your command => same error as you.
- Start a listener on the same machine from another command window, joining the same multicast group on the loopback interface (`tsp -I ip  -l 127.0.0.1 224.250.1.1:6003 ...`).
- Re-run your command => it works now

This is a bit silly (just another idiotic network behaviour from Windows) because you may want to run the sender first and start the listener sometime later. It works on Linux, not on Windows.

Strangely enough, the command works again after the listener has stopped and has dropped the multicast group. It seems that sending multicast on the loopback interface works only when at least one listener had once joined the multicast group since the machine booted. Windows networking...

Just googling for that message, it seems to be well known. You will find some clues in [this discussion](https://social.technet.microsoft.com/Forums/Exchange/en-US/a6691bb1-eb71-41eb-b739-dd136758bb3c/sending-multicast-to-localhost-windows-10?forum=win10itpronetworking)

I guess that the behaviour has always been the same for all TSDuck versions. If it had already workd for you, this is probably because a listener on the same multicast address had been run on the same machine before running the sender.

Sefy76, 2021-05-06:
Thenk you @lelegard, you've been great help as always!

closed 2021-05-06 by Sefy76