tsduck/tsduck #780: [Q] Merge two HLS streams - PAT is not merged

skynetbeta23, 2021-05-07:
Hi everyone.

My goal is to merge two HLS streams with VBR in a single CBR transport stream and save it in a file.

I use the following command:

tsp -I hls --highest-bitrate https://streamcdnb5-8e7439fdb1694c8da3a0fd63e4dda518.msvdn.net/rainews1/hls/rainews_2400/chunklist.m3u8 | tsstuff --bitrate 20000000 | tsp -P merge --no-pcr-restamp "tsp -I hls --highest-bitrate https://nbcnews2.akamaized.net/hls/live/723426/NBCNewsPlaymaker24x7Linear99a3a827-ua/VIDEO_1_4596000.m3u8 -P svrename --id 2 1" -P pcradjust --bitrate 20000000 -P pcrbitrate -O file testfile_mux.ts

So I get the first stream (with VBR around 2.5 Mbps), I stuff it to 20 Mbps, I merge the second stream (with VBR around 3 Mbps) and finally I fine-tune the CBR to 20Mbps and save the resulting stream in a file.

My issue is that the PAT is not getting merged, so while all PIDs from the second stream (including its PMT) are included in the final stream, the PAT only refers the PMT of the first stream.

For testing purposes I recorded on file two chunks of the two streams:

tsp -I hls --highest-bitrate https://streamcdnb5-8e7439fdb1694c8da3a0fd63e4dda518.msvdn.net/rainews1/hls/rainews_2400/chunklist.m3u8 -O file stream_a.ts

tsp -I hls --highest-bitrate https://nbcnews2.akamaized.net/hls/live/723426/NBCNewsPlaymaker24x7Linear99a3a827-ua/VIDEO_1_4596000.m3u8  -O file stream_b.ts

Then I made the same operation on the two files:

tsp -I file stream_a.ts | tsstuff --bitrate 20000000 | tsp -P merge --no-pcr-restamp "tsp -I file stream_b.ts -P svrename --id 2 1" -P pcradjust --bitrate 20000000 -P pcrbitrate -O file testfile_mux.ts

With this "offline" version things work fine and the PAT includes both PMT tables, so both services can be displayed.

I can't understand why I cannot make it work on the "live" version.

Thanks for any help

lelegard, 2021-05-07:
Hi @skynetbeta23,

I just ran the same commands as yours (plain copy/paste from your post and I got the PAT correctly merged in the two cases.

How did you come to the conclusion that the PAT was not merged in the live case? Did you use `tsanalyze`, displayed all PAT, only the first one?

I ask the question in these terms because there may be some confusion. This is a _merge_ of one secondary stream into the first main stream. At the beginning, the main stream contains an input PAT (one service in your case). This PAT is passed all the way down. Then, at some early stage, the `merge` plugin discovers a PAT in the secondary streams and merges it with the main PAT, producing an updated version of the PAT with all services from both streams.

Depending on the order of arrival of the first packet in PID zero from both streams and the interval between the two first packets (one from each stream), you may see two consecutive PAT's or only one from the beginning.

Example, using the file I merged using your live command:
~~~
$ tstables testfile_mux.ts --pid 0 --packet-index

* First TS packet: 0, last: 0
* PAT, TID 0 (0x00), PID 0 (0x0000)
  Version: 0, sections: 1, total size: 16 bytes
  - Section 0:
    TS id:       1 (0x0001)
    Program:     1 (0x0001)  PID: 4095 (0x0FFF)

* First TS packet: 25,846, last: 25,846
* PAT, TID 0 (0x00), PID 0 (0x0000)
  Version: 1, sections: 1, total size: 20 bytes
  - Section 0:
    TS id:       1 (0x0001)
    Program:     1 (0x0001)  PID: 4095 (0x0FFF)
    Program:     2 (0x0002)  PID:  100 (0x0064)
~~~

Since the PAT version is correctly incremented, any conformant receiver shall detect the change and detect the new service.

Note that the two PAT's are distant by 25946 packets, meaning 1.9 second at 20 Mb/s. Given that two commands are separately started, each one connecting to a remote HLS server with probably variable response times, crossing Wifi, NAT and VPN in my case, I cannot be surprised that they don't receive the first packet within the same second.

skynetbeta23, 2021-05-07:
Hi  lelegard , thanks for your support.

Yes, you're right. PAT is indeed merged. I ran tstables and I confirm the result. 

I apologize, I used an analysis software which probably detects only PAT in version 0 and never updates it; the same apparently is made by VLC.

I take advantage of this post to ask you if the command I'm running is indeed the best one to get a 20 Mbps CBR stream from the two HLS streams.

That's because I made a new analysis on the file with a tool called StreamXpert and I find many errors:

 - PCR repetition error

 - PCR discontinuity error

 - Interval between PAT sections is 2.02 seconds (standard limit 500 ms)

And as practical issue I cannot display Program 2, it has many glitches.

So I suppose there's something wrong in my command. I also tried excluding the --no-pcr-restamp flag from the command, with no success.

Thanks

lelegard, 2021-05-07:
> I take advantage of this post to ask you if the command I'm running is indeed the best one to get a 20 Mbps CBR stream from the two HLS streams.

This topic has been raised many times. What you need is a MUX and TSDuck is not a MUX. "Merging" streams means merging the dirty way, which may be sufficient for data streams. "Muxing" audio/video streams requires precise timing control, which is quite another story. Since you have StreamXpert, you may have another Dektec software named MuxXpert which has the reputation of being an excellent software MUX. I know that it is expensive but if MuxXpert costs €1000 and TSDuck is free, there is a reason: muxing is difficult.

> I apologize, I used an analysis software which probably detects only PAT in version 0 and never updates it; the same apparently is made by VLC.

If you want to remove the first PAT (the one with only one service) and keep only the larger one, you may use this trick: add the following two plugins after the `merge` plugin:
~~~
-P filter --pid 0 --set-label 0 -P filter --only-label 0 --pattern 0000B00D --search-payload --search-offset 0 --negate --stuffing
~~~
I let you read the documentation as an exercise to better understand all the bizarre TSDuck features which can be used to address no less bizarre situations...

closed 2021-06-01 by skynetbeta23