tsduck/tsduck #782: [Q] What does "stderr: * Warning: regulate: out of sequence PCR, maybe source was cycling, restarting regulation" mean?

amit2maha1, 2021-05-09:
we are using tsswitch to play a string of files in sequence. We are getting this warning sometimes when we switch between video files. What does it mean?

lelegard, 2021-05-09:
You didn't mention the context. This is a `regulate` plugin message, not a `tsswitch` message (it only uses input and output plugins). So, I just guess that the message comes from a `tsp` command either at the output of `tsswitch` or at one of its inputs.

General cause: the `regulate` plugin suspends the execution on a regular basis to adapt the _processing speed_ to the _playout speed_. The regulation is based on the bitrate or PCR's. In your case, the message indicates regulation based on PCR's (option `--pcr-synchronous`). The PCR are clock timestamps and the regulation is then based on those timestamps. In a regular TS, the PCR's are strictly incremental (wrap-up at 42 bits) and the increment represents a fraction of a second. This message indicates that the difference between two consecutive PCR's is so huge or so bizarre that they cannot come from the same stream. This is a clear symptom that, between these two PCR's, the source TS has changed.

This is just a warning. The regulation restarts on the new PCR reference. It just warns you that the small interval between these two PCR's (a fraction of a second) was not properly regulated.

If the message comes from an input `tsp` command, it is probably because you used `-I file --infinite`, meaning looping back the file. The last PCR of the file is followed that its first one, not in sequence of course.

If the message comes from an output `tsp` command, it is absolutely normal when the input switches from one source to the next one. There is no reason that the last PCR of file N and the first PCR of file N+1 are continuous. There is a leap.

Anyway, in all cases, the reason is normal, this is just a warning, you may safely ignore it.

Note: If your only purpose is to play a string of files in sequence, there is no need to use `tsswitch`, the `file` input plugin can do that: `tsp -I file file1.ts file2.ts file3.ts ...`. RTFM as usual. `tsswitch` is typically used to aggregate different kinds of input, implement input backup, allow remote control.

amit2maha1, 2021-05-09:
> You didn't mention the context. This is a `regulate` plugin message, not a `tsswitch` message (it only uses input and output plugins). So, I just guess that the message comes from a `tsp` command [...]

first of all, Thank you for the reply.

I see, you are correct `-I file --infinite` is the cause of it. 
Ok now, what if we fed the output of this tsswitch to ffmpeg to create an HLS stream. Do you think it would cause any packet corruption on ffmpeg part? I know this isnt the right place to ask about ffmpeg but maybe you know something about it?

We are doing this and it is causing `Packet corrupt (stream = 0, dts = 46211540).` error in ffmpeg side when creating the HLS stream.

lelegard, 2021-05-09:
If you jump from one stream into another, just like that, in the middle of a stream, you cut a video frame in the middle and restart with probably the middle of another video frame. So, this is quite normal that ffmpeg complains about corruption. The actual question is: is this just a transient error and ffmpeg continues or does it stop?

If the error makes ffmpeg stop, you have to find a way to smoothen the transition. If this is only a matter of time stamps, you may try the plugin `pcradjust` (please RTFM and try the various options by yourself) but the result is not perfect. If ffmpeg stops because of video corruption, then you have to find a solution using video tools or make sure in advance that file beginning and end are properly cut.

amit2maha1, 2021-05-09:
> pcradjust

I see, I will try `pcradjust`. But I think the problem is in recordings, because we are recording from an SDI input. Each recording file starts with black screen, only after 1 secs of black screen it starts recording any content. Do you have suggestions to compensate for this when starting recording with `tsp` ?

thanks

closed 2026-04-26 by github-actions[bot]