tsduck/tsduck #784: [Q] DVB input TBS 5580 not working

skynetbeta23, 2021-05-13:
Hi everyone,

I have and USB device "TBS 5580" on Ubuntu 20.04 with working original TBS driver.

'''
~$ ls /dev/dvb/adapter0

ca0  demux0  dvr0  frontend0  frontend1  net0

~$ tslsdvb
0: /dev/dvb/adapter0 ("TBS 5580 CI USB2.0 DVB-T/T2/C/C2/ISDB-T, TBS-Tech TBS 5580 2.00", DVB-T, DVB-T2, ISDB-T, DVB-C, 2)
1: /dev/dvb/adapter0:1:1:1 ("TBS 5580 CI USB2.0 DVB-S/S2/S2X, TBS-Tech TBS 5580 2.00", DVB-S, DVB-S2, DSS)
'''

I should be using the satellite frontend (DVB-S / DVB-S2) but i can't select it.

I have tried these commands with no success:

'''
tsp -I dvb \
--device-name "/dev/dvb/adapter0:1:1:1" \
--lnb universal \
--delivery-system DVB-S \
--frequency 10992000000 \
--polarity vertical \
--symbol-rate 27.5 \
--receive-timeout 5000 \
--signal-timeout 5 \
-P zap "Rai 1" --audio ita --no-subtitles \
-O file --max-duration 60 channel.ts

* Error: dvb: error opening /dev/dvb/adapter0/dvr1: No such file or directory
'''

'''
tsp -I dvb \
--device-name "/dev/dvb/adapter0:1:0:0" \
--lnb universal \
--delivery-system DVB-S \
--frequency 10992000000 \
--polarity vertical \
--symbol-rate 27.5 \
--receive-timeout 5000 \
--signal-timeout 5 \
-P zap "Rai 1" --audio ita --no-subtitles \
-O file --max-duration 60 channel.ts

* Error: dvb: tuning error on /dev/dvb/adapter0/frontend1: Invalid argument
'''

'''
tsp -I dvb \
--device-name "TBS 5580 CI USB2.0 DVB-S/S2/S2X, TBS-Tech TBS 5580 2.00" \
--lnb universal \
--delivery-system DVB-S \
--frequency 10992000000 \
--polarity vertical \
--symbol-rate 27.5 \
--receive-timeout 5000 \
--signal-timeout 5 \
-P zap "Rai 1" --audio ita --no-subtitles \
-O file --max-duration 60 channel.ts

* Error: dvb: error opening /dev/dvb/adapter0:1:1:1/frontend0: No such file or directory
'''

How can I use the satellite frontend ? 

In the "TSduck" manual (page 409) the device TBS 5580 is marked as "Tested OK" so I suppose I'm missing something. 

Can someone please give me an hand ?

lelegard, 2021-05-13:
Could you please post the output of `ls -lR /dev/dvb` ?

lelegard, 2021-05-13:
And please note that, for the sake of consistency, all units are "natural", i.e. Hz, bits/seconds, symbols/seconds, etc. No kilo, or mega or giga is used by default. So you must use `--symbol-rate 27500000` not `27.5`. Not related to the dvr issue of course.

skynetbeta23, 2021-05-13:
> And please note that, for the sake of consistency, all units are "natural", i.e. Hz, bits/seconds, symbols/seconds, etc. No kilo, or mega or giga is used by default. So you must use --symbol-rate 27500000 not 27.5

Yes, that was the issue. I was confused by the default value in the --symbol-rate description in the manual.

~$ ls -lR /dev/dvb
/dev/dvb:
total 0
drwxr-xr-x 2 root root 160 mag 13 10:41 adapter0

```
/dev/dvb/adapter0:
total 0
crw-rw----+ 1 root video 212, 3 mag 13 10:41 ca0
crw-rw----+ 1 root video 212, 0 mag 13 10:41 demux0
crw-rw----+ 1 root video 212, 1 mag 13 10:41 dvr0
crw-rw----+ 1 root video 212, 4 mag 13 10:41 frontend0
crw-rw----+ 1 root video 212, 5 mag 13 10:41 frontend1
crw-rw----+ 1 root video 212, 2 mag 13 10:41 net0

```

Everything works now. Thanks!

lelegard, 2021-05-13:
>  I was confused by the default value in the --symbol-rate description in the manual.

The manual says _"Symbol rate in symbols/second "_. I don't know how to make it more clear. 27.5 means 27.5 mega-symbols per second. This is the standard order of magnitude for satellite transmission. 27.5 symbols per second wouldn't transmit much...

Concerning the device naming, the Linux driver for this device is a but confusing. It seems that there is only one tuner here. The documentation says "multi-standard single tuner". Usually, the driver would expose only one frontend (there should be one frontend per tuner). For some bizarre reason, they chose to expose one frontend interface for DVB-T/T2/C/C2/ISDB-T and a distinct frontend interface for DVB-S/S2/S2X. But in fact, there is only one hardware. You cannot use DVB-T and DVB-S at the same time for instance, although having them on two frontends would suggest that you could. And since there is in fact one single data path, they expose only one demux and one dvr. 

This is why you must use `/dev/dvb/adapter0:1:0:0`, meaning "use frontend 1 but demux 0 and dvr 0".

The usual configuration of a driver is to have one frontendN per hardware tuner, exposing all modulations for that tuner, and one demuxN and one dvrN per tuner. So, by default, TSDuck uses the same N for demux and dvr when you use frontend N. The possibility to specify a non-default combinations such as `/dev/dvb/adapter0:1:0:0` was done to support bizarre drivers such as this one.

closed 2026-04-26 by github-actions[bot]