tsduck/tsduck #785: [Q] does the dektec output plugin doesn't terminate tsp ?

izharm, 2021-05-13:
when using the "until" plugin with a DekTec output plugin (in ASI mode), the output plugin doesn't terminate and as a consequence the tsp process doesn't terminate.

is this normal behavior of DekTec output plugin?

Izhar

lelegard, 2021-05-13:
> is this normal behavior of DekTec output plugin?

Definitely not. In fact, this is not under control of the output plugin. An output plugin just implements `start()`, `send()` and `stop()` primitives.

The orchestration is done at tsp level. When a plugin in the middle (the `until` plugin here) decides to terminate, it returns that information to tsp. Then, tsp propagates the termination both ways. Downward to the output plugin, an "end of input" condition is propagated, as if the input plugin had no more data.

As long as the "output executor" (the tsp thread that calls the output plugin) receives packets, it calls the `send()` function in the plugin. When it receives an "end of data", the output executor thread calls the `stop()` function and terminates. This is as simple as that.

To investigate what's going on, I suggest `tsp -d` or even `tsp -d10` (debug level 10). It is very verbose, so take care, use a short `until` condition. Please post the last part of the log, starting a bit before the last `waitWork` message of the output plugin.

izharm, 2021-05-18:
here you go;

```
 Debug[2]: ip: received UDP packet, source: 172.20.8.241:11111, destination: 172.20.8.1:51105, timestamp: 1,621,336,174,884,761
* Debug[10]: ip: passPackets(count = 7, bitrate = 5,199,983, input_end = false, aborted = false)
* Debug[10]: ip: waitWork(min_pkt_cnt = 1, ...)
* Debug[10]: ip: waitWork(min_pkt_cnt = 1, pkt_first = 17,878, pkt_cnt = 71,362, bitrate = 0, input_end = false, aborted = false, timeout = false)
* Debug[10]: until: waitWork(min_pkt_cnt = 1, pkt_first = 17,871, pkt_cnt = 7, bitrate = 5,199,983, input_end = false, aborted = false, timeout = false)
* Debug[10]: until: passPackets(count = 7, bitrate = 5,199,983, input_end = false, aborted = false)
* Debug[10]: until: waitWork(min_pkt_cnt = 1, ...)
* Debug[10]: dektec: waitWork(min_pkt_cnt = 1, pkt_first = 17,871, pkt_cnt = 7, bitrate = 5,199,983, input_end = false, aborted = false, timeout = false)
* Debug[10]: dektec: passPackets(count = 7, bitrate = 0, input_end = false, aborted = false)
* Debug[10]: dektec: waitWork(min_pkt_cnt = 1, ...)
* Debug[2]: ip: received UDP packet, source: 172.20.8.241:11111, destination: 172.20.8.1:51105, timestamp: 1,621,336,174,886,780
* Debug: ip: input: got bitrate 5,199,983 b/s
* Debug[10]: ip: passPackets(count = 7, bitrate = 5,199,983, input_end = false, aborted = false)
* Debug[10]: ip: waitWork(min_pkt_cnt = 1, ...)
* Debug[10]: until: waitWork(min_pkt_cnt = 1, pkt_first = 17,878, pkt_cnt = 7, bitrate = 5,199,983, input_end = false, aborted = false, timeout = false)
* Debug: until: plugin requests termination
* Debug[10]: until: passPackets(count = 1, bitrate = 5,199,983, input_end = true, aborted = true)
* Debug[10]: ip: waitWork(min_pkt_cnt = 1, pkt_first = 17,885, pkt_cnt = 71,355, bitrate = 0, input_end = false, aborted = false, timeout = false)
* Debug: until: packet processing thread terminated after 17,880 packets, 17,879 passed, 0 dropped, 0 nullified
* Debug[10]: dektec: waitWork(min_pkt_cnt = 1, pkt_first = 17,878, pkt_cnt = 1, bitrate = 5,199,983, input_end = true, aborted = false, timeout = false)
* Debug[10]: dektec: passPackets(count = 1, bitrate = 0, input_end = false, aborted = false)
* Debug[10]: dektec: waitWork(min_pkt_cnt = 1, ...)
* Debug[10]: dektec: waitWork(min_pkt_cnt = 1, pkt_first = 17,879, pkt_cnt = 0, bitrate = 5,199,983, input_end = true, aborted = false, timeout = false)
* dektec: terminating DTA-2174B output
* Debug[2]: ip: received UDP packet, source: 172.20.8.241:11111, destination: 172.20.8.1:51105, timestamp: 1,621,336,174,888,805
* Debug[10]: ip: passPackets(count = 7, bitrate = 5,199,983, input_end = false, aborted = false)
* Debug[10]: ip: waitWork(min_pkt_cnt = 1, ...)
* Debug[10]: ip: waitWork(min_pkt_cnt = 1, pkt_first = 17,892, pkt_cnt = 71,348, bitrate = 0, input_end = false, aborted = true, timeout = false)
* Debug: ip: input thread aborted after 17,892 packets
^C* tsp: user interrupt, terminating...
* Debug: aborting all plugins...
* Debug: aborting plugin ip
* Debug: ip: aborting IP input
* Debug: aborting plugin until
* Debug: aborting plugin dektec
^C

```

lelegard, 2021-05-18:
Hi @izharm,

Thanks for the traces. The termination is processed correctly. All plugins terminate. The `dektec` output plugin also terminates but it seems that the Dektec API hangs in the final detach operation.

Does it occur all the time with the DTA-2174B?

Could you try the option `-O dektec --instant-detach ...`? As indicated in the documentation, this option is recommended on Dektec modulators which sometimes hang during detach while ASI devices usually don't. Maybe the DTA-2174B has a detach behaviour which is similar to the modulators.

The only issue with `--instant-detach` is that the current output operation may be interrupted before all bytes are output. So, the last packets may not be output on the ASI port. Depending on what you are expecting, this may be a problem or not. If this option solves your problem, then this is the same issue as with some Dektec modulators and I see no other option.

izharm, 2021-05-18:
> Does it occur all the time with the DTA-2174B?

yes it does.

> Could you try the option -O dektec --instant-detach ...?

yes, this solved the problem.

thanks.

closed 2021-05-18 by izharm