tsduck/tsduck #786: [Q] Output to custom hardware device ...

kappasm, 2021-05-14:
Hi, 

I have a custom device on linux which can be similar to a Decktec or Hides, on which I am writing the output of a treatment with tsp.

To write to the device I use the "-O file /dev/dvbdev". My device exposes a device of type "char" in "/dev".

The stream manipulated by tsp is written but in some moments I get this error message: 

`
* Error: file: error writing /dev/dvbdev: Message too long (90)
`

Is there a flag for tsp that limits the size of the write buffer ?

lelegard, 2021-05-14:
> Is there a flag for tsp that limits the size of the write buffer ?

Not as such. But you can try `--max-flushed-packets` at the `tsp` level. It is not exactly the same. It means that `tsp` does not wait for more than this number of TS packets to pass from one plugin to another (including the output plugin). As a consequence, this reduces the size of most output write requests. But this is not a guarantee. It more packets are already present, they are all passed to the output plugin.

It would be trivial to add some option to limit in a guaranteed way the size of each output request (regardless of the output plugin).

That being said, this error is returned by the driver of your custom device. If the error means that the size of the `write()` request is too large, I would say that this is a defect of the driver. A Linux driver has the right to limit the output size but there is a proper way to do that: just output what you can output and return the output size. As far as size is concerned, writing is not a binary operation, success or fail. The system call `write()` is designed to return the actual output size. This means that the size parameter is just a request, the driver should output what it can output and returns the actual output size. By immediately returning a "message too long" error, your driver is not correctly implemented.

kappasm, 2021-05-14:
Unfortunately the driver was written some time ago and I didn't even write it myself. I have no idea where the sources are and how to do it. 

I will try the flags `--max-flushed-packets` you reported to me.

In my opinion it should be enough .

Thanks very much.

lelegard, 2021-05-15:
Hi @kappasm 

I just added an option `--max-output-packets` to `tsp`. It made sense because, first, the option `--max-input-packets` already existed without output counterpart and, second, the option `--max-flushed-packets` may be fine as a workaround but it impacts all packet transfers between all plugins, not just the output.

kappasm, 2021-05-17:
Hi @lelegard 

thanks very much for this new flag. As soon as you release the new version I will definitely try it. 

Thanks very much.

lelegard, 2021-05-17:
> As soon as you release the new version I will definitely try it.

You can use the latest nightly builds for Ubuntu and Windows from here: https://tsduck.io/download/prerelease/

All commits go through the same QA testing process as official releases. Nightly builds are not worse than official releases when you want to use new features (or you might say that official releases are not better than nightly builds...)

For other Linux flavors, you may rebuild from these instructions: https://tsduck.io/doxy/building.html

kappasm, 2021-05-18:
Hi @lelegard

excellent choice the new flag "--max-output-packets". I tried and it works perfectly, you just have to be careful to put it on the last "tsp".

lelegard, 2021-05-18:
> you just have to be careful to put it on the last "tsp".

What do you mean?

kappasm, 2021-05-18:
> 
> 
> > you just have to be careful to put it on the last "tsp".
> 
> What do you mean?

I use a chain "tsp ... | tsstuff ... | tsp --max-output-packets xx ... -O file / dev / dvbdev"

The "--max-output-packets" flag is placed on the last "tsp".

lelegard, 2021-05-18:
OK, got it. Yes, it must be in the tsp command which addresses your output device, of course.