tsduck/tsduck #788: [BUG] tsdektec fails to set DTA-115 output mode

philpem, 2021-05-16:
#### Bug description:
`tsdektec` doesn't set the ASI port to output mode on the Dektec DTA-115

#### How to reproduce:
- Run `tsdektec -o0 0` (assumes the DTA-115 is device 0)

#### Expected behavior:
- Dektec card set to ASI output mode -- ASI port is channel 0 output, Modulator is channel 1 output

#### Errors and logs:
~~~
$ tsdektec -av

DTAPI: 5.43.2.149, Dta: 4.28.1.271, DtaNw: -1.-1.-1.-1

* Device 0: DTA-115 (DTA-115 Modulator with VHF/UHF Upconverter)
  Physical ports: 2
  Channels: input: 1, output: 1
  Input 0: DTA-115 port 1 (top), ASI (Uni-directional input, Uni-directional
      output, ASI output signal can be inverted, MPEG-2 transport stream,
      DVB-ASI transport stream, Internal TS rate clock reference)
  Output 0: DTA-115 port 2, Modulator (VHF-band 47-470MHz, UHF-band 400-862MHz,
      Uni-directional output, MPEG-2 transport stream, Modulator output, ATSC
      8-VSB modulation, DVB-T modulation, QAM-A modulation, QAM-B modulation,
      QAM-C modulation, Adjustable output level, Continuous wave, Mute RF output
      signal, Internal RF clock reference, Internal TS rate clock reference)
  Subsystem id: 0x0073 (DTA-115)
  Subsystem vendor id: 0x1A0E
  Device id: 0x9056
  Vendor id: 0x10B5
  Serial number: REDACTED
  Firmware version: 2 (0x00000002)
  Firmware variant: 0 (0x00000000)
  PCI bus: 4, slot: 1
  Customer id: 303010
  Engineering change level: Rev 3
  Manufacture id: 03
  Production date: 2010.12
  Part number: DTA-115
  Serial number: 4115006837
  Crystal stability: RF:1ppm;Sym:2ppm

$ tsdektec -o0 0

$ tsp -v -m --bitrate 8000000 --buffer-size-mb 128 -I null -P regulate -O dektec --device 0 --channel 0
* [MON] 2021/05/16 14:48, resource monitoring started
* tsp: buffer failed to lock into physical memory (12: Cannot allocate memory), risk of real-time issue
* regulate: minimum packet burst duration is 10,000,000 nano-seconds
* null: initial input bitrate is 8,000,000 b/s
* dektec: using Dektec device 0, output channel 0 (DTA-115 port 2)
* Error: dektec: unspecified modulation type for DTA-115
* [MON] 2021/05/16 14:48, resource monitoring terminated

# TSDuck has clearly not set up the ASI port as an output. Asking for a modulation type shows that TSDuck has selected the DVB-T modulator.
# With the ASI set as output, Channel 0 should be the ASI port, and Channel 1 should be the DVB-T modulator

~~~

Running DtPlay, instructed to output to ASI (device 1 port 1) switches the output to ASI successfully, and from that point TSDuck can output to the ASI port.

#### Environment:
 - OS: Linux
 - OS version: Linux Mint 20 "Ulyana"
 - TSDuck full version: 3.26-2349
 - Installation type: Official release binary, Debian package from tsduck website

#### Additional information:
There seems to be no issue playing out ASI with DtPlay; in fact it can be used to switch the card into ASI mode as a workaround (play a short null stream through the card with DtPlay).

lelegard, 2021-05-16:
Hi @philpem 

Thanks for reporting this.

Your command is `tsdektec -o0 0`, meaning "set port 0 in output mode on Dektec device 0".

The output of `tsdektec -av` is:
~~~
* Device 0: DTA-115 (DTA-115 Modulator with VHF/UHF Upconverter)
  Physical ports: 2
  Channels: input: 1, output: 1
  Input 0: DTA-115 port 1 (top), ASI (Uni-directional input, Uni-directional
...
~~~

As you may see, the output says _"port 1 (top)"_. So, you should use `-o 1`. Please retry with `-o 1` and then use `tsdektec -av` again to check if the port has successfully changed direction.

To explain the different concepts:
- A Dektec "device" is physical device. Numbering starts at 0. The number depends on the placement on the PCIe bus or the plugin order for USB devices. This is allocated by the Dektec driver.
- A "port" is a physical socket on the device. Numbering starts at 1. The number is assigned by Dektec as a built-in characteristics of the device. We cannot change this.
- An input or output "channel" is a capability of a port at a given time. Numbering starts at 0. The allocation is done by TSDuck, in the order of ports, considering their current configuration (input or output for bi-directional ports).

I realise that there is one issue with `tsdektec`. The input/output port numbers should not allow 0. Allowed values should start at 1 since port 0 never exists. To make things worse, the value 0 is internally considered as "value not set, option not used", which explains why `tsdektec` does not do anything, without even an error message. I just changed this. Zero will no longer be accepted as a port number and you will get the appropriate error message.

One additional point: In issue #771, there was an error setting input/output mode on a DTA-2174B (with a proper error message). This was due to recent evolutions in the Dektec software for devices such as DTA-2174B with internal signal routing capabilities. In that case, it was no longer sufficient to request "output mode" (or input). An additional parameter was necessary to say "output to/from where", with output to the physical socket being one of the options. The problem was fixed in TSDuck and "output to socket" (or input) is now always explicit.

I do not think that this will impact your DTA-115. But, in case you get an error message when setting the output mode, you should try the latest nightly builds (3.27-something) or rebuild from the source repo.

philpem, 2021-05-16:
I actually tried "-o1" first -- this is what I got:
```
$ tsdektec -o1
* Error setting port 1 to output mode: DTAPI_E_INVALID_ARG (DTAPI status 4140)
1 playout@playout-dvbt:~/Videos/recast$ 
```
I'm not sure why it's complaining about the argument being invalid... I figured it meant the port number, which is why I subtracted 1.

I'll try the latest nightly.

philpem, 2021-05-16:
I've just tried the latest nightly, version `3.27-2384`.

```
playout@playout-dvbt:~/tsduck$ tsdektec -o1 0
playout@playout-dvbt:~/tsduck$ tsdektec -av

Segmentation fault (core dumped)
```

With the exception of `tsdektec -av` crashing, `tsdektec -o1 0` seems to work as expected, and sets the ASI port to output mode.

lelegard, 2021-05-16:
The error `DTAPI_E_INVALID_ARG ` is indeed the one from #771 (good to know that it does not only impact new devices such as DTA-2174B but also older devices without internal routing). So, you need the latest version.

The `tsdektec` crash is new to me.

Did you recompile from the repo or used the Ubuntu binary from https://tsduck.io/download/prerelease ? I do not know Linux Mint 20 and I am not sure that it is compatible with binaries built on Ubuntu 20. Since you used the Debian binary from the latest version, this may make a difference.

If you used the Ubuntu binary, I suggest to rebuild on your system. Details here: https://tsduck.io/doxy/building.html

If rebuilding is an issue for you, I will manually rebuild a Debian version.

lelegard, 2021-05-16:
I just uploaded version 3.27-2385 for Debian here: https://tsduck.io/download/prerelease

philpem, 2021-05-16:
Mint is an Ubuntu derivative - Mint 20 is based on Ubuntu 20.04 LTS.

I used the packages on https://tsduck.io/download/prerelease/ - 
- tsduck-dev_3.27-2384.ubuntu20_amd64.deb
- tsduck_3.27-2384.ubuntu20_amd64.deb

lelegard, 2021-05-16:
@philpem, since you mentioned in your original post that you used the Debian packages, could you try the latest Debian binaries I just posted? Or, better, could your recompile from sources?

@izharm, I need your help here. Since you use Ubuntu 20 and the latest Dektec fixes on it, do you have any issue running `tsdektec -av`?

lelegard, 2021-05-16:
@philpem, I have quickly setup a Mint VM and rebuilt TSDuck on it (the eighth Linux VM, the eighth distro, on my Mac host). The binaries are here: https://tsduck.io/download/prerelease/

I do not have Dektec devices at home and I cannot test `tsdektec -av`. Or more precisely it just displays the DTAPI version number (without crashing).

izharm, 2021-05-17:
> @philpem, since you mentioned in your original post that you used the Debian packages, could you try the latest Debian binaries I just posted? Or, better, could your recompile from sources? [...]

Last I check, a few days ago, it worked just fine.  I can retest latest changes tomorrow.

philpem, 2021-05-17:
I've built from source (Git ec03e08) and `tsdektec` works fine.
The Mint 20 package (`tsduck_3.27-2385.mint20_amd64.deb`) also works fine.

The binary in the `3.27-2384.ubuntu20` package crashes - which is weird, as `tsduck_3.26-2349.ubuntu20_amd64.deb` (latest Ubuntu 20 release DEB) works fine (aside from the output-mode bug).

What version of Ubuntu are the packages built on?
I ask because "Ubuntu 20" could mean 20.04 or 20.10, and there's a big difference.
As Mint is based on 20.04, packages for that should install and work cleanly. If you're building on 20.10, then both Mint 20 and Ubuntu 20.04 users will have issues.

I usually build packages in Docker containers to avoid these kind of issues...

lelegard, 2021-05-17:
> What version of Ubuntu are the packages built on?

The Ubuntu and Windows nightly builds are automatically produced using the GitHub Actions CI/CD environment. The Ubuntu GitHub runners always use LTS versions, currently 20.04. The exact list of pre-installed packages can be found somewhere in the GitHub documentation. So, it starts from a pretty standard and clean environment. The script `build/install-prerequisites.sh` is executed first, then `make`. The workflow is available here: https://github.com/tsduck/tsduck/blob/master/.github/workflows/nightly-build.yml

The "official" releases are produced on my environment using VM's with latest versions. Next release will be built on Ubuntu 21.04.

The production of Linux binaries is mostly a help for users. There are so many different distros, with so many different versions and alternatives that I cannot address them all. On Linux, there is always the possibility for everyone to easily rebuild.

On Windows, many users do not have the setup to rebuild. I automated the build the same way as on Linux: scripts `install-prerequisites.ps1` and `build-installers.ps1` used in GitHub Actions CI/CD. Fortunately, the installers seem to work fine on non-prehistoric versions of Windows, despite its myriad of versions and sub-versions.

izharm, 2021-05-18:
@lelegard, still need my help ?

lelegard, 2021-05-18:
> @lelegard, still need my help ?

@izharm, no thanks. There was a suspicious crash from the latest build of `tsdektec`, similar to the one you use. But @philpem made it work with a native build on Mint. The problem seems to be linked to binaries which are built on Ubuntu 20.04 and installed on Mint 20. The exact root cause of this inconsistency remains unknown, though.

conorOBS, 2022-02-24:
@lelegard I've recently built the latest master branch and running "tsdektec -av" results in a segmentation fault. I'm running Ubuntu 20.04, let me know if you need any more details.

closed 2022-09-23 by lelegard