tsduck/tsduck #790: [Q] Secondary process monitoring and HLS reconnect behaviour

skynetbeta23, 2021-05-19:
Hi lelegard,

I'm keeping on learning your great tool suite and I have two more questions.

1) I read in documentation that the Merge plugin starts a new process that obviously handles the merged command part. What happens if the merged command fails/crashes? Does the main tsp process recognize the situation and restarts the process?

An example:

```
tsp -I hls --connection-timeout 5000 --receive-timeout 5000 http://192.168.1.66/files/stream/streaming_channel_0.m3u8 --live | tsstuff  --bitrate 10000000 | tsp -P regulate --pcr-synchronous \
-P merge "tsp -I hls --connection-timeout 5000 --receive-timeout 5000 http://192.168.1.66/files/stream/streaming_channel_1.m3u8 --live" \
-O file testout.ts
```

If the second tsp process (started by the merge plugin) exits/crashes, does the main tsp process (the one starting the merge) recognize it and restarts it?

I ask this because I may write a script that monitors the main tsp process and restart it if it fails, but I cannot have access over the second tsp process (started by the main tsp process), whose PID I don't know.

2) A general question / enhancement request about HLS input plugin.

I'm making experiments with the LAN connection.

`tsp -I hls --connection-timeout 60000 --receive-timeout 60000 http://192.168.1.66/files/stream/streaming_channel_0.m3u8 --live | tsstuff  --bitrate 10000000 | tsp -P regulate --pcr-synchronous -O file testout.ts`

If I disconnect the HLS server from the network, the tsp process waits for the connection or receive timeout before stopping, as expected. May it possible to have the HLS plugin trying to reconnect indefinitely until manually stopped?

If , instead, I disconnect  the device running the tsp command, no matter the connection/receive timeout value, when I reconnect the cable (before the expiration time of course) I get an error: 
`
Error: hls: download error, Couldn't connect to server, Failed to connect to 192.168.1.66 port 80: No route to host`

An then the script stops.

I would expect it should resume the download if the elapsed time is less than--receive-timeout, similarly to the first case.

Am I doing something wrong?

Thanks for your help!

lelegard, 2021-05-19:
Hi @skynetbeta23,

Let's take the points one by one.

> What happens if the merged command fails/crashes? Does the main tsp process recognize the situation and restarts the process?

Currently no. When the command terminates (normal end of stream or crash), this is considered as the end of the stream to merge. The documentation says:
> __--terminate__
> Terminate packet processing when the merged stream is terminated. By default, when packet insertion is complete, the transmission continues and the stuffing is no longer modified.

So, unless `--terminate` is specified, the processing simply continues without anything more to merge.

A new option `--restart` can be considered to endlessly restart the merge command when it stops. No distinction will be made between normal termination and crash (too ambiguous, too many corner cases).

> I may write a script that monitors the main tsp process and restart it if it fails

Usually, I use a quite simple method for that: run `tsp` from a script which does something like that:
~~~
while true; do
    tsp ....
fi
~~~
No need for a complex external monitoring. Just loop on restart when it fails. Usually, I add some precaution to avoid a CPU-bound loop when the command immediately fails on startup (invalid parameters, non-existent files, etc.) Just check the execution time and if it is less than a few seconds, you may safely assume that something goes wrong from the beginning and will likely goes wrong again next time. In that case, either exit or wait a few seconds in the hope that the non-existent resources will appear.

> A general question / enhancement request about HLS input plugin.

First, we must clearly understand the relationship between the network errors and their impacts on the TCP/IP connections at application level. The `tsp` application only deals with TCP connections (in the case of the `hls` plugin). In fact, the plugin does not really manage the TCP connections in that case. Since HLS is HTTP-based, it relies on the system-specific HTTP library (WinInet on Windows, libcurl elsewhere). So, the question you need to ask is: what is the impact of such or such network error on the system HTTP library?

About restarting, HLS is different from UDP/IP reception. Reconnecting is fine, but reconnecting to what? On a network protocol perspective, there is no such thing as an _HLS server_. We simply download the contents of files from their URL, either playlist text files or TS segment files. So, what does mean "reconnect"? Simply restart the processing will probably result in restarting from the beginning of the first segment in the current playlist. This would probably never be the right choice. At best, the plugin could remember the URL of the current segment and the index of the last received TS packet in that segment. In case of failure, restarting would mean 1) reload the playlist, 2) check if the previous current segment is still there and where, 3) if the segment is still there, reload it, skipping packets until the previous current one, 4) if the segment is no longer there, restart at the current first one.

So, the _precise_ technical specification of "restart" in the case of HLS can be quite complicated to specify and implement.

> `Error: hls: download error, Couldn't connect to server, Failed to connect to 192.168.1.66 port 80: No route to host`
> An then the script stops.
> I would expect it should resume the download if the elapsed time is less than--receive-timeout, similarly to the first case.

This is a fatal network error. In your specific case, the cable is plugged again, network datagrams are flowing again in the pipe, the current pending I/O is woken up, and since the system reconfiguration of the link is not immediately complete there is an error ("no route to host"). The current I/O was waiting for something, packet or error, to complete. You get the error. There is nothing we can do.

I understand your need: resilience and fail-over. This is the characteristics of high-quality production tools. This is quite feasible. I have developed such systems many times in a professional context and there is a incredibly high number of details and system-specific behaviours to handle. This can result in an expensive development. TSDuck is just a lab/test/demo/debug tool. It will never have that level of fine-tuning for all cases of operational errors. Some "restart" options may be added here and there to avoid a fatal termination. But remember that the devil is in the details...

skynetbeta23, 2021-05-19:
Hi @lelegard,

thanks for the fast reply.

> A new option --restart can be considered to endlessly restart the merge command when it stops. No distinction will be made between normal termination and crash (too ambiguous, too many corner cases).

That would be great, thanks if you find time to implement it.

> About restarting, HLS is different from UDP/IP reception. Reconnecting is fine, but reconnecting to what? On a network protocol perspective, there is no such thing as an HLS server. We simply [...]

I agree with your concerns. For my needs a simple reconnect action would be fine, my goal is just to avoid tsp terminating; the process could just restart. That could be a possible enhancement for the future, even if I can automatically restart the process as you suggest.

> I understand your need: resilience and fail-over. This is the characteristics of high-quality production tools. This is quite feasible. I have developed such systems many times in a professional [...]

I know we ask so much from you every time, but we find this tool so powerful that we always want more features :-D

Thanks

lelegard, 2021-05-19:
> For my needs a simple reconnect action would be fine, my goal is just to avoid tsp terminating; the process could just restart. 

If you have a monitor restarting tsp on failure (or a simple script which loops around tsp), the process will be restarted anyway. I there is no "clever HLS restart", meaning simply restarting the playlist, the result is the same.

I agree that having a restart at the input plugin level could be more elegant. But when you have that concern with the HLS input plugin, anyone could have the same request on any other input plugin. Or output plugin. Why should we complexify all plugins when a simple loop around tsp in a script can achieve a similar result?

The restart issue is different in the merge plugin since there was no tsp termination here. This is why I proposed to add the restart option. But this is limited to that specific plugin.

lelegard, 2021-05-19:
Option `--restart` added to plugin `merge`.

skynetbeta23, 2021-05-20:
Hi @lelegard ,

Thanks for the improvement.

I compiled the new version 3.27-2393 but unfortunately the merge plugin seems to not be working properly.

```
tsp -I hls --connection-timeout 60000 --receive-timeout 60000 https://streamcdnb5-8e7439fdb1694c8da3a0fd63e4dda518.msvdn.net/rainews1/hls/rainews_2400/chunklist.m3u8 --live | \
tsstuff  --bitrate 15000000 --buffer-size 16777216 | \
tsp --max-output-packets 1024 -P regulate --pcr-synchronous \
-P merge --pcr-reset-backwards "tsp -I hls --connection-timeout 60000 --receive-timeout 60000 https://nbcnews2.akamaized.net/hls/live/723426/NBCNewsPlaymaker24x7Linear99a3a827-ua/VIDEO_1_4596000.m3u8 --live -P svrename --id 2 1 -P regulate --pcr-synchronous" \
-P regulate --bitrate 15000000 \
-O file test_merge.ts 
```

The second process is correctly created and I can confirm that the --restart option works great, but the second stream is not merged at all in the main one.

Can you please verify ? Thanks

lelegard, 2021-05-20:
Hi @skynetbeta23, you are right, sorry about that. I just fixed it in the repo.

Insufficient tests indeed... The problem with the automated test suite is that it can currently include only purely deterministic tests. And `merge` is not a deterministic plugin, the placement of merged packets may vary due to timing consideration. I did manual testing this time.

skynetbeta23, 2021-05-21:
Hi @lelegard ,

Everything is fine now. Thanks again

closed 2026-04-26 by github-actions[bot]