tsduck/tsduck #791: [Q] Merge default method inaccuracy PCR increases

Petr330, 2021-05-21:
Hi Lelegard.
I want to thank you for excellent software and for your work to improve it.
I am an engineer using your program as a backup multiplexer for television broadcasting.
Plugin Merge multiplexing 4 programs constant bitrate (CBR) streams. DVB stream is generated to transmitter.
The command line has the form:
`nice -n -20 \
tsp -v -t --log-plugin-index --bitrate 10060000 --realtime --max-input-packets 161 --max-flushed-packets 161 \
-I null \
-P regulate --packet-burst 14 \
-P merge --accel 0 --pcr-reset-backwards "tsp --realtime --max-input-packets 161 --max-flushed-packets 161 -I ip xxxxxx --local-address xxxxxx -P filter -s --pid 5100-5102" \
-P merge --accel 0 --pcr-reset-backwards "tsp --realtime --max-input-packets 161 --max-flushed-packets 161 -I ip xxxxxx --local-address xxxxxx -P filter -s --pid 5200-5202" \
-P merge --accel 0 --pcr-reset-backwards "tsp --realtime --max-input-packets 161 --max-flushed-packets 161 -I ip xxxxxx --local-address xxxxxx -P filter -s --pid 5300-5302" \
-P merge --accel 0 "tsp --realtime --buffer-size-mb 1 --max-input-packets 161 --max-flushed-packets 161 -I ip xxxxxx --local-address xxxxxx -P remap 1020-1023=5400 -P svrename 0x15 --id 0x04 -P filter -s --pid 5400-5403" \
-P skip 40000 \
-P pat --create --bitrate 5000 --tsid 1 --add-service 1/5100 --add-service 2/5200 --add-service 3/5300 --add-service 4/5400 \
-P sdt --create -s 1 --name service1 --provider xxx --original-network-id 65399 --ts-id 1 \
-P sdt -s 2 --name service2 --provider xxx \
-P sdt -s 3 --name service3 --provider xxx \
-P sdt --default-charset ISO-8859-5 -s 4 --name 'service4' --provider xxx \
-P nit -c --build-service-list-descriptors --network-id 111 --network-name xxx \
-O ip --enforce-burst --packet-burst 7 udpxxxxxx --local-address xxxxxx`

The program works well, continuously 24 hours a day. Jitter is provided within as input signals. But on work there are some questions:
1. When merged transport stream the default method inaccuracy increases PCR and DTS, PMT. PCR lags lags more and more. Checked on the releases of different versions of tsduck starting from v2237 and above, version OS Ubuntu Server 20.04. The work of the TSP Plugins merge according to the incremental method is not observed.
2. Request for improvement. Merge plugin with option --pcr-reset-backwards works well, but I have in one of the services 'Service4' PID PCR and video different. Is it possible to improve the --pcr-reset-backwards option so as to indicate PCR PID and PID video or just PID video if they are different.

I carefully read your messages. In many ways, the command line is made according to your recommendations. Parameters - --max-input-packets 161 --max-flushed-packets 161 and others were selected experimentally for the minimum jitter at the output. There is a problem with the stability of the output signal with any small activity of the OS (ubuntu-server) for example, the search for updates or restarting services at the beginning of the day at the output appears jumps of a jitter stream, although the load on the processor is not prevailing to 10%. The nice -n -20 command has improved a bit. Do not consider the arrogance, I know that you do not want yet or cannot improve tsduck for the multiplexer. I am not a programmer's engineer. The question is whether there is an opportunity to stabilize the output DVB stream or to decide its dependence on small loads on the processor or make a de-jitter plugin. I will also be grateful for recommendations for improving or changing above the specified command line.
I apologize for not accuracy in English. English I know weak. Translated Google.
I add a screenshot fixed jump values of PCR. If you need recorded TS streams, specify where to download there are records of 60 s. It is about 75 MB.
[image]

lars18th, 2021-05-22:
Hi,

From y point of view the cause of this "jitter" (that actually is an added delta over time) is caused because the merged streams are increasing some delay. Perhaps as the TSDuck uses several threads the subprocess then are lossing the synchronization of with the master thread.

If this is true, then perhaps a `--flush-sync` parameter in the `merge` plugin can provide a workaround. The idea is to flush all packets at the interval indicated by this "flush-sync". Or another option could be to define a max-buffer size and enforce the flushing when it arrives to this limit.

You (@lelegard) think that this assumption is true?

Petr330, 2021-05-22:
Hello.
Your assumption is very similar to the truth. When checking on a more powerful computer, only two services were lost synchronization - "Service2" (where the increased frequency of PCR) and "Service4" with non-standard PID PCR were lost. Yes, and TSP Plugins Merge with the --incremental-pcr-restamp option perfectly works without losing synchronization on both computers. The only thing that upsets me, I cannot for the "Service4" to perform --pcr-reset-backwards.

lars18th, 2021-05-22:
Hi @Petr330 ,

In this case, until @lelegard will finish his implementation of the more powerful muxer (this is on the road, but I don't know anything about ETA) you can try this way to a more robust "muxing":

- Instead of directly merging the four services, create 4 different processes that generate each one 1 SPTS with CBR injecting your services and generating all the service data. You can output these streams to the network (multicast or unicast).
- After that create a new process that does ONLY the merging of these four services. When the input are CBR it will be more easy to complete the processing without lossing packets. You now send it another time to the network.

Using this technique if you don't have sufficient power in your server to guarantee the outcomes then you can use more than one (in fact until 5 servers). That's the solution that I can suggest with the current implementation of the TSDuck. Another option is to do similar (create the SPTS using TSDuck) and use one external muxer.

Regards.

Petr330, 2021-05-23:
Hi, lars18th.
Thanks for the recommendations. I plan to optimize the command line and reduce the number of plug-ins and according to processes. Technically, I have a solution. I am route the MPTS from the server output to the IRD 2900, which converts the signal to ASI and then on the DVB-T2 transmitter. It works well. The MPTS filed directly periodically (3-5 times a day) by the transmitter log of "Dropping TS Packets, Fail" has signal skipping.

lars18th, 2021-05-24:
> The MPTS filed directly periodically (3-5 times a day) by the transmitter log of "Dropping TS Packets, Fail" has signal skipping.

This indicates that the server that is generating the MPTS has troubles to maintain the realtime flow.

Petr330, 2021-05-24:
> This indicates that the server that is generating the MPTS has troubles to maintain the realtime flow.

lars18th, 2021-05-25:
Hi @Petr330 ,

In realtime processing, sometimes the root cause is not in the CPU power, but in other components. For example, please check you NIC. The physical device or the driver configuration can have a direct impact. Another option is to use some buffering. For example, if the input comes from the network, try to buffer around 1 second and the use the output to do the mux.

Petr330, 2021-05-25:
> Another option is to use some buffering. For example, if the input comes from the network, try to buffer around 1 second and the use the output to do the mux.

lars18th, 2021-05-26:
Hi @Petr330 ,

Make the adjust of a server to run in realtime is not a simple/easy task. My suggestion is to first do a _lot_ of testing. For example, please try first to execute a simple _restreaming_ from the network to the network (using a simple `tsp -I ip -O ip`). And with an external tool check if the stream is adding some delay overtime. If this check fails, then you will need to adjust the OS. But if this test goes right, then the problem is with the TSDuck processing.

Try to check one-after-one.

Petr330, 2021-05-26:
Hi @lars18th.
Thank you. This is a good idea. I'll try to do that.

Petr330, 2021-06-01:
Conducted tests (3 days of work) tsp -I ip -O ip. Also observes a delay effect of 12-14 ms when the activity of OS 3-4 times a day appears. Normal TC flow delay up to 3 ms. ОS: Ubuntu-server 20.04, CPU: Intel(R) Core(TM) i3-3220 3.30GHz, System memory size: 8GiB. Settings NIC and socket buffer size to maximum buffer.

lars18th, 2021-06-01:
Hi @Petr330 ,

Are you sure that you're using a plain `tsp -I ip -O ip`? For example, if you include some `-P regulate` then this is the cause of the delays. However, if you're really using the clean network input to network output, then something requires to be adjusted.

So, please @lelegard , can you try to identify the problem? Perhaps the tsp tool requires to incorporate some "clock drift correction" to overcome these large executions. What you think?

Petr330, 2021-06-01:
I used a command line:
`tsp --realtime -v -t -I ip 226.0.0.7:1234 -b 8388608 --local-address 192.168.200.24  -O ip --enforce-burst --packet-burst 7 226.4.4.5:1234 --local-address 192.168.200.24`
PCR measurements carried out so:
`tsp   -v -t --realtime  -I ip 226.4.4.5:1234 --local-address 192.168.200.24 -P pcrverify -t --input-synchronous -j 4000 -O drop`
I think  need to make the output buffer and then hard synchronized by the external TSP -O IP source at least for CBR flows. This would remove the jitter as a positive and minus and no envy of the jitter input signals. Well, this is a dilettante opinion in programming. I understand that it is not easy.

lelegard, 2021-06-01:
> can you try to identify the problem? Perhaps the tsp tool requires to incorporate some "clock drift correction" to overcome these large executions. What you think?

For a basic `tsp -I ip -O ip` (fine tuning options below), there is nothing to "correct". There are two threads: one for input, one for output.

- Input thread:
  - Blocking read on UDP socket until a UDP datagram is received.
  - Write the extracted TS packets in the global buffer
  - Signal the presence of new packets in the global buffer
  - Loop back
- Output thread:
  - Wait until some packets are available in the buffer
  - Send the packets in UDP datagrams
  - Signal the presence of free space in the global buffer.
  - Loop back

Nothing is done on the TS packets, no additional delay is introduced.

Proposed fine tuning options: `tsp --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7`

lars18th, 2021-06-01:
Hi @Petr330 ,

> I used a command line:
> `tsp --realtime -v -t -I ip 226.0.0.7:1234 -b 8388608 --local-address 192.168.200.24 -O ip --enforce-burst --packet-burst 7 226.4.4.5:1234 --local-address 192.168.200.24`

Sorry but this is not a **clean** test. You need to test this:
`tsp -I ip 226.0.0.7:1234 --local-address 192.168.200.24 -O ip 226.4.4.5:1234 --local-address 192.168.200.24`

Why you need to remove the bitrate, for example? Because this is not relevant in this flow. The clock is derived from the incoming packets. And you don't want to add any delay. Futhermore, after this _initial test_ try to add the fine tuning options recommended by @lelegard . Then the output packets will be more _smothed_ instead of chunks. But anyway what you need to check is if the TSDuck process is adding some _accumulative delay_. And to not add it, it's mandatory to not generate some waiting (for example with `-P regulate` or similar filters).

I hope this helps you.

Petr330, 2021-06-01:
Hi @lelegard .
Thank you for responding. I will definitely conduct tests with fine tuning and write the results.
I have a question to improve the TS plugin merge.
Request for improvement. Merge plugin with option --pcr-reset-backwards works well, but I have in one of the services 'Service4' PID PCR and video different. Is it possible to improve the --pcr-reset-backwards option so as to indicate PCR PID and PID video or just PID video if they are different.
Perhaps this to make.

lelegard, 2021-06-01:
> Request for improvement. Merge plugin with option --pcr-reset-backwards works well, but I have in one of the services 'Service4' PID PCR and video different. Is it possible to improve the --pcr-reset-backwards option so as to indicate PCR PID and PID video or just PID video if they are different.

It already works that way. For each service, the declared PCR PID (in the PMT) is used as reference. We never assume that the PCR and PTS/DTS are on the same PID. The option `--pcr-reset-backwards` means that we decide to reset the PCR adjustment on a PID when a PTS or DTS _in any other PID using this PID as PCR PID_ would move backward the PCR. [See the implementation here](https://github.com/tsduck/tsduck/blob/55ad652476a796d195ac70994f55e6569e2665d0/src/libtsduck/dtv/tsPCRMerger.cpp#L131)

Petr330, 2021-06-01:
I have a version of TsDuck  - Version 3.26-2349 this feature is implemented or I need to upgrade to the latest version of GitHub.
Thanks for the info.

lelegard, 2021-06-01:
> I have a version of TsDuck - Version 3.26-2349 this feature is implemented or I need to upgrade to the latest version of GitHub.

The behaviour I described above is present in version -2349.

Petr330, 2021-06-02:
Hi.
Made tests with a new version of TSDuck version 3.27-2409. After 12 hours of Service4, the service does not work option --pcr-reset-backwards PCR the value starts to be ahead of the PTS DTS. 3 other services work normally. The test continues.

Start
[image]

After 12 hours
[image]

Petr330, 2021-06-06:
Hi @lelegard.
I looked at your link to work tsPCRMerger.cpp. For my case, when the PCR PID 5403 is specified in PMT, and the PID video 5401 condition in line 139 (if (it->second->pcr_pid == pid)) will never be fulfilled and will not reset the PCR, the --pcr-reset-backwards does not work. According to this, please add the --target-pid option to indicate the target flow with DTS/PTS. PID with PCR is already specified in PMT. I understand that the program will complicate as it is necessary to work with two threads. I tried the work of the -P pcrcopy plugin and insert PCR in the PID video and changed the PMT, option --pcr-reset-backwards is works, but synchronization is not high quality.

lelegard, 2021-06-06:
> For my case, when the PCR PID 5403 is specified in PMT, and the PID video 5401 condition in line 139 (if (it->second->pcr_pid == pid)) will never be fulfilled and will not reset the PCR, the --pcr-reset-backwards does not work.

I think the code is correct. In each service, there is one PCR PID, identified in the PMT. It can be the video PID or a dedicated PID (a dedicated PID in your case but it does not make a difference). This code is executed when a packet from the PCR PID contains a PCR. The code looks at all other PID's in the services (or more generally all PID's which use that PCR PID if it is shared) and if any PTS or DTS in any _other_ PID (for instance your video PID) would become lower than the adjusted PCR.

So, if the video PID and PCR PID are dissociated, it works. The problem would be the same with a video PID containing PCR's because audio PID's also contain PTS/DTS and they are similarly compared to the PCR in the video PID (or any other PID which is declared in the PMT as PCR PID).

Petr330, 2021-06-06:
Hi @lelegard.
In learned faster to make tests. I make a bitrate of the main stream a little less than attached. Forcing faster to work --pcr-reset-backwards and not wait 12 hours. Flow I join SPTS. Here are screenshots:
pcr pid and pcr video in PMT  match, --pcr-reset-backwards is works
[image]
pcr pid and pcr video in PMT do not match, --pcr-reset-backwards does not work
[image]

You `write:

> The code looks at all other PID's in the services (or more generally all PID's which use that PCR PID if it is shared) and if any PTS or DTS in any other PID (for instance your video PID) would become lower than the adjusted PCR, then the "reset backward" is applied.

The test shows that --pcr-reset-backwards works only with video and then if his PID coincides with PCR PID.
It also observes this in the messages of the tsp in mode --debug=2.

lelegard, 2021-06-06:
When you mean that the PCR PID is different from the video PID, that means that:
- The PCR PID is declared as such in the PMT.
- There is no PCR in the video PID.

Is that the case in your stream? You can verify both information with `tsanalyze`.

Petr330, 2021-06-06:
> When you mean that the PCR PID is different from the video PID, that means that:
    The PCR PID is declared as such in the PMT.
    There is no PCR in the video PID.

Yes it is!

lelegard, 2021-06-06:
Could you provide materials to reproduce this, TS file and tsp command or script?

Petr330, 2021-06-06:
The tsp command
`tsp -d -v -t --log-plugin-index --bitrate 2480000 --realtime --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 \
-I null \
-P regulate \
-P merge --restart --accel 0 --pcr-reset-backwards --incremental-pcr-restamp "tsp --realtime --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 \
-I ip 225.0.0.2:22222 -b 16777216 --local-address 192.168.200.24 -P remap 1020-1023=5400 -P svrename 0x15 --id 0x04" \
-P pat --create --bitrate 5000 --tsid 2 --add-service 4/5400 \
-P sdt --create -s 4 --default-charset ISO-8859-5 --name 'Service4' --original-network-id 10 --ts-id 2 \
-P skip 40000 \
-O ip --enforce-burst --packet-burst 7 226.0.0.4:4444 --local-address 192.168.200.24`

Information with tsanalyze:
[image]
[image]

lelegard, 2021-06-07:
> What duration you need a TS file 

I cannot tell. Enough to reproduce the issue. If you want any problem to be analyzed (I mean _any_ problem, not only about TSDuck), you should provide the materials so that someone else can reproduce the problem in his own environment. In your case, that probably means one TS file and a few scripts.

First, make sure that the provided materials are actually sufficient to reproduce the problem. The most reliable way is that you first run the scripts yourself and ensure that the problem is still there. Then, you may transmit these materials and safely assume that the receiver will be able to analyze the problem.

> where to upload it?

Any public file transfer site such as wetransfer.com is fine.

Petr330, 2021-06-07:
Hi @lelegard.
Thanks for the explanation. Command line for test. 
`tsp -d -v -t --log-plugin-index --bitrate 2480000 --realtime --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 \
-I null \
-P regulate \
-P merge --restart --accel 0 --pcr-reset-backwards --incremental-pcr-restamp "tsp -I file --infinite /home/user1/tv5_ip225_070621.ts -P pcrbitrate --dts -P regulate -P remap 1020-1023=5400 -P svrename 0x15 --id 0x04" \
-P pat --create --bitrate 5000 --tsid 2 --add-service 4/5400 \
-P sdt --create -s 4 --default-charset ISO-8859-5 --name 'Service4' --original-network-id 10 --ts-id 2 \
-P skip 40000 \
-O ip --enforce-burst --packet-burst 7 226.0.0.4:4444 --local-address 192.168.200.24`

Link  [https://we.tl/t-9FNx7fIURE](url)  to the TS file with which I spent the test. The problem with the option in the test is.

lelegard, 2021-06-07:
Hi @Petr330 

Thanks for the files. There was something, indeed. A nasty bug in the `merge` plugin. Something silly, calling the PSI merger before the PCR merger. In some cases, the PSI merger needs to remove PAT packets from the merged stream. The PCR merge no longer received the PAT, couldn't locate the PMT, couldn't make PID / PCR associations. This must have impacted audio PID's too, even if the PCR were in the video PID. Inverting the invocation order of PSI and PCR mergers fixed it.

I fixed it in commit e9059404c6be0edf4279a17960969aa45550d678, version 3.27-2419.

Petr330, 2021-06-07:
Hi @lelegard 
I am very glad that I could help. I need some time to conduct tests on live streams. Thanks for your work.

Petr330, 2021-06-21:
Hi @lelegard 
I did several tests on real option --pcr-reset-backwards signals now working well but noticed its frequent triggering. After testing different variants, I found an error in the calculation of PCR when the signals were merged.
Description:    Ubuntu 20.04.2 LTS
tsp:                TSDuck - version 3.27-2419       compiling on OS
Сommand line:
`tsp -d2 -v -t --log-plugin-index --bitrate 10060000 --realtime --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 \
-I null \
-P regulate --packet-burst 14 \
-P merge --accel 0 --transparent "tsp --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I file /home/user1/test_spts_z_210621.ts -P pcrbitrate -P regulate" \
-P skip 40000 \
-P until --seconds 1100 \
-O ip --enforce-burst --packet-burst 7 226.0.0.4:4444 --local-address 192.168.200.24 &>>log_d2_test_spts_z_210621.txt`

Fragments of the file log tsp:
`1
* 2021/06/21 16:38:00 - Debug[2]: merge[2]: adjusted PCR by -97,225 (-3 ms) in PID 0x13ED (5101)
* 2021/06/21 16:38:01 - Debug[2]: merge[2]: adjusted PCR by +12,362 (+0 ms) in PID 0x13ED (5101)
* 2021/06/21 16:38:01 - Debug[2]: merge[2]: adjusted PCR by -72,091 (-2 ms) in PID 0x13ED (5101)
* 2021/06/21 16:38:01 - Debug[2]: merge[2]: adjusted PCR by -95,995 (-3 ms) in PID 0x13ED (5101)
* 2021/06/21 16:38:01 - Debug[2]: merge[2]: adjusted PCR by -31,095 (-1 ms) in PID 0x13ED (5101)
* 2021/06/21 16:38:01 - Debug[2]: merge[2]: adjusted PCR by -95,365 (-3 ms) in PID 0x13ED (5101)
* 2021/06/21 16:38:01 - Debug[2]: merge[2]: adjusted PCR by -18,336,717,865 (-679,137 ms) in PID 0x13ED (5101)
* 2021/06/21 16:38:01 - Debug[2]: merge[2]: adjusted PCR by -18,336,798,281 (-679,140 ms) in PID 0x13ED (5101)
* 2021/06/21 16:38:01 - Debug[2]: merge[2]: adjusted PCR by -18,336,680,905 (-679,136 ms) in PID 0x13ED (5101)
* 2021/06/21 16:38:01 - Debug[2]: merge[2]: adjusted PCR by -18,336,761,321 (-679,139 ms) in PID 0x13ED (5101)
* 2021/06/21 16:38:01 - Debug[2]: merge[2]: adjusted PCR by -18,336,821,271 (-679,141 ms) in PID 0x13ED (5101)

2
* 2021/06/21 16:49:20 - Debug[2]: merge[2]: adjusted PCR by -18,336,818,686 (-679,141 ms) in PID 0x13ED (5101)
* 2021/06/21 16:49:20 - Debug[2]: merge[2]: adjusted PCR by -18,336,721,492 (-679,137 ms) in PID 0x13ED (5101)
* 2021/06/21 16:49:20 - Debug[2]: merge[2]: adjusted PCR by -18,336,801,909 (-679,140 ms) in PID 0x13ED (5101)
* 2021/06/21 16:49:20 - Debug[2]: merge[2]: adjusted PCR by -18,336,684,532 (-679,136 ms) in PID 0x13ED (5101)
* 2021/06/21 16:49:20 - Debug[2]: merge[2]: adjusted PCR by -36,673,488,679 (-1,358,277 ms) in PID 0x13ED (5101)
* 2021/06/21 16:49:20 - Debug[2]: merge[2]: adjusted PCR by -36,673,544,876 (-1,358,279 ms) in PID 0x13ED (5101)
* 2021/06/21 16:49:20 - Debug[2]: merge[2]: adjusted PCR by -36,673,451,719 (-1,358,275 ms) in PID 0x13ED (5101)
* 2021/06/21 16:49:20 - Debug[2]: merge[2]: adjusted PCR by -36,673,532,135 (-1,358,278 ms) in PID 0x13ED (5101)`

[image]

Add options TSP plugin merge  --incremental-pcr-restamp
[image]

You can download a 20-minute test file here:
https://we.tl/t-uzOj6IBMVB

lelegard, 2021-06-27:
Hi,

Thanks for the sample TS. The problem comes from the existing PCR in file `test_spts_z_210621.ts`. At some points, there are huge PCR discontinuities, a jump of approximately 10 minutes in the future. This was not correctly handled by the `merge` plugin. I fixed this. Now, in case PCR discontinuity in the input, the PCR adjustment is reset, restarting from the new input PCR value.

Interestingly, if the discontinuity was in the past instead of the future, the problem did not happen. The side effect of a jump in the past was that the adjusted PCR became ahead of the input PTS/DTS and the PCR adjustment was automatically reset.

Petr330, 2021-09-22:
Hi, @lelegard .
I spent tests option --incremental-pcr-restamp is disabled (the default method restamping PCR's), remarked the frequent reset of the PCR calculation depends on the value bitrate in the main stream.
Description: Ubuntu 20.04.2 LTS
tsp: TSDuck - version 3.28-2534 compiling on OS
Сommand line:
`tsp -v -t --log-plugin-index --bitrate 30000000 --realtime --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 \
-I null \
-P regulate --packet-burst 14 \
-P merge --accel 0 --transparent "tsp --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I file /home/user1/test_spts_z_210621.ts -P pcrbitrate -P regulate" \
-P skip 40000 \
-P until --seconds 1100 \
-O ip --enforce-burst --packet-burst 7 226.0.0.4:4444 --local-address 192.168.220.22`
Fragments of the file log 1 tsp:

`* 2021/09/22 11:36:03 - regulate[1]: minimum packet burst duration is 10,000,000 nano-seconds
* 2021/09/22 11:36:03 - null[0]: initial input bitrate is 30,000,000 b/s
* 2021/09/22 11:36:03 - regulate[1]: regulated at bitrate 30,000,000 b/s
* 2021/09/22 11:36:03 - merge[2]: main stream bitrate reset to 30,000,000 b/s (was 0 b/s)
* 2021/09/22 11:36:03 - merge[2]: merged stream bitrate reset to 2,560,000 b/s (was 0 b/s)
* 2021/09/22 11:37:57 - merge[2]: resetting PCR restamping in PID 0x13ED (5101) after possible 
* 2021/09/22 11:39:51 - merge[2]: resetting PCR restamping in PID 0x13ED (5101) after possible discontinuity in original PCR
* 2021/09/22 11:41:45 - merge[2]: resetting PCR restamping in PID 0x13ED (5101) after possible discontinuity in original PCR
* 2021/09/22 11:43:39 - merge[2]: resetting PCR restamping in PID 0x13ED (5101) after possible discontinuity in original PCR
* 2021/09/22 11:45:33 - merge[2]: resetting PCR restamping in PID 0x13ED (5101) after possible discontinuity in original PCR
* 2021/09/22 11:47:27 - merge[2]: resetting PCR restamping in PID 0x13ED (5101) after possible discontinuity in original PCR
* 2021/09/22 11:49:20 - merge[2]: resetting PCR restamping in PID 0x13ED (5101) after possible discontinuity in original PCR
* 2021/09/22 11:51:14 - merge[2]: resetting PCR restamping in PID 0x13ED (5101) after possible discontinuity in original PCR
* 2021/09/22 11:53:08 - merge[2]: resetting PCR restamping in PID 0x13ED (5101) after possible discontinuity in original PCR`

After analyzing the operation tsPCRMerger.cpp and the PCR calculation, I found a periodic overflow of the long int "(((main_packet_index - base_pkt) * PKT_SIZE_BITS * SYSTEM_CLOCK_FREQ) / main_bitrate).toInt()". This value goes into a negative area, which leads to discontinuity in original PCR.

tsPCRMerger.cpp
`string 127.  ctx->last_pcr = base_pcr + (((main_packet_index - base_pkt) * PKT_SIZE_BITS * SYSTEM_CLOCK_FREQ) / main_bitrate).toInt();`

I changed the calculation of the PCR replaced the string 127.

uint64_t fact_pcr = ((PKT_SIZE_BITS * SYSTEM_CLOCK_FREQ * 1000) / main_bitrate).toInt();
uint64_t delta_pcr = (main_packet_index - base_pkt) * fact_pcr / 1000;
ctx->last_pcr = base_pcr + delta_pcr;

Multiplication by 1000 (three characters are taken into account during the division) are necessary to increase the accuracy of the calculation.
In this version of tsPCRMerger.cpp, no values ​​overflow. Maybe you will make the best solution?

Fragments of the file log 2 tsp:

`* 2021/09/22 12:44:44 - regulate[1]: minimum packet burst duration is 10,000,000 nano-seconds
* 2021/09/22 12:44:44 - null[0]: initial input bitrate is 30,000,000 b/s
* 2021/09/22 12:44:44 - regulate[1]: regulated at bitrate 30,000,000 b/s
* 2021/09/22 12:44:44 - merge[2]: main stream bitrate reset to 30,000,000 b/s (was 0 b/s)
* 2021/09/22 12:44:44 - merge[2]: merged stream bitrate reset to 2,560,000 b/s (was 0 b/s)`

You can download a test file here:
https://we.tl/t-NX8GwdTNfb

lelegard, 2021-09-22:
Hi @Petr330 

Could you please try again after rebuilding TSDuck this way (and revert to the original code):
~~~
make distclean
make -j10 BITRATE_FLOAT=1
~~~
This way, bitrate values are implemented as double float instead of 64-bit fixed-point integers.

There are recurrent overflow issues in corner cases. I wonder if I should switch by default to a double float representation.

lelegard, 2021-09-22:
In fact, the code at line 127 should be:
~~~
        ctx->last_pcr = base_pcr + ((BitRate(main_packet_index - base_pkt) * PKT_SIZE_BITS * SYSTEM_CLOCK_FREQ) / main_bitrate).toInt();
~~~
The conversion `BitRate(main_packet_index - base_pkt)` ensures that floats are used before the integer overflow.

Petr330, 2021-09-23:
Hi, @lelegard .
I spent tests. Compiled the program TSDuck as you suggested.
~~~
make distclean
make -j10 BITRATE_FLOAT=1 NOPCSC=1 NOCURL=1 NODTAPI=1 NOTELETEXT=1 deb
~~~
 With a bitrate of 30,000,000 test passed well, but the same team with a bitrate of 10060000 with time PCR begins to quickly accumulate a mistake.
You can download a log file --debug=2  the original code here:
https://we.tl/t-5ZhUI3O8UP
The same test with edited file as I indicated above gave a good result.
Compiled the program
~~~
make distclean
make -j10  NOPCSC=1 NOCURL=1 NODTAPI=1 NOTELETEXT=1 deb
~~~
You can download a log file --debug=2  the edit code here:
https://we.tl/t-n5AHUxf15D

lelegard, 2021-09-23:
> Compiled the program TSDuck as you suggested.

Did you also apply the suggested modification first:

> The conversion `BitRate(main_packet_index - base_pkt)` ensures that floats are used before the integer overflow.

Without this, the integer overflow remains. With the conversion first, the conversion to float occurs before ` * PKT_SIZE_BITS * SYSTEM_CLOCK_FREQ` and no overflow occurs (the maximum value of a 64-bit float is around 10^308).

Anyway, to get rid of intermediate integer overflows in computations involving bitrates, I am in the process of switching the floating-point model as the default one for bitrates. See the section 2.2 of the user's guide for the bitrate representation dilemma. Based on experience from many users, it seems that the floating point model is the most robust one.

Petr330, 2021-09-23:
The conversion BitRate(main_packet_index - base_pkt) not tried. 
Now I will spend the test.

Petr330, 2021-09-23:
I wanted to get your advice. Nowhere in technical literature I can not find a restriction on the setting of PTS from PTS. TSDuck limits this value as SYSTEM_CLOCK_SUBFREQ = 90,000 or 1 second. Can this value increase to 2 seconds?

lelegard, 2021-09-23:
>  Nowhere in technical literature I can not find a restriction on the setting of PCR from PTS. TSDuck limits this value as (pdts - subpcr) > SYSTEM_CLOCK_SUBFREQ = 90,000 or 1 second.

This is arbitrary, indeed.

The idea is to detect when the PCR adjustment becomes unrealistic, meaning that the input PCR may have jumped (as well as the PTS/DTS) with some discontinuity. In case of such a jump, we must stop adjusting the PCR based on the previous values of the PCR because the PTS/DTS would go out of sync.

Usually the difference between the PTS/DTS and the PCR is a few milliseconds. A difference of 1 seconds seems unrealistic to me. That means that the video codec would need to buffer several megabits of data in the decoding pipe. That's why I considered that 1 second was a realistic indication of a synchronization issue. But if some video codec specialist can give another value based on current video codec technology, I will change it.

Petr330, 2021-09-24:
I spent several tests including with real signals. Results can be found in the logs. Changing the formula for calculating PCR "(BitRate(main_packet_index - base_pkt))" did not affect the payoff. Over time, the error accumulates. In my version of the PCR calculation, the error accumulates on a lot of slower.

You can download a log file --debug=2 test 1 SPTS:
https://we.tl/t-Ab6nZtcUdv

log file --debug=2 test 2 TS realtime Mux 4-SPTS 30 minute. BITRATE_FLOAT=1, BitRate(main_packet_index - base_pkt):
https://we.tl/t-w4VqNdlvSF

log file --debug=2 test 3 TS realtime Mux 4-SPTS 30 minute. BITRATE_FLOAT=0, my version of the PCR calculation:
https://we.tl/t-U9xeAMjlyu

lelegard, 2021-09-28:
> Changing the formula for calculating PCR "(BitRate(main_packet_index - base_pkt))" did not affect the payoff.

This does not make sense with floating point representation of bitrates.  Please check with the following command:
~~~
$ tsversion --version=bitrate
64-bit fixed-point value with up to 1 decimals
~~~
or
~~~
$ tsversion --version=bitrate
64-bit floating-point value
~~~
I have definitely switched the default representation of bitrates to floating-point. The latest state of the repo should be the new base.

Petr330, 2022-02-23:
Hi, @lelegard .
I found out what the problem of the accuracy of the calculation of PCR in my case. The difference in the values of the bitrate when calculating PCR and the reality of the thread. Check it is very simple.
~~~
# tsversion --version=bitrate
64-bit floating-point value
	# tsp -v -t --bitrate 10060000 -I null -P regulate -P bitrate_monitor -p 1 -t 1 -O drop
* 2022/02/23 10:01:13 - regulate: minimum packet burst duration is 10,000,000 nano-seconds
* 2022/02/23 10:01:13 - null: initial input bitrate is 10,060,000 b/s
* 2022/02/23 10:01:13 - regulate: regulated at bitrate 10,060,000 b/s
* 2022/02/23 10:01:15 - bitrate_monitor: 2022/02/23 10:01:15, TS bitrate: 10,076,800 bits/s
* 2022/02/23 10:01:16 - bitrate_monitor: 2022/02/23 10:01:16, TS bitrate: 10,076,800 bits/s
* 2022/02/23 10:01:17 - bitrate_monitor: 2022/02/23 10:01:17, TS bitrate: 10,076,800 bits/s
* 2022/02/23 10:01:18 - bitrate_monitor: 2022/02/23 10:01:18, TS bitrate: 10,076,800 bits/s
^C* 2022/02/23 10:01:18 - tsp: user interrupt, terminating...
~~~
When calculating PCR, the value of the bitrate is 10060000. The real value of the generated flow bitrate is 10076800 hence the inaccuracy of PCR.
For TSDuck version with integers, accuracy is higher because the bitrate value is closer.
~~~
	# tsversion --version=bitrate
64-bit unsigned integer value
	# tsp -v -t --bitrate 10060000 -I null -P regulate -P bitrate_monitor -p 1 -t 1 -O drop
* 2022/02/23 10:02:26 - regulate: minimum packet burst duration is 10,000,000 nano-seconds
* 2022/02/23 10:02:26 - null: initial input bitrate is 10,060,000 b/s
* 2022/02/23 10:02:26 - regulate: regulated at bitrate 10,060,000 b/s
* 2022/02/23 10:02:28 - bitrate_monitor: 2022/02/23 10:02:28, TS bitrate: 10,018,144 bits/s
* 2022/02/23 10:02:29 - bitrate_monitor: 2022/02/23 10:02:29, TS bitrate: 10,058,752 bits/s
* 2022/02/23 10:02:30 - bitrate_monitor: 2022/02/23 10:02:30, TS bitrate: 10,060,256 bits/s
* 2022/02/23 10:02:31 - bitrate_monitor: 2022/02/23 10:02:31, TS bitrate: 10,060,256 bits/s
* 2022/02/23 10:02:32 - bitrate_monitor: 2022/02/23 10:02:32, TS bitrate: 10,060,256 bits/s
* 2022/02/23 10:02:33 - bitrate_monitor: 2022/02/23 10:02:33, TS bitrate: 10,060,256 bits/s
* 2022/02/23 10:02:34 - bitrate_monitor: 2022/02/23 10:02:34, TS bitrate: 10,060,256 bits/s
* 2022/02/23 10:02:35 - bitrate_monitor: 2022/02/23 10:02:35, TS bitrate: 10,058,752 bits/s
* 2022/02/23 10:02:36 - bitrate_monitor: 2022/02/23 10:02:36, TS bitrate: 10,060,256 bits/s
* 2022/02/23 10:02:37 - bitrate_monitor: 2022/02/23 10:02:37, TS bitrate: 10,060,256 bits/s
^C* 2022/02/23 10:02:37 - tsp: user interrupt, terminating...
~~~
Maybe someone will be useful to this information. To increase the accuracy of the calculation of PCR, you must select the value of the bitrate with an integer package for 1 second.
For example:
PKT_SIZE = 188
PKT_SIZE_BITS = 8 * PKT_SIZE = 1504
bitrate = 70 * 1504 = 10528000
~~~
	# tsp -v -t --bitrate 10528000 -I null -P regulate -P bitrate_monitor -p 1 -t 1 -O drop
* 2022/02/23 10:01:33 - regulate: minimum packet burst duration is 10,000,000 nano-seconds
* 2022/02/23 10:01:33 - null: initial input bitrate is 10,528,000 b/s
* 2022/02/23 10:01:33 - regulate: regulated at bitrate 10,528,000 b/s
* 2022/02/23 10:01:35 - bitrate_monitor: 2022/02/23 10:01:35, TS bitrate: 10,528,000 bits/s
* 2022/02/23 10:01:36 - bitrate_monitor: 2022/02/23 10:01:36, TS bitrate: 10,528,000 bits/s
* 2022/02/23 10:01:37 - bitrate_monitor: 2022/02/23 10:01:37, TS bitrate: 10,528,000 bits/s
* 2022/02/23 10:01:38 - bitrate_monitor: 2022/02/23 10:01:38, TS bitrate: 10,528,000 bits/s
* 2022/02/23 10:01:39 - bitrate_monitor: 2022/02/23 10:01:39, TS bitrate: 10,528,000 bits/s
^C* 2022/02/23 10:01:39 - tsp: user interrupt, terminating...
~~~
In this case, the accuracy of the calculation of PCR is very high for any assembly of TSDuck.
I want to thank you once again for your work and excellent software.

closed 2022-02-28 by Petr330