tsduck/tsduck #792: EPG import[Q]

gregor20sw, 2021-05-22:
Hello, can you give me an example of how I import an xml with epg into a .ts file, and how I enter utc time. Many Thanks

lars18th, 2021-05-22:
Hi @gregor20sw ,

Just to remark: The TSDuck toolkit doesn't supports the XMLTV format. It only supports the own XML format describing the tables with EIT events. If you want to use the first, then you can't do it. If you want the to use the second, then please read the documentation. It has examples.

I hope this helps.

gregor20sw, 2021-05-22:
Hi **@lars18th**
Many Thanks
I exported EIT as xml, and did not manage to import it back into a .ts can you please give me a few connection points what a command looks like. or where to go for a meal. how can I import EIT.

lars18th, 2021-05-22:
Hi,

> I exported EIT as xml,

Using TSDuck? Then OK.

> and did not manage to import it back into a .ts can you please give me a few connection points what a command looks like. or where to go for a meal. how can I import EIT.

If you have some EIT data in a XML file, then to "insert" it in the TS you need to use the `inject` plugin. Please, read the documentation and search for this plugin: https://tsduck.io/download/docs/tsduck.pdf

Regards.

gregor20sw, 2021-05-22:
can I create my own mpegts using ffmpeg own EIT EPG and insert it into the .ts files using TSDuck?

I've been trying some information in my self-made mpegts files for days

so I extract the eit from a tv stream
`tsp -I http http://ip:8001/1:0:19:2B5C:41B:1:FFFF0000:0:0:0: -P until --seconds 10 | tstables --pid 18  --xml-output /home/EPG/test.xml`

so i import it again
`tsp -I file /home/13261.ts -r 1 -P inject --xml /home/EPG/test.xml --replace --pid 18 -O file /home/13261.ts`

the file gets bigger, but i can't read it out
`tsp -I file /home/13261.ts  -P tables -O drop`

it is the same output as before the inject
```
* SDT Actual, TID 66 (0x42), PID 17 (0x0011)
  Version: 0, sections: 1, total size: 40 bytes
  - Section 0:
    Transport Stream Id: 1 (0x0001)
    Original Network Id: 65281 (0xFF01)
    Service Id: 1 (0x0001), EITs: no, EITp/f: no, CA mode: free
    Running status: running
    - Descriptor 0: Service (0x48, 72), 18 bytes
      Service type: 0x01 (Digital television service)
      Service: "Service01", Provider: "FFmpeg"

* PAT, TID 0 (0x00), PID 0 (0x0000)
  Version: 0, sections: 1, total size: 16 bytes
  - Section 0:
    TS id:       1 (0x0001)
    Program:     1 (0x0001)  PID: 4096 (0x1000)

* PMT, TID 2 (0x02), PID 4096 (0x1000)
  Version: 0, sections: 1, total size: 26 bytes
  - Section 0:
    Program: 1 (0x0001), PCR PID: 256 (0x0100)
    Elementary stream: type 0x1B (AVC video), PID: 256 (0x0100)
    Elementary stream: type 0x0F (MPEG-2 AAC Audio), PID: 257 (0x0101)

```

closed 2021-05-23 by gregor20sw