tsduck/tsduck #793: [Q] is it possible to shift audio position in TS bitstream forward

clime, 2021-05-22:
Hello,

for AV synchronization testing purposes, I would like to move all audio packets forward in TS bitstream. Is it possible? I.e. i would like to maintain PTS and everything but I want to delay sending audio with respect to video. I am not very experienced with tsduck. Hopefully, my question makes at least a bit of sense.

lars18th, 2021-05-22:
Hi,

This is an interesting idea. In fact what you want to do is a PACKET SHIFT at pid level. But this only has sense with CBR. Perhaps @lelegard can comment if this idea is feasible or not: store N packets in a FIFO and rewrite them in the same pid.

clime, 2021-05-23:
I imagine, the packets would be shifted and in their original place NULL packets would be inserted to maintain CBR (and perhaps level of used NULL bytes will change throughout the whole bitstream?)

But I am not sure why CBR is required. Imho it could work the same just no NULL stuffing would be needed? It would change current bitrate throughout the stream but that wouldn't be a problem at least for me.

Anywyay, I am ok with both CBR and VBR :).

>  store N packets in a FIFO and rewrite them in the same pid.

I don't completely understand this at the moment but hopefully somebody else will :).

lars18th, 2021-05-23:
Hi,

> Anywyay, I am ok with both CBR and VBR :).

Hi, to obtain _correct_ outcomes, from my point of view, the CBR of the pid it's mandatory.

> > store N packets in a FIFO and rewrite them in the same pid.
> 
> I don't completely understand this at the moment but hopefully somebody else will :).

That's the strategy to implement this technically.

I hope @lelegard will want to implement this.
Regards.

lelegard, 2021-05-26:
Hi,

To comment on previous posts, yes, CBR is preferred for the shifted PID. It's obvious. It's similar to an old slide rule or a caliper where you shift the middle bar back and forth. If the middle bar does not have a constant width (ie. CBR), it won't slide smoothly.

Moreover, if you want to shift a PID forward, others would want to shift it backward. Since it is impossible to go back in the past, that means that you need to shift the whole TS forward, except the selected PID (which becomes backward shifted, de facto).

lars18th, 2021-05-26:
Hi @lelegard ,

> Moreover, if you want to shift a PID forward, others would want to shift it backward. Since it is impossible to go back in the past, that means that you need to shift the whole TS forward, except the selected PID (which becomes backward shifted, de facto).

Please, don't think that "others would want to shift backward". This is _directly_ impossible (as you commented). However, as you have pointed as well to go backward you can shift all forward except your target pid. But in this case it's impossible if not all sub-streams are CBR (a not very common scenario). So, I agree with you: this doesn't have sense.

But anyway I found very interesting to do a fine-grade resynch with this functionality. So I hope you want to consider this enhancement.

lelegard, 2021-05-26:
> But in this case it's impossible if not all sub-streams are CBR (a not very common scenario). 

Well, we just need all other PID's to be _globally_ CBR. In the case of a broadcast stream, the TS is CBR, the audio PID's are individually CBR and the video PID's are individually VBR but their sum is CBR (stat mux). This is a case where you can shift forward all PID's around some fixed audio PID's.

Anyway, if the feature is implemented, it will do the shift blindly. The user may expect a correct result when the shifted PID's are globally CBR. Otherwise, the result will depend on the TS bitrate variations.

lelegard, 2021-05-29:
A `pidshift` plugin has been added. Please try it.

clime, 2021-05-29:
> A `pidshift` plugin has been added. Please try it.

Thanks, I have some problems using tsduck for my use-case to actually be able to try pidshift. I have the following script:

```
#!/bin/bash

ffmpeg -y \
    -f lavfi -i "color=size=320x240:rate=30:color=red:duration=1" \
    -f lavfi -i "aevalsrc=2:duration=1:nb_samples=1600:s=48000:c=stereo" \
    -f lavfi -i "color=size=320x240:rate=30:color=green:duration=1" \
    -f lavfi -i "aevalsrc=0:duration=1:nb_samples=1600:s=48000:c=stereo" \
    -f lavfi -i "color=size=320x240:rate=30:color=blue:duration=1" \
    -f lavfi -i "aevalsrc=-2:duration=1:nb_samples=1600:s=48000:c=stereo" \
    -filter_complex "[0:v][1:a][2:v][3:a][4:v][5:a]concat=n=3:v=1:a=1[v][a]" \
    -map "[v]" -map "[a]" \
    -intra \
    -c:v h264 \
    -c:a s302m -strict -2 \
    -f mpegts sample.ts

ffmpeg \
    -re -stream_loop -1 -i sample.ts -c:a copy -c:v copy -f mpegts udp://127.0.0.1:5100
```

to generate input signal by ffmpeg.

Next I tried:

```
tsp -I ip 0.0.0.0:5100 -O ip 127.0.0.1:5000
```

to try to employ tsp in some basic way so that I can try pidshift afterwards. However, when I then do:

```
$ ffplay udp://127.0.0.1:5000
```

I get:

```
[mpegts @ 0x7fe620000c80] decoding for stream 1 failed
[mpegts @ 0x7fe620000c80] Could not find codec parameters for stream 1 (Audio: s302m (BSSD / 0x44535342), 0 channels): unspecified sample format
```

...for audio (video works). This doesn't happen, if I just do:

```
ffplay udp://127.0.0.1:5100
```

...both audio and video work in this case.

I am not sure, maybe it's because  the s302m stream is muxed by ffmpeg as "private"?

lelegard, 2021-05-29:
To view the PSI which were generated by ffmpeg: `tspsi sample.ts`

The command `tsp -I ip 0.0.0.0:5100 -O ip 127.0.0.1:5000` is absolutely transparent. It does not modify any single byte or packet to the transport stream.

clime, 2021-05-29:
> To view the PSI which were generated by ffmpeg: `tspsi sample.ts`

```
$ tspsi sample.ts

* SDT Actual, TID 66 (0x42), PID 17 (0x0011)
  Version: 0, sections: 1, total size: 40 bytes
  - Section 0:
    Transport Stream Id: 1 (0x0001)
    Original Network Id: 65281 (0xFF01)
    Service Id: 1 (0x0001), EITs: no, EITp/f: no, CA mode: free
    Running status: running
    - Descriptor 0: Service (0x48, 72), 18 bytes
      Service type: 0x01 (Digital television service)
      Service: "Service01", Provider: "FFmpeg"

* PAT, TID 0 (0x00), PID 0 (0x0000)
  Version: 0, sections: 1, total size: 16 bytes
  - Section 0:
    TS id:       1 (0x0001)
    Program:     1 (0x0001)  PID: 4096 (0x1000)

* PMT, TID 2 (0x02), PID 4096 (0x1000)
  Version: 0, sections: 1, total size: 32 bytes
  - Section 0:
    Program: 1 (0x0001), PCR PID: 256 (0x0100)
    Elementary stream: type 0x1B (AVC video), PID: 256 (0x0100)
    Elementary stream: type 0x06 (MPEG-2 PES private data), PID: 257 (0x0101)
    - Descriptor 0: Registration (0x05, 5), 4 bytes
      Format identifier: 0x42535344 ("BSSD")
```

> The command `tsp -I ip 0.0.0.0:5100 -O ip 127.0.0.1:5000` is absolutely transparent. It does not modify any single byte or packet to the transport stream.

Strange that it leads to different results with ffplay....maybe bitrate plays a role somehow. ffmpeg produces VBR. Idk what might be the cause...

clime, 2021-05-29:
I am getting also this output from ffplay:

```
[s302m @ 0x7f0e10027e40] frame has invalid header
[mpegts @ 0x7f0e10000940] PES packet size mismatchsq=    0B f=0/0   
[mpegts @ 0x7f0e10000940] Packet corrupt (stream = 1, dts = 136098000).
[s302m @ 0x7f0e10027e40] frame has invalid header
```

Also for video stream, i get messages like:

```
[mpegts @ 0x7fef38000c80] Packet corrupt (stream = 0, dts = 3906000).
[mpegts @ 0x7fef38000c80] Packet corrupt (stream = 0, dts = 3996000).
[mpegts @ 0x7fef38000c80] Packet corrupt (stream = 0, dts = 4086000).
```

but video displays at least.

I am not getting these messages with ffplay when attaching to ffmpeg source directly (without the tsp proxy in the middle). I tried to play around with ffmpeg's muxer option (`-pcr_period`, `-muxrate`) but these have no effect on the problem.

lelegard, 2021-05-30:
To minimize latency, try:
~~~
tsp --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I ip 0.0.0.0:5100 -O ip 127.0.0.1:5000 --enforce-burst
~~~

clime, 2021-05-30:
> To minimize latency, try:
> 
> ```
> tsp --initial-input-packets 7 --max-input-packets 7 --max-flushed-packets 7 -I ip 0.0.0.0:5100 -O ip 127.0.0.1:5000 --enforce-burst
> ```

Good to know how decrease latency, it doesn't help with the issue, still getting the same messages and no audio...

lelegard, 2021-05-30:
This command does really nothing: It reads one UDP datagram (typically 7 TS packets), pass it immediately to the output thread which sends it immediately. And so on, in parallel, without waiting. The problem must be elsewhere, maybe in the network socket buffering. Remember that UDP is unreliable by definition and network tuning is essential.

clime, 2021-05-30:
Well, UDP on localhost is going to be reliable. I can replicate the problem on two differrent machines.

`$ socat UDP-LISTEN:5100 UDP:127.0.0.1:5000` works as expected.

`$ tsp -I ip :5100 -O ip 127.0.0.1:5000` causes ffplay to spit errors.

Idk, what is the cause but regrettably, I won't be able to test pidshift.

lars18th, 2021-05-31:
Hi @clime ,

Please, take note that ffmpeg is a very good tool, but it's innefficient to generate DTV conformant streams. For example, never try to send MPEG-TS streams over UDP without using the parameter `?pkt_size=1316`. Another question is: why you not use MP2  as the audio codec? In fact, I don't understand your use case. So please, can you explain it with more detail? I feel all the troubles that you have are related to the MPEG-TS generator (in this case the ffmpeg tool) and not by the TSDuck.

clime, 2021-05-31:
I cannot use mp2 as audio codec because i need raw audio format. Encoding disturbs audio signal, which would complicate checking for audio/video alignment later.

`?pkt_size=1316` fixed the problem. Probly related to: https://trac.ffmpeg.org/ticket/2748#:~:text=All%20IP%20UDP%20packets%20should,(1%20MPEG%20TS%20packet).

I performed test with DVBInspector and command ` tsp -P pidshift --pid 257 --time 3000 -I ip :5100 -O ip 127.0.0.1:5000` and I could see the pes packets are shifted. Thanks

clime, 2021-06-01:
Looking again at the output from DVBInspector, it seems pidshift rather shifts backwards instead of forwards:

[image]

Command used was:

` tsp -P pidshift --pid 257 --time 3000 -I ip :5100 -O ip 127.0.0.1:5000`

The original (produced by ffmpeg) looks like this (it has the green triangles exactly behind the blue circles):

[image]

The audio should have been shifted above the pcr line, not below.

Backward audio shift does nothing, it seems:

`tsp -P pidshift --pid 257 --backward --time 3000 -I ip :5100 -O ip 127.0.0.1:5000`

[image]

Video shift also doesn't seem to do anything:

` tsp -P pidshift --pid 256 --time 3000 -I ip :5100 -O ip 127.0.0.1:5000`

[image]

lelegard, 2021-06-01:
It depends on what you mean by "forward", "backward" and "does nothing".

Shifting "forward" a PID means shifting all packets of that PID forward in the TS.

Let's take an example. On an input TS, at packets index 100, 200 and 300 _in the TS_ you have packets from an audio PID. At index 100, 200, 300 _in the TS_, you have packets number 10, 20, 30 _in the audio PID_. These packets have PTS1, PTS2 and PTS3. Now, assuming that we shift the audio PID _forward_ in the TS, by an amount of 100 packets, you get this:

| Index in TS | Index in PID (before) | PTS in PID (before) | Index in PID (after) | PTS in PID (after)
| - | - | - | - | - 
| 100 | 10 | PTS1 | null packet | n/a
| 200 | 20 | PTS2 | 10 | PTS1
| 300 | 30 | PTS3 | 20 | PTS2

So, "forward" depends on your standpoint. If your standpoint is "packet 200 in TS", the PTS which was PTS2 becomes PTS1. So, you may say "from my standpoint on the side of the road, the audio PID _packets_ shifted forward while the audio PID _time stamps_ shifted backward". But if your standpoint is the audio PID, let's say in packet 10, then you may say "from my standpoint in the audio PID, I moved forward in the TS from packet 100 to packet 200". There was a guy named Einstein who called that "relativity" 😉 

Similarly, when you say "shift backward does nothing", the documentation says (please read it carefully) that shifting a PID backward is impossible by design and is implemented as shifting forward all the rest of the TS. So, again, if your standpoint is the TS file, you will see that the audio PID did not shift (but everything else did) and if your standpoint is the rest of the TS (video for instance), you will see that the audio PID shifted backward. And again, when you see the packets shifting backward, you see the timestamp values shifting forward.

So, please read the plugin documentation carefully, understand the meaning of "backward" and "forward" which are opposite for packets and time stamps. In all cases, audio and video PID's shift from each other, in one direction or the other, "backward" and "forward" just being relative words.

clime, 2021-06-01:
> It depends on what you mean by "forward", "backward" and "does nothing".
>
> Shifting "forward" a PID means shifting all packets of that PID forward in the TS.
>
> Let's take an example. On an input TS, [...]

Well it seems I looked at it from the point of receiver while you look at it from the point of static file?

Anyway, is there a way, to move the audio pid above the video pid 3 secs in the graph above? This is what I meant - that audio will be sent much sooner in then video.

> Similarly, when you say "shift backward does nothing", the documentation says (please read it carefully) that shifting a PID backward is impossible by design and is implemented as shifting forward [...]

For command `tsp -P pidshift --pid 257 --backward --time 3000 -I ip :5100 -O ip 127.0.0.1:5000` (see my prev comment) i see that audio did not shift with respect to video per the graph - the line overlap - they shouldn't overlap.

> 
> So, please read the plugin documentation carefully, understand the meaning of "backward" and "forward" which are opposite for packets and time stamps. In all cases, audio and video PID's shift from each other, in one direction or the other, "backward" and "forward" just being relative words.

The docs should explain what is meant by the individual words then. Also I think it would be better if forward is taken from the point of MPEG TS receiver which is quite natural interpretation. The current default behavior tends to move pids behind PCR producing an invalid stream.s

lars18th, 2021-06-01:
Hi,

> The docs should explain what is meant by the individual words then. Also I think it would be better if forward is taken from the point of MPEG TS receiver which is quite natural interpretation. The current default behavior tends to move pids behind PCR producing an invalid stream.

From my point of view, the "natural interpretation" is that the TS packet is shifted "back in time". That's because it's impossible to advance (in the real world) the time. So that's the common interpretation of _any_ engineer. But the problem is how we can interpret this in the _output stream_. In this case, what you see could be contraintuitive, as the shifted packets arrive _later_ but with the _same_ timestamp.

lelegard, 2021-06-02:
@clime, @lars18th, anyone's "natural" interpretation is one's own interpretation of course !

I have updated the user's guide to clarify that moving packets forward means moving the timestamps values backward and vice versa.

clime, 2021-06-02:
Great! 

So how can I achieve approximately this:

[image]

i.e. move the yellow line above the blue line given that they overlap initially (i.e. on tsp input they overlap)? No combination that I had tried did that. I meanwhile did it by some other means but originally I wanted to use tsduck for it.

lelegard, 2021-06-02:
I have tried the same kind of test, using DVBinspector to get the same type of graph and `pidshift` does the job.

With a default forward shift, the audio line is going down on the graph. With a `--backward` shift, the audio line is going up the PCR line (in fact, the PCR line and everything else are shifted to the right while the audio line stays in place, with the result of the audio line being above the PCR line).

clime, 2021-06-02:
I was able to get the expected results now (i.e shift the audio line above video with --backward). Sries, i messed up something before :/. I guess this can be closed then.

lelegard, 2021-06-02:
> I was able to get the expected results now (i.e shift the audio line above video with --backward). Sries, i messed up something before :/. I guess this can be closed then.

Cool, I am glad to read that.

closed 2021-06-02 by lelegard