tsduck/tsduck #795: Can't compile TSDUCK on CentOs 7[Q]

Plombirius, 2021-05-26:
I can't compile tsduck on Centos 7.9.002 with gcc 4.8.5 20150623 (Red Hat 4.8.5-44) (GCC). I tried it on TS Duck ver 3.26 and 3.25

I have used this command:  make NOPCSC=1 NOCURL=1 NODTAPI=1 -j 4 and 
have got this error message:

>   [CXX] dtv/signalization/tsTablePatchXML.cpp
>   [CXX] dtv/signalization/tsTablesDisplay.cpp
> plugin/tsTSProcessorArgs.cpp: In member function ‘virtual bool ts::TSProcessorArgs::loadArgs(ts::DuckContext&, ts::Args&)’:
> plugin/tsTSProcessorArgs.cpp:225:52: error: no matching function for call to ‘ts::Args::getFixedValue(ts::BitRate&, const char16_t [8], int)’
>      args.getFixedValue(fixed_bitrate, u"bitrate", 0);
>                                                     ^
> plugin/tsTSProcessorArgs.cpp:225:52: note: candidate is:
> In file included from plugin/tsPlugin.h:36:0,
>                  from plugin/tsPluginOptions.h:36,
>                  from plugin/tsTSProcessorArgs.h:37,
>                  from plugin/tsTSProcessorArgs.cpp:30:
> /home/vkantor/tsduck-tsduck-ebd3e5d/src/libtsduck/base/tsArgs.h:954:14: note: template<class FIXED> void ts::Args::getFixedValue(FIXED&, const UChar*, ts::FixedPoint<typename FIXED::int_t, FIXED:: PRECISION>, size_t) const
>          void getFixedValue(FIXED& value,
>               ^
> /home/vkantor/tsduck-tsduck-ebd3e5d/src/libtsduck/base/tsArgs.h:954:14: note:   template argument deduction/substitution failed:
> plugin/tsTSProcessorArgs.cpp:225:52: note:   mismatched types ‘ts::FixedPoint<typename FIXED::int_t, FIXED:: PRECISION>’ and ‘int’
>      args.getFixedValue(fixed_bitrate, u"bitrate", 0);
>                                                     ^
>   [CXX] dtv/signalization/tsTablesLogger.cpp
> make[2]: *** [/home/vkantor/tsduck-tsduck-ebd3e5d/bin/release-x86_64-srt03/objs-libtsduck/tsTSProcessorArgs.o] Error 1
> make[2]: *** Waiting for unfinished jobs....
> make[1]: *** [default] Error 2
> make: *** [default] Ошибка 2
> 

I tried it on 3.26 and 3.25. So what do I need to do for compile it successfully?

lelegard, 2021-05-26:
This looks like a GCC 4.8.5 bug. This is a prehistoric version of GCC (current one is 11.1.1 on recent distros).

I no longer have CentOS 7 systems but I have a CentOS 8 systems with several old versions of GCC, side by side, including 4.8.5. I can reproduce your issue. GCC 4.8.5 erroneously reports errors on valid code which was recently added to TSDuck.

I have modified the source code as a workaround for this old GCC bug. It works on my CentOS 7 with GCC 4.8.5. Please try to pull the repo and rebuild in your environment.

Plombirius, 2021-05-27:
I have built it successfully. Thank you very much.
I prefer CentOs 8, but we were forced back to CentOs 7 due they set EoL of CentOs 8 on 31.12.2021 :(

closed 2022-09-23 by lelegard