tsduck/tsduck #796: [Q] New Modulation Types

RolsimGit, 2021-05-26:
There are new Modulation Types

NBC 8PSK
NBC QPSK

What can i do for it

Kind regards Rolsim

lelegard, 2021-05-27:
Hi @RolsimGit 

I am not aware of those. Are they different from 8-PSK and QPSK as seen in DVB-S2?

How would you like to use them with TSDuck? Is there any USB or PCIe tuner supporting them (if different from DVB-S2)?

We also need support for new modulations in the operating system. I just checked the Linux TV API on a kernel 5.12.6 (Fedora 34 distro) and I don't see anything new (nothing more that 8-PSK and QPSK for DVB-S2). On Windows, DirectShow hasn't been seriously improved for years and I doubt that new modulations will ever be supported. Only Linux is closely following the evolution of modulations. Windows doesn't care.

RolsimGit, 2021-05-27:
[image]
BDA knows the two modulation types as

BDA_MOD_NBC_8PSK and
BDA_MOD_NBC_QPSK

I get no signal from transponders with this modulation type

Example on Astra 19.2

11009000000, POL_VERTICAL  , PSK_8, ROLLOFF_AUTO, FEC_3_4, 23500000, DS_DVB_S2 // NBC 8PSK
12090000000, POL_VERTICAL  , PSK_8, ROLLOFF_25  , FEC_2_3, 29700000, DS_DVB_S2 // NBC 8PSK
12304500000, POL_HORIZONTAL, QPSK , ROLLOFF_35  , FEC_9_10,27500000, DS_DVB_S2 // NBC QPSK

and more.

lelegard, 2021-05-27:
OK, `BDA_MOD_NBC_8PSK` and `BDA_MOD_NBC_QPSK` are not new. They are old definitions in Windows DirectShow with NBC meaning "Non-Backward Compatible" versions of QPSK and 8-PSK. I have no idea what it was. I have not seen any reference to that elsewhere.

From your original post, I thought that NBC (the US TV network) had released some new modulation schemes. Sorry for the confusion.

> I get no signal from transponders with this modulation type

What makes you think that a given transponder would use this "Non-Backward Compatible" version of QPSK and 8-PSK? In which way is this "Non-Backward Compatible" from the corresponding DVB-S2 modulations?

RolsimGit, 2021-05-27:
[image]

As a test, I check all transponders in my program. The parameters of the not receivable transponders I have compared e.g. with ProgDVB, because I have here ine signal. Therefore I have come to these modulation types.
[image]

lelegard, 2021-05-27:
The transponder you show, Astra 1M 19.2E 12.090, is a [CanalSat transponder according to Lyngsat](https://www.lyngsat.com/muxes/Astra-1M_Europe_12090-V.html). The CanalSat transponders are usually very standard.

Using ProgDVB, when you specify 8PSK, does this work? Or does it work only with NBC-8PSK? 

When using TSDuck, did you specify `--roll-off 0.25`? I see from your table that this transponder uses 0.25 while the default is 0.35. Lyngsat does not specify the roll-off factor, unfortunately.

RolsimGit, 2021-05-27:
tsp -O file test.ts -P until --seconds 20 -I dvb --delivery-system DVB-S2 --lnb extended --frequency 12009000000 --polarity vertical --symbol-rate 29700000 --device-name :1

This works.

ProgDVB has not 8PSK,, only NBC-8PSK ok, my mistake.

But all parameters in my program lead to the same failure on this transponder.

I continue to research and share my progress. However, I would like to express my thanks here for the extremely nice help.

lelegard, 2021-05-27:
> But all parameters in my program lead to the same failure on this transponder.

Do you mean you write a C++ program using the TSDuck library and it does not work?

If this is the case, read the documentation of the `dvb` plugin, carefully spot the default values for each parameter and also set the same default values in your `ModulationArgs`, not only the parameters you specify on the command line. The command line parameters have default values to avoid some typing but C++ code needs all values.

RolsimGit, 2021-05-27:
I've got it;)

TAGDISEQC tagDiseqc[4] =
{
//	{0, _T("Astra 19.2")  , _T("Astra")   , ART_ASTRA_19_2  , ART_SATELLIT},
	{0, _T("Astra 19.2")  , **_T("extended")** , ART_ASTRA_19_2  , ART_SATELLIT},

I have now changed the parameters for the LNB. It works now fine.

closed 2022-09-23 by lelegard